espressif/esp-idf #15287: Strange warnings

diplfranzhoepfinger, 2025-01-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

```
NOTICE: Processing 1 dependencies:
NOTICE: [1/1] idf (5.4.0)
-- Project sdkconfig file /home/franz/git/hr/CTIS/CTIS/build_ESP32_dual_out/sdkconfig
Loading defaults file /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults...
warning: unknown kconfig symbol 'BT_CTRL_BLE_MAX_ACT' assigned to '10' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_BLE_STATIC_ACL_TX_BUF_NB' assigned to '12' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_PINNED_TO_CORE_1' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_DFT_TX_POWER_LEVEL_P21' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_DISCARD_THRSHOLD' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'BT_CTRL_BLE_SCAN_DUPL' assigned to 'n' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'SPIRAM_FETCH_INSTRUCTIONS' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
warning: unknown kconfig symbol 'SPIRAM_RODATA' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults
Loading defaults file /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32...
warning: unknown kconfig symbol 'BT_CTRL_BLE_MAX_ACT' assigned to '10' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_STATIC_ACL_TX_BUF_NB' assigned to '12' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_PINNED_TO_CORE_1' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_DFT_TX_POWER_LEVEL_P21' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_DISCARD_THRSHOLD' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_SCAN_DUPL' assigned to 'n' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'SPIRAM_FETCH_INSTRUCTIONS' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'SPIRAM_RODATA' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
Loading defaults file /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32...
warning: unknown kconfig symbol 'BT_CTRL_BLE_MAX_ACT' assigned to '10' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_STATIC_ACL_TX_BUF_NB' assigned to '12' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_PINNED_TO_CORE_1' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_DFT_TX_POWER_LEVEL_P21' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_DISCARD_THRSHOLD' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'BT_CTRL_BLE_SCAN_DUPL' assigned to 'n' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'SPIRAM_FETCH_INSTRUCTIONS' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
warning: unknown kconfig symbol 'SPIRAM_RODATA' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.esp32
Loading defaults file /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out...
warning: unknown kconfig symbol 'BT_CTRL_BLE_MAX_ACT' assigned to '10' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_BLE_STATIC_ACL_TX_BUF_NB' assigned to '12' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_PINNED_TO_CORE_1' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_DFT_TX_POWER_LEVEL_P21' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_BLE_ADV_REPORT_DISCARD_THRSHOLD' assigned to '1000' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'BT_CTRL_BLE_SCAN_DUPL' assigned to 'n' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'SPIRAM_FETCH_INSTRUCTIONS' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
warning: unknown kconfig symbol 'SPIRAM_RODATA' assigned to 'y' in /home/franz/git/hr/CTIS/CTIS/sdkconfig.defaults.ESP32_dual_out
-- Compiler supported targets: xtensa-esp-elf

```

ESP-Marius, 2025-01-27:
> in fact, these Params are set correctly,

Are you sure? From what I see you are compiling for esp32 and e.g. `SPIRAM_FETCH_INSTRUCTIONS` or `BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_NUM` do not seems to exists for that target. So it correctly warns you that these configs are unknown.

dobairoland, 2025-03-27:
I think we can close this for inactivity.

closed 2025-03-27 by dobairoland

diplfranzhoepfinger, 2025-03-27:
Hi,

in fact: this come from a Dual Build with ESP32 and ESP32-S3.
there it is impossible to use a sdkconfig.defaults File,
but must use a  sdkconfig.defaults.esp32 and a  sdkconfig.defaults.esp32s3
in addition.

then this Problem is gone.