espressif/esp-idf #15289: Component registry is not reachable while requesting for resolve dependencies via CMake

hathorr-hub, 2025-01-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We are following to create a project using the information has been given in this link: [LINK](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md)

We are using VSCode ESP_IDF for the implementation. 

While we are trying to build the project we are getting error for component which is defined in the idf_component.yml

[image]

Error is shown below :

[image]

Note: we have gone through the available solutions provided but not helping to solve my issue.

mantriyogesh, 2025-01-27:
As the error points out, are you connected to internet?

mantriyogesh, 2025-01-27:
Additionally, as https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md#31-adding-components

Suggests, please remove extconn from idf_component.yml to use ESP-Hosted-MCU.

mantriyogesh, 2025-01-28:
@hathorr-hub, any updates on this please?

hathorr-hub, 2025-02-03:
Hello. Yes we are able to resolve this with this command "pip install pip_system_certs".

mantriyogesh, 2025-02-03:
Thank you for the update @hathorr-hub , If you are unblocked on this, Can we close the issue for now? You always can re-open or file a new issue for different problem.

Alvin1Zhang, 2025-03-27:
Thanks for reporting, will close due to inactivity, feel free to reopen.

closed 2025-03-27 by Alvin1Zhang