espressif/esp-idf #15290: [USB][ABORT][v5.1.4] root_port_handle_events() hits HUB_DRIVER_STATE_ROOT_RECOVERY

chipweinberger, 2025-01-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

esp32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

no crash

### What is the actual behavior?

occasionally, crashes in hub.c

### Steps to reproduce.

1. plug & replug usb devices

### Debug Logs.

```plain
==================== CURRENT THREAD STACK =====================
#0  0x40376056 in panic_abort (details=0x3c3c73da "abort() was called at PC 0x4208cf73 on core 1") at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/panic.c:466
#1  0x40384838 in esp_system_abort (details=0x3c3c73da "abort() was called at PC 0x4208cf73 on core 1") at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4038dbca in abort () at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/newlib/abort.c:38
#3  0x4208cf76 in root_port_handle_events (root_port_hdl=<optimized out>) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/usb/hub.c:892
#4  hub_process () at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/usb/hub.c:1109
#5  0x42087a74 in usb_host_lib_handle_events (timeout_ticks=<optimized out>, event_flags_ret=<optimized out>) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/usb/usb_host.c:539
#6  0x4204639a in usbHostTaskLoop (arg=<optimized out>) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jcommon/usb/pd_usbh_service.cpp:74
#7  0x403858a9 in vPortTaskWrapper (pxCode=0x42046338 <usbHostTaskLoop(void*)>, pvParameters=0x0) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

chipweinberger, 2025-01-27:
I filed this same issue *nearly 3 years ago* and it's still not fixed.

CAN YOU PLEASE FIX IT!

https://github.com/espressif/esp-idf/issues/9707

https://github.com/espressif/esp-idf/issues/8748

chipweinberger, 2025-01-27:
@peter-marcisovsky 
@Dazza0 
@igrr

chipweinberger, 2025-01-27:
The original issues from ~3 years ago were repro'd using an **AKAI Professional LPK25 piano**: 

https://www.amazon.com/gp/product/B002M8GBDI?th=1

The customer repo is on a **Yamaha P-525**

https://www.amazon.com/dp/B0CJ5NPG1M/

roma-jam, 2025-01-27:
Hi @chipweinberger , 

thanks to bringing it up again. 
We refactored quite a lot the hub root handling, starting from the `release v5.2`. 
But it was not backported to `v5.1`, the changes are not there and the code is pretty much as in `v4.4`. 

The difficult part is that the problem is quite hard to be reporduced with regular usb devices..

So, if you have a chance, is that possible to try it out with any release, starting from `v5.2`?

chipweinberger, 2025-01-27:
Thanks for the reply.

This product is shipping already, so I can't easily switch to v5.2 without very good reason.

IMO, this should be fixed in v5.1 since that is still a currently supported release.

chipweinberger, 2025-01-27:
Even on master, it's still largely the same code. `abort();    // Should never occur` still exists.

```
static void root_port_handle_events(hcd_port_handle_t root_port_hdl)
{
    hcd_port_event_t port_event = hcd_port_handle_event(root_port_hdl);
    switch (port_event) {
    case HCD_PORT_EVENT_NONE:
        // Nothing to do
        break;
    case HCD_PORT_EVENT_CONNECTION: {
        if (hcd_port_command(root_port_hdl, HCD_PORT_CMD_RESET) != ESP_OK) {
            ESP_LOGE(HUB_DRIVER_TAG, "Root port reset failed");
            goto reset_err;
        }
        ESP_LOGD(HUB_DRIVER_TAG, "Root port reset");
        usb_speed_t speed;
        if (hcd_port_get_speed(p_hub_driver_obj->constant.root_port_hdl, &speed) != ESP_OK) {
            goto new_dev_err;
        }

        if (new_dev_tree_node(NULL, 0, speed) != ESP_OK) {
            ESP_LOGE(HUB_DRIVER_TAG, "Failed to add new device");
            goto new_dev_err;
        }

        // Change Port state
        HUB_DRIVER_ENTER_CRITICAL();
        p_hub_driver_obj->dynamic.root_port_state = ROOT_PORT_STATE_ENABLED;
        HUB_DRIVER_EXIT_CRITICAL();
        break;
new_dev_err:
        // We allow this to fail in case a disconnect/port error happens while disabling.
        hcd_port_command(p_hub_driver_obj->constant.root_port_hdl, HCD_PORT_CMD_DISABLE);
reset_err:
        break;
    }
    case HCD_PORT_EVENT_DISCONNECTION:
    case HCD_PORT_EVENT_ERROR:
    case HCD_PORT_EVENT_OVERCURRENT: {
        bool port_has_device = false;
        HUB_DRIVER_ENTER_CRITICAL();
        switch (p_hub_driver_obj->dynamic.root_port_state) {
        case ROOT_PORT_STATE_POWERED: // This occurred before enumeration
        case ROOT_PORT_STATE_DISABLED: // This occurred after the device has already been disabled
            // Therefore, there's no device object to clean up, and we can go straight to port recovery
            p_hub_driver_obj->dynamic.port_reqs |= PORT_REQ_RECOVER;
            p_hub_driver_obj->dynamic.flags.actions |= HUB_DRIVER_ACTION_ROOT_REQ;
            break;
        case ROOT_PORT_STATE_NOT_POWERED: // The user turned off ports' power. Indicate to USBH that the device is gone
        case ROOT_PORT_STATE_ENABLED: // There is an enabled (active) device. Indicate to USBH that the device is gone
            port_has_device = true;
            break;
        default:
            abort();    // Should never occur
            break;
        }
        HUB_DRIVER_EXIT_CRITICAL();

        if (port_has_device) {
            ESP_ERROR_CHECK(dev_tree_node_dev_gone(NULL, 0));
        }

        break;
    }
    default:
        abort();    // Should never occur
        break;
    }
}
```

tore-espressif, 2025-02-20:
Hello @chipweinberger 

We are having hard time reproducing the problem. But after some thorough static analysis we found out that it can crash in the way you are describing if the driver gets 2 errors in very short time. That is if it gets 2nd error before it recovered from 1st error

Could you please try this fix and let me know if it helps? It is based on idf 5.1.4, you can try it in this branch https://github.com/tore-espressif/esp-idf/tree/fix/hub_error_handling

[image]

chipweinberger, 2025-02-20:
Oh, okay, I see you adjusted the abort condition to allow RECOVERY.

Yes, my investigation [here](https://github.com/espressif/esp-idf/issues/9707#issuecomment-1236010030) confirms that will definitely prevent the abort.

But of course there is still the other quesion of - will the USB device successfully recover? I'll test it.

roma-jam, 2025-02-20:
@chipweinberger 

after refactoring we do not set the ROOT_PORT_STATE_RECOVERY anywhere, so the root hub port is never in this state. 

So, if there will be several root hub port events in a row where we need to recover the port after, we set up the flag that port requires recovering and later change it's state to ROOT_PORT_STATE_POWERED (if port wasn't powered off intentionally via `usb_host_lib_set_root_port_power` call)

Anyway, feel free to try it out and share the result.

closed 2025-03-03 by tore-espressif