espressif/esp-idf #15294: NIMBLE ASSERT PROBLEM

filzek, 2025-01-27:
Could you guys fix the assert NIMBLE problem? 
We do not want that the NIMBLE takes the ASSERT to crash always, its impossible to handle the asserts locks inside the NIMBLE the way it is managed, so, without ASSERT it works perfect, but turning on the ASSERT it will not work.

can provide tons of example:
assert failed: ble_hs_conn_find ble_hs_conn.c:302 (ble_hs_locked_by_cur_task())

always same problem, we just have a running core and tasks in the secondary core, and the problem happen no matter which core or task , the problem happen. 

can you guys could provide a true solution to solve this on ESP32 ?

filzek, 2025-01-27:
assert failed: ble_hs_conn_find ble_hs_conn.c:302 (ble_hs_locked_by_cur_task())

Backtrace: 0x400824d8:0x3ffd7720 0x40099355:0x3ffd7740 0x4009d38d:0x3ffd7760 0x4014678e:0x3ffd7890 0x4013b02a:0x3ffd78b0 0x4013dbff:0x3ffd78d0 0x4013dc83:0x3ffd7930 0x401378da:0x3ffd7990 0x40137d47:0x3ffd79b0 0x40136e65:0x3ffd79d0 0x401ffaab:0x3ffd79f0 0x40088149:0x3ffd7a10 0x400dcab1:0x3ffd7a40
--- 0x400824d8: panic_abort at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_system/panic.c:454
0x40099355: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_system/port/esp_system_chip.c:92
0x4009d38d: __assert_func at C:/Espressif/frameworks/esp-idf-v5.4/components/newlib/assert.c:80
0x4014678e: ble_hs_conn_find at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_conn.c:302 (discriminator 1)
 (inlined by) ble_hs_conn_find at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_conn.c:294 (discriminator 1)
0x4013b02a: ble_gattc_connection_broken at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:5382
0x4013dbff: ble_gap_conn_broken at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1550
0x4013dc83: ble_gap_rx_disconn_complete at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1595 (discriminator 4)
0x401378da: ble_hs_hci_evt_disconn_complete at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:311
 (inlined by) ble_hs_hci_evt_disconn_complete at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:238
0x40137d47: ble_hs_hci_evt_process at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1336
0x40136e65: ble_hs_event_rx_hci_ev at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:531
0x401ffaab: npl_freertos_event_run at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451
0x40088149: ble_npl_event_run at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at C:/Espressif/frameworks/esp-idf-v5.4/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:309

KaeLL, 2025-01-27:
#15014, although it seems like you already know that.

strange969, 2025-01-28:
Hi @filzek, 

We are actively debugging this issue. As seen this issue is related to enabling of asserts. This is upstream mynewt code and no changes so far from our side are done under this flag. By default, esp-idf has this flag disabled. 

We suspect incorrect handling of nested locks resulting in this behaviour. Nimble code has references to nested locks to maintain atomicity of resources and hence failure in this path can lead to asserts in various use cases. 

Attached change should fix this problem as seen in our local testing. However, while we test in more detail to cover all applications/use cases, we would appreciate if you too can give a try of this patch at your end to see if it helps. In case
it doesn't request you to kindly let us know your testing sequence.

[assert_crash_fix.txt](https://github.com/user-attachments/files/18569086/assert_crash_fix.txt)

filzek, 2025-01-28:
> Hi [@filzek](https://github.com/filzek),
>
> We are actively debugging this issue. As seen this issue is related to enabling of asserts. This is upstream mynewt code and no changes so far from our side [...]

We have modified the assert behavior in NimBLE Mynewt so that assertions are effectively disabled, making it similar to having debugging turned off.

For now, we believe the best practice is to keep NimBLE asserts disabled until all fixes are fully reviewed and merged into the main or release branches.

We will test the patches you proposed next week.

strange969, 2025-03-27:
Closing issue, changes mentioned in comment above have been [merged](https://github.com/espressif/esp-idf/commit/30589c306c285ffa12000332aeab2c833525aa19). Please feel to reopen in case of any further updates are required

closed 2025-03-27 by rahult-github