espressif/esp-idf #15296: `-O2` build causing warnings and errors

KaeLL, 2025-01-28:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1655-gc5865270b5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build successfully, without warnings.

### What is the actual behavior?

Warnings from the sdmmc driver, and a bunch of errors from the NimBLE library

### Steps to reproduce.

<details>
<summary>defconfig</summary>

```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.5.0 Project Minimal Configuration
#
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota_encrypted.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_DISABLE=y
CONFIG_COMPILER_ASSERT_NDEBUG_EVALUATE=n
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=0
CONFIG_BT_NIMBLE_MAX_CCCDS=0
CONFIG_BT_NIMBLE_ROLE_CENTRAL=n
CONFIG_BT_NIMBLE_ROLE_BROADCASTER=n
CONFIG_BT_NIMBLE_SECURITY_ENABLE=n
CONFIG_BT_NIMBLE_SVC_GAP_DEVICE_NAME="test"
CONFIG_BT_NIMBLE_GATT_MAX_PROCS=1
CONFIG_BT_NIMBLE_HS_FLOW_CTRL=n
CONFIG_BT_NIMBLE_WHITELIST_SIZE=1
CONFIG_BT_NIMBLE_HIGH_DUTY_ADV_ITVL=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=1
CONFIG_BTDM_CTRL_MODEM_SLEEP=n
CONFIG_BTDM_BLE_SCAN_DUPL=n
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP=n
CONFIG_ESP_ERR_TO_NAME_LOOKUP=n
CONFIG_SPI_MASTER_IN_IRAM=y
[... 150 lines cut ...]
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:765:45: error: array subscript idx is outside array bounds of 'struct ble_store_value_local_irk[0]' [-Werror=array-bounds=]
  765 |     *value_irk = ble_store_config_local_irks[idx];
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_read':
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:71:5: note: while referencing 'ble_store_config_local_irks'
   71 |     ble_store_config_local_irks[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'ble_store_config_read_csfc',
    inlined from 'ble_store_config_read' at /home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:1056:14:
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:978:41: error: array subscript idx is outside array bounds of 'struct ble_store_value_csfc[0]' [-Werror=array-bounds=]
  978 |     *value_csfc = ble_store_config_csfcs[idx];
      |                   ~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config_priv.h: In function 'ble_store_config_read':
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config_priv.h:40:5: note: while referencing 'ble_store_config_csfcs'
   40 |     ble_store_config_csfcs[MYNEWT_VAL(BLE_STORE_MAX_CSFCS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_host_check_status':
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:9260:5: error: array subscript 0 is outside array bounds of 'ble_addr_t[0]' [-Werror=array-bounds=]
 9260 |     ble_store_util_bonded_peers(&oldest_peer_id_addr[0],
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 9261 |                     &num_peers, MYNEWT_VAL(BLE_STORE_MAX_BONDS));
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nvmd/Documents/code/.idf/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:9217:16: note: while referencing 'oldest_peer_id_addr'
 9217 |     ble_addr_t oldest_peer_id_addr[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
```

Shreeyash-17, 2025-01-29:
@KaeLL I have tried to reproduce this issue using the provided sdkconfig, but on my side, it is building correctly without an error/warning.

Please note, I found some configuration in your defconfig that is not part of our master branch. 

CONFIG_SPI_MASTER_IN_IRAM=y
CONFIG_SPI_SLAVE_ISR_IN_IRAM=n
CONFIG_ETH_DMA_BUFFER_SIZE=256
CONFIG_ETH_DMA_RX_BUFFER_NUM=3
CONFIG_ETH_DMA_TX_BUFFER_NUM=3
CONFIG_ETH_USE_SPI_ETHERNET=n
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=2048
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
CONFIG_ESP_COREDUMP_CHECK_BOOT=n
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=30

Could you please share your entire sdkconfig file so that I can compare it? Also, which application/example are you trying to build?

Below is the console log after monitoring :
[console_log.txt](https://github.com/user-attachments/files/18585807/console_log.txt)

KaeLL, 2025-01-29:
The NimBLE errors are another instance of #14712, whose fix, for whatever reason, hasn't been merged yet.

I'm not building any IDF examples, it's a build of one of the closed-source apps I work on. Hence the defconfig.

rahult-github, 2025-01-30:
> The NimBLE errors are another instance of https://github.com/espressif/esp-idf/issues/14712, whose fix, for whatever reason, hasn't been merged yet.

The changes for the mentioned issue are already part of source code [here](https://github.com/espressif/esp-idf/commit/f8ccb29d1210d35f23257d20cac1f040a0662b86) and [here](https://github.com/espressif/esp-nimble/commit/3c8aa5456efefabe0f7343fae6512a0ee51b48dd). 

But looks like some more fixes may be needed.  Understood you can't share close-source apps information, but we are looking for a way to reproduce issue so as to fix it correctly .

Shreeyash-17, 2025-03-07:
Closing issue since idf examples/configuration changes do not show such behavior. I tried different combinations but was not able to reproduce this issue. Please feel free to reopen if you can share anyway for us to reproduce

closed 2025-03-07 by rahult-github