espressif/esp-idf #15297: Advertise my own service uuid during Bluetooth Mesh / BLE coexistence

lebrislo, 2025-01-29:
### Is your feature request related to a problem?

Hi,

Here's my problem explained : https://github.com/espressif/esp-idf/issues/15217

### Describe the solution you'd like.

I would be a benefit to be allowed to add my own ble service uuid along with the current mesh proxy service (0x1828) in the BLE advertising data.

The expected behavior is exactly what's happening when advertising with BLE only  with `esp_ble_adv_data_t`. It would be great to add my services uuid inside `esp_ble_mesh_ble_adv_param_t`.

forx157, 2025-03-31:
Ok, I will discuss this with the team and subsequent updates will be updated in this issue.

lebrislo, 2025-10-30:
Hi, I coming back to you on this matter. Have you discussed with the teams about this subject ? Thanks