espressif/esp-idf #15298: Ring Buffer documentation: clarification needed about multi-thread safety

DurvalMenezes, 2025-01-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In [the documentation for xRingbufferGetCurFreeSize](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#_CPPv425xRingbufferGetCurFreeSize15RingbufHandle_t), there's the following:

> Warning
> This API is not thread safe. So, if multiple threads are accessing the same ring buffer, it is the application's responsibility to ensure atomic access to this API and the subsequent Send

By "this API", does it mean just `xRingbufferGetCurFreeSize()` API function, or the entire ring buffer API including all its other functions? 

In other words, if I don't use `xRingbufferGetCurFreeSize()` but instead just use `xRingbufferSend()` and `xRingbufferReceive()` to the same ring buffer concurrently in separate threads, do these threads need to use semaphore or a similar mechanism to ensure atomic access to the ring buffer? 

TIA!

KonstantinKondrashov, 2025-01-30:
Hi @DurvalMenezes!
The comment stating "This API is not thread safe." appears to be incorrect because [`xRingbufferGetCurFreeSize()`](https://github.com/espressif/esp-idf/blob/master/components/esp_ringbuf/ringbuf.c#L1270-L1272) uses a critical section (via portENTER_CRITICAL and portEXIT_CRITICAL) to protect access to the ring buffer's internal state. So it is now a thread-safe function. Probably, the description was not updated after a fix. We will update the comment.

DurvalMenezes, 2025-01-30:
Thanks @KonstantinKondrashov! From your response, I understand not only  xRingbufferGetCurFreeSize(), but most/all of the other functions in this API are thread-safe, and I am closing this issue and proceeding  accordingly -- but if I'm wrong, please let me know! 👍