espressif/esp-idf #15299: BLE_HCI_SCAN_FILT_MAX conflict with BLE_HCI_SCAN_FILT_USE_WL_INITA

UmutUnay, 2025-01-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

- [x] Checked ESP-IDF Documents
- [x] Checked Related Issues
- [x] Working on Windows 10

I am working on esp-idf v5.3.2 . I need to cancel the duplicate BLE extended advertisements captured by central device, I have searched for it and can't seem to find the solution (None worked for me, I have tried all). I have opened the CONFIG_BT_CTRL_BLE_SCAN_DUPL and related options, but none of it helped. **Any help would be appreciated.** I also use NimBLE stack as BLE stack.

When I was looking to the mynewt documents I saw there is BLE_HCI_SCAN_FILT_MAX for filtering data, but when I tried to use it, my central device didn't capture any of my advertisements. When I looked into the problem I saw two macro's sharing the same value conflicting each other. Picture below in the file
`"C:\Espressif\frameworks\esp-idf-v5.3.2\components\bt\host\nimble\nimble\nimble\include\nimble\hci_common.h" in lines between 1446-1447.`

Also it is not defined in
`"C:\Espressif\frameworks\esp-idf-v5.3.2\components\bt\host\nimble\nimble\apps\btshell\src\cmd.c" (image 2) in lines 1176-1182.`

However this part says it is a valid policy in
`"C:\Espressif\frameworks\esp-idf-v5.3.2\components\bt\host\nimble\nimble\nimble\host\include\host\ble_gap.h" in lines between 2245- 2299 in the function "ble_gap_ext_disc"` (No image provided but you can look at it).

I assume that you didn't write that macro in the esp-idf.  What should I do?

[image]
                                                              image - 1

[image]
                                                              image - 2

rahult-github, 2025-01-29:
Hi @UmutUnay , 

I think you need to enable scan filter policy while enabling scanning.  Set the filter policy while calling your ext scan API. You can refer information of it in BT spec version 6.0 , vol 4, Part E, 7.8.65

[image]

And more specifically: 

[image]

UmutUnay, 2025-01-30:
The thing is I already use it. I call `ble_gap_ext_disc` function with a filter policy (currently BLE_HCI_SCAN_FILT_NO_WL) with filter_duplicates as 1. This should prevent the duplicates to be appear, am I wrong? Also to mention that when I make CONFIG_BT_CTRL_SCAN_DUPL_CACHE_SIZE from 100 to 1000, nothing changed so I don't think this is a cache issue.

And about the main issue, is BLE_HCI_SCAN_FILT_MAX definition missing? If not where?

rahult-github, 2025-01-30:
Hello @UmutUnay ,  

> The thing is I already use it. I call ble_gap_ext_disc function with a filter policy (currently BLE_HCI_SCAN_FILT_NO_WL) with filter_duplicates as 1. This should prevent the duplicates to be appear, am I wrong? 

Ok, may i request you to please share your application code that is not working ? 

> And about the main issue, is BLE_HCI_SCAN_FILT_MAX definition missing? If not where? 

This is not an issue. This is standard C coding mechanism where the highest value is used as max and checks are made against this value to detect if value passed is valid or not. something like below: 

    /* Check scanner filter policy */
    if (disc_params->filter_policy > BLE_HCI_SCAN_FILT_MAX) {
        return BLE_HS_EINVAL;
    }

UmutUnay, 2025-01-31:
Hello @rahult-github ,
Thank you for your messages before, and I am sorry for my late message.

> Ok, may i request you to please share your application code that is not working ?

It is a simple function call with a `filter_policy `set to BLE_HCI_SCAN_FILT_NO_WL, and `filter_duplicates `set to 1.

> This is not an issue. This is standard C coding mechanism where the highest value is used as max and checks are made against this value to detect if value passed is valid or not.

After I wrote the message I then see that part and understood it, but I didn't want to edit my message.

I have tested it more and find out when I disabled the `filter_duplicates`, it gave me too much of the same advertisement, hence I am sure that it works to a certain degree. Then I tested it with more cache size 100 to 1000, but in five attempts, in average it is quite similar. Then I tested in cache sizes 10 and 1, similar in average. After these tests, I changed it to 100 then changed the peripheral device's `ble_gap_ext_adv_params`'s intervals to make it advertise less, but this makes my central device less consistent about catching the advertisements. Should I just eliminate duplicates on application level?

closed 2025-02-19 by UmutUnay