espressif/esp-idf #15304: [USB][v5.1.4] Yamaha P-525 returns `HUB: Bad transfer status 1: CHECK_SHORT_DEV_DESC`

chipweinberger, 2025-01-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

esp32-s3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

USB communication should succeed.

### What is the actual behavior?

```
    (103239) HUB: Bad transfer status 1: CHECK_SHORT_DEV_DESC
    (103243) HUB: Stage failed: CHECK_SHORT_DEV_DESC
```

### Steps to reproduce.

1. plug in Yamaha P-525

### Debug Logs.

```plain

```

### More Information.

This is all the relevant information I have. This is a customer report.

chipweinberger, 2025-01-29:
@tore-espressif 
@peter-marcisovsky 
@roma-jam 

Thanks for your time.

roma-jam, 2025-01-29:
Hi @chipweinberger, 

Thanks for reporting the problem. 

Could you try to increase the root port delays in menuconfig?
`Component config` -> `USB-OTG `-> `Root Hub configuration`
 
Step 1. Increase only `Reset recovery delay` to 3000 ms. If that helps, you can decrease the value and find out which one works. If not:
Step 2. Increase `Reset hold` to 1000 ms. If not helps: 
Step 3. Increase `SetAddress() recovery time` to 1000 ms. 

Let me know if that helps.

chipweinberger, 2025-01-29:
Thanks for the suggestions.

Question - My product needs to work with hundreds if not thousands of different piano models.

Could changing these hurt compatibility with other models?

I need one set of values that works with all digital pianos.

Thanks for your help.

chipweinberger, 2025-01-29:
If one set of values cannot work, then i need a way for these to be configurable at runtime by the end user .

chipweinberger, 2025-01-29:
I'll also need a way to change these at runtime, so that the same firmware can support all models.

roma-jam, 2025-01-29:
Hi @chipweinberger, 

Yes, that's right indeed. 
This will increase the overall time from plugging the device to the port and getting the enumeration done. It won't affect any devices, which works with less delay. 

Right now there is a tendency to move from build-time configuration to the run-time and as a part of this process the delay values as a configurable parameter also could be eliminated. Unfortunately there is no information about when this could be implemented.

roma-jam, 2025-02-04:
Hi @chipweinberger,

Does the delay changing help?

tore-espressif, 2025-02-04:
@chipweinberger I reproduced the error on a different digital piano.

Unfortunately, it seems to be device problem. My windows PC cannot enumerate it too, this is a screenshot from USB Device Tree Viewer after the problem occurs

[image]

I had to turn the piano off and back on to fix the problem.

Could you please verify that it is not the USB device's problem? You can try like this
1. Plug and unplug the USB device to ESP32 repeatedly until the problem occurs
2. Do NOT turn off the piano and plug it into PC

If the PC cannot enumerate the device, it is the USB device (piano) problem

chipweinberger, 2025-02-04:
> Hi @chipweinberger,
> 
> Does the delay changing help? 

I don't have this piano personally, so it's more difficult. 

But in the next couple weeks I plan to test it.

chipweinberger, 2025-02-04:
@tore-espressif I had the customer plug and unplug many times. Their piano was able to connect to a Mac, but not ESP32-S3.

chipweinberger, 2025-02-04:
IMO Espressif should consider acquiring this piano for further testing.

chipweinberger, 2025-08-25:
I've never gotten a chance to test it sadly

chipweinberger, 2026-09-15:
related: https://github.com/espressif/esp-idf/issues/15938