espressif/esp-idf #15306: Dedicated GPIO not work with inline assembly function on ESP32S3

windowsair, 2025-01-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

I'm trying to use Dedicated GPIO and I've found that inline assembly functions produce different results than non-inline assembly functions:

```c
void setHigh()
{
    __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
}

// inline version
static inline void setHigh()
{
    __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
}
```

Use objdump to see the results of the non-inline and inline versions separately:
```asm
void setHigh()
{
4037ecc8:	004136        	entry	a1, 32
    __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
4037eccb:	754ff4        	lsi	f15, a15, 0x1d4
}
4037ecce:	f01d      	retw.n

// inline version
   __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
42009220:	754ff4        	lsi	f15, a15, 0x1d4
```

According to TRM, the instruction `754ff4` generated by both function is consistent and correct.
[image]

**But now only the non-inline version works.**

I inserted a large number of `nop` instructions into the inline version, considering that there may be some delay in function calls. But that didn't work:

```c
// inline version with nop. NOT WORK
static inline void setHigh()
{
    __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
    __asm__ __volatile__ ("nop" );
    __asm__ __volatile__ ("nop" );
// more than 10 nop here
}
```

Also, calling the `cpu_ll_write_dedic_gpio_mask` function in esp-idf has the same behavior, and once the `inline` attribute of `cpu_ll_write_dedic_gpio_mask` is removed, everything works fine.

### What is the actual behavior?

See above.

### Steps to reproduce.

See above.

### Debug Logs.

```plain

```

suda-morris, 2025-02-03:
> But now only the non-inline version works.

Do you mean, you can't see the GPIO being "toggle/set/clear" by the CPU instruction?  Could you please share a minimal project for us to reproduce? Because we also want to see if you have setup the GPIO correctly.

windowsair, 2025-02-03:
> > But now only the non-inline version works.
> 
> Do you mean, you can't see the GPIO being "toggle/set/clear" by the CPU instruction? Could you please share a minimal project for us to reproduce? Because we also want to see if you have setup the GPIO correctly.

It looks like the time it takes to run inline and non-inline instructions is different, which messes up the timing a bit. I need to insert more nop instructions than expected to make the non-inline version work.

The TRM and esp-idf documentation doesn't say what is the typical time to execute this instruction, i.e. what is the fastest rate of the Dedicated GPIO. I'll verify this with a logic analyzer after the holidays.

suda-morris, 2025-02-05:
@windowsair when you using the dedicate GPIO instructions to simulate any waveforms, please wrap them in a critical section. for example: https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/examples/peripherals/dedicated_gpio/soft_uart/components/soft_uart/soft_uart.c#L154-L156

windowsair, 2025-02-05:
Hi @suda-morris, Thank you for your fast reply! The problem is that the inline and non-inline versions have different execution times. The inline version does not have the extra time overhead of function calls, which causes the output to be too fast, resulting in timing errors.

For those who want to use Dedicated GPIO, here are my real world measurements:

>  The sampling rate may not be enough.

ESP32S3 CPU run as 240MHz:

### 1. No nops
```c
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
...
```

[image]

**About 133MHz**

### 2. 1 nops

```c
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );
 __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );
 __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );
...
```

[image]

**About 57MHz**

### 3 . 2 nops

```c
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );__asm__ __volatile__ ("nop" );
 __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );__asm__ __volatile__ ("nop" );
__asm__ __volatile__ ("ee.clr_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );__asm__ __volatile__ ("nop" );
 __asm__ __volatile__ ("ee.set_bit_gpio_out 0xff" );
__asm__ __volatile__ ("nop" );__asm__ __volatile__ ("nop" );
```

[image]

**About 40MHz**

It can be seen that the GPIO can be flipped once per CPU cycle.

Close this. Thank you!

closed 2025-02-05 by windowsair