espressif/esp-idf #15312: Ethernet Example DHCP Isuue

diplfranzhoepfinger, 2025-01-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic

ESP-IDF v5.4 

Issue: 

using 2times  W5500

https://gist.github.com/diplfranzhoepfinger/6fc3e9050228a7696de2a1af00b4d7e6

ETH 1 all OK, 
ETH 2 only show UP and DOWN, but no DHCP. 

tested on 2 totally different Hardware: 

- 1 Self made Board,
- 1 M5 Stack with 2 W5500 Modules

so i guess it is a SW Problem. 

Thanks, 
Franz

kostaond, 2025-02-03:
@diplfranzhoepfinger thanks for reporting the issue. There is really a problem, https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c#L109. 
```c
eth_netif_glues[i] = esp_eth_new_netif_glue(eth_handles[0]);
```
should be
```c
eth_netif_glues[i] = esp_eth_new_netif_glue(eth_handles[i]);
```

Alvin1Zhang, 2025-02-25:
Thanks for reporting and sorry for the slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/39f34a65a9c3b94480eb89b77737c13dbe124531, and fix on release/5.4 is available at https://github.com/espressif/esp-idf/commit/462ed8353dddca6247a045a8af282de4502c832e (it will be included in the v5.4.1, and regarding the release timeframe of v5.4.1, please help refer to https://github.com/espressif/esp-idf/blob/master/ROADMAP.md#esp-idf-bugfix-releases), thanks.

closed 2025-02-25 by Alvin1Zhang