espressif/esp-idf #15314: w5500.mac: w5500_spi_write(140): spi transmit failed

diplfranzhoepfinger, 2025-01-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

W5500
plus 
ESP32-S3-DevKitC-1-N32R8V

I sometimes get this: 

it comes and goes ... sometimes it runs fine a day long, 
then it comes every try ... 

still could be a HW Issue, like Cabling. 

```
E (170722) w5500.mac: w5500_spi_write(140): spi transmit failed
E (170722) w5500.mac: w5500_write_buffer(258): write TX buffer failed
E (170723) w5500.mac: emac_w5500_transmit(583): write frame failed
```

this is my Pinout: 

```
#
# Example Ethernet Configuration
#
CONFIG_ENV_GPIO_RANGE_MIN=0
CONFIG_ENV_GPIO_RANGE_MAX=48
CONFIG_ENV_GPIO_IN_RANGE_MAX=48
CONFIG_ENV_GPIO_OUT_RANGE_MAX=48
CONFIG_EXAMPLE_USE_SPI_ETHERNET=y
CONFIG_EXAMPLE_SPI_ETHERNETS_NUM=1
# CONFIG_EXAMPLE_USE_DM9051 is not set
# CONFIG_EXAMPLE_USE_KSZ8851SNL is not set
CONFIG_EXAMPLE_USE_W5500=y
CONFIG_EXAMPLE_ETH_SPI_HOST=1
CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO=10
CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO=11
CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO=14
CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ=16
CONFIG_EXAMPLE_ETH_SPI_CS0_GPIO=46
CONFIG_EXAMPLE_ETH_SPI_INT0_GPIO=48
CONFIG_EXAMPLE_ETH_SPI_POLLING0_MS=0
CONFIG_EXAMPLE_ETH_SPI_PHY_RST0_GPIO=39
CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR0=1
# end of Example Ethernet Configuration
```

nopnop2002, 2025-02-01:
Looking at the source code, I don't think there are any problems with the source.

https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/spi/w5500/esp_eth_mac_w5500.c#L141

diplfranzhoepfinger, 2025-02-02:
I think it was a Memory Problem.
Moving some other Parts to SPIRAM, the Error is gone. 

might be good to have a better Error Diagnosis here.

nopnop2002, 2025-02-02:
You can check for memory leaks with the code below.
```
#include <stdio.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

void traceHeap() {
    static uint32_t _free_heap_size = 0;
    if (_free_heap_size == 0) _free_heap_size = esp_get_free_heap_size();

    int _diff_free_heap_size = _free_heap_size - esp_get_free_heap_size();
    ESP_LOGI(__FUNCTION__, "_diff_free_heap_size=%d", _diff_free_heap_size);
    ESP_LOGI(__FUNCTION__, "esp_get_free_heap_size() : %6"PRIu32"\n", esp_get_free_heap_size() );
    ESP_LOGI(__FUNCTION__, "esp_get_minimum_free_heap_size() : %6"PRIu32"\n", esp_get_minimum_free_heap_size() );
    ESP_LOGI(__FUNCTION__, "xPortGetFreeHeapSize() : %6zd\n", xPortGetFreeHeapSize() );
    ESP_LOGI(__FUNCTION__, "xPortGetMinimumEverFreeHeapSize() : %6zd\n", xPortGetMinimumEverFreeHeapSize() );
    ESP_LOGI(__FUNCTION__, "heap_caps_get_free_size(MALLOC_CAP_32BIT) : %6d\n", heap_caps_get_free_size(MALLOC_CAP_32BIT) );
    // that is the amount of stack that remained unused when the task stack was at its greatest (deepest) value.
    ESP_LOGI(__FUNCTION__, "uxTaskGetStackHighWaterMark() : %6d\n", uxTaskGetStackHighWaterMark(NULL));
}

void app_main()
{
    while(1) {
        traceHeap();
        vTaskDelay(100);
    }
}
```

diplfranzhoepfinger, 2025-02-02:
good idea....

kostaond, 2025-03-28:
> I think it was a Memory Problem. Moving some other Parts to SPIRAM, the Error is gone.
> 
> might be good to have a better Error Diagnosis here.

@diplfranzhoepfinger would you have any suggestion, or something which helped you when your were dealing with the issue, how to improve the error report?