espressif/esp-idf #15315: How to get the transition time of ZigBee level control clusters move actions

pvginkel, 2025-01-31:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to control the transition of ZigBee level control clusters myself.

This is currently implemented by the ZigBee stack. I'm seeing log messages like this, a few in sequence as the level changes:

```
ZB_TRACE_LOG[0]: zcl/zcl_continuous_value_change_commands.c:203    steps_number = 5
```

The level of the ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID attribute is increased in steps to reach the target level. The steps however are quite big so the transition are very noticeable. I would like to control this myself. I would like to just get one notification that the level has changed and the associated transition time, instead of multiple attribute changes of the ESP_ZB_ZCL_ATTR_LEVEL_CONTROL_CURRENT_LEVEL_ID attribute.

How can this be achieved?

xieqinan, 2025-02-05:
Hi @pvginkel ,

> How can this be achieved?

The transition process of ZCL Level Control is handled by the Zigbee stack by default. There is no dedicated API to retrieve the transition time of a Level Control action.

> . The steps however are quite big so the transition are very noticeable.

Could you please share the log of the level change? I believe the level should transition smoothly from the original value to the target value. There might be something missing.

pvginkel, 2025-02-09:
I'm not sure what logs you'd like. I've enabled ZCL logging like this:

```c
esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_DEBUG, ESP_ZB_TRACE_SUBSYSTEM_ZCL);
```

Does that make sense? I'm sending command 4 on cluster 8 with the following parameters:

```json
{
  "level": 254,
  "transtime": 30
}
```

It just makes two steps if I have debugging enabled. If I don't have logging enabled, it moves to full brightness in roughly 20 steps.

The log looks like this:

```
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:702    >> zb_zcl_reporting_timer_handler
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:724    min_interval expired rep_info 0xa
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:751    Zero max_interval time, skip
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:773    0 entries to report
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:640    >> zb_zcl_adjust_reporting_timer
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:671    reschedule 0, next_event_time 0
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:690    << zb_zcl_adjust_reporting_timer
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:782    << zb_zcl_reporting_timer_handler
ZB_TRACE_LOG[3]: zcl/zcl_common.c:592    >>get_endpoint_desc 0x408125e4 0x4081eaf0 ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:597    device_ctx 0x4081eaf0, ep_count 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:599    ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:609    <<get_endpoint_desc desc 0x4081eb10
ZB_TRACE_LOG[2]: zcl/zcl_common.c:1060    > zb_zcl_is_target_endpoint: ep_desc 0x4081eb10, profile_id 0x104
ZB_TRACE_LOG[2]: zcl/zcl_common.c:1120    < zb_zcl_is_target_endpoint, match 1
ZB_TRACE_LOG[1]: zcl/zcl_common.c:1638    > zb_zcl_process_device_command 14
ZB_TRACE_LOG[1]: zcl/zcl_main.c:192    >> zb_zcl_parse_header: param 14 h 0x0 sz 6, dst endp 10
ZB_TRACE_LOG[1]: zcl/zcl_main.c:264    << zb_zcl_parse_header ret = 0
ZB_TRACE_LOG[1]: zcl/zcl_common.c:1572    > zb_zcl_process_parsed_zcl_cmd 14
ZB_TRACE_LOG[2]: zcl/zcl_common.c:1328    > ep_process_zcl_cmd, param 14
ZB_TRACE_LOG[2]: zcl/zcl_common.c:1333    cmd_id: 4, cluster_id: 0x8, is_common_command: 0
ZB_TRACE_LOG[3]: zcl/zcl_common.c:592    >>get_endpoint_desc 0x408125e4 0x4081eaf0 ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:597    device_ctx 0x4081eaf0, ep_count 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:599    ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:609    <<get_endpoint_desc desc 0x4081eb10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:592    >>get_endpoint_desc 0x408125e4 0x4081eaf0 ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:597    device_ctx 0x4081eaf0, ep_count 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:599    ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:609    <<get_endpoint_desc desc 0x4081eb10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:1681    >>get_cluster_desc ep_desc 0x4081eb10, cluster_id 0x0 cluster_role 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:1696    <<get_cluster_desc cluster_desc 0x4081eb38
ZB_TRACE_LOG[2]: zcl/zcl_general_commands.c:604    >> zb_zcl_get_attr_desc_manuf_a ep 10, cluster_id 0x0, attr_id 0x12, manuf_code 0xffff
ZB_TRACE_LOG[3]: zcl/zcl_common.c:592    >>get_endpoint_desc 0x408125e4 0x4081eaf0 ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:597    device_ctx 0x4081eaf0, ep_count 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:599    ep_id 10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:609    <<get_endpoint_desc desc 0x4081eb10
ZB_TRACE_LOG[3]: zcl/zcl_common.c:1681    >>get_cluster_desc ep_desc 0x4081eb10, cluster_id 0x0 cluster_role 1
ZB_TRACE_LOG[3]: zcl/zcl_common.c:1696    <<get_cluster_desc cluster_desc 0x4081eb38
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:560    >> zb_zcl_get_attr_desc_manuf cluster_desc 0x4081eb38, attr_id 0x12, manuf_code 0xffff, cluster id: 0x0
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0xfffd: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0x0: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0x7: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0x4: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0x5: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:571    Checking attribute with id 0xffff: is_attr_ids_match 0, is_manuf_codes_match 1, is_ms_bit_set 0, is_ms_bit_set_correctly 1
ZB_TRACE_LOG[3]: zcl/zcl_general_commands.c:588    << zb_zcl_get_attr_desc_manuf, attr_desc 0x0
ZB_TRACE_LOG[2]: zcl/zcl_general_commands.c:615    << zb_zcl_get_attr_desc_manuf_a ret 0x0
ZB_TRACE_LOG[1]: zcl/zcl_basic_commands.c:120    > zb_zcl_check_is_device_enabled ep_id 10, cmd_id 4, cluster_id 8, is_common_command 0
ZB_TRACE_LOG[1]: zcl/zcl_basic_commands.c:144    < zb_zcl_check_is_device_enabled ret 1
ZB_TRACE_LOG[2]: zcl/zcl_common.c:1455    ep 10, dev handler 0x4200ead8
--- 0x4200ead8: ZigBeeDevice::zbTask()::{lambda(unsigned char)#1}::_FUN(unsigned char) at C:/Projects/esp-libs/esp-zha-support/src/ZigBeeDevice.cpp:132
[... 1126 lines cut ...]
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:640    >> zb_zcl_adjust_reporting_timer
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:671    reschedule 1, next_event_time 61721637
[... cut ...]
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:724    min_interval expired rep_info 0xa
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:751    Zero max_interval time, skip
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:773    0 entries to report
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:640    >> zb_zcl_adjust_reporting_timer
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:671    reschedule 0, next_event_time 0
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:690    << zb_zcl_adjust_reporting_timer
ZB_TRACE_LOG[1]: zcl/zcl_reporting.c:782    << zb_zcl_reporting_timer_handler
```