espressif/esp-idf #15316: W5500 and MAC Address

diplfranzhoepfinger, 2025-02-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://wiznet.io/products/network-modules/wiz550io

this Wiznet Module does print a MAC Address on its Back. 

it seems the Ethernet Sample of ESP-IDF just ignore this MAC ? 

right ? 

thanks, 
Franz

nopnop2002, 2025-02-03:
You can get the MAC address from here.
I don't have time to try it out.

[image]

kostaond, 2025-03-28:
MAC address of W5500 is zeroed on boot as pointed out by @nopnop2002. Therefore, we set it's MAC address by software means during Ethernet driver initialization in our examples: https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/components/ethernet_init/ethernet_init.c#L275 However, you can set whatever address you want using `esp_eth_ioctl`, see example at https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/components/ethernet_init/ethernet_init.c#L223

closed 2025-03-28 by kostaond