espressif/esp-idf #15318: is there any idf command option to generate cscope.files or compile_commands.json file ?

iysheng, 2025-02-02:
### Is your feature request related to a problem?

I always use vim to develop with idf. I found it's hard to jump when view these code. So I wonder is there any command option to genrate cscope.files or compile_commands.json file to achive jump with cscope tool or clangd ?

dobairoland, 2025-02-03:
Hi @iysheng. If you run `idf.py reconfigure` or `idf.py build` then `build/compile_commands.json` should be available in your project. It looks like the Vscode extension also uses it (https://developer.espressif.com/blog/vscode-esp-idf-v1.8.0/#code-navigation-without-full-project-build). Please let us know if this works for you.

I don't think we have support for generating `cscope.files` right now.

igrr, 2025-02-03:
And if you need tips on configuring clangd, please check https://github.com/espressif/esp-idf/issues/6721 for recipes.

iysheng, 2025-02-05:
> And if you need tips on configuring clangd, please check [#6721](https://github.com/espressif/esp-idf/issues/6721) for recipes.

tks, I viewed this issue https://github.com/espressif/esp-idf/issues/6721 .
First I do command ``ln -s build/compile_commands.json .`` to link  ``build/compile_commands.json`` file in project root dirctory.
Then I create ``.clangd`` file in project root file which has build directory. and add theses content in the ``.clangd`` file, now it's ok.
```
CompileFlags:
  # adjust for your compile_commands.json location
  CompilationDatabase: build/
  Remove:
    [
      -fno-tree-switch-conversion,
      -mtext-section-literals,
      -fstrict-volatile-bitfields,
      -mlong-calls,
      -fno-shrink-wrap,
      -march=rv32imc_zicsr_zifencei,
    ]
```

Besides, I wonder that though I have set ``CompilationDatabase: build/`` in ``.clangd`` file  , But if there is no compile_commands.json file in project root dirctory, I still couldn't do function jump.

iysheng, 2025-02-05:
I found another issue, when I want to jump ``esp_board_init`` function defination. I just jump to the function def in the header file ``hardware_driver/include/esp_board_init.h``. How could this happen ?