espressif/esp-idf #15319: MTU in esp-idf

indexds, 2025-02-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm looking to make my ethernet netif's mtu match that of wireguard at 1420 to prevent fragmentation, but the default is 1500 and that doesn't seem changeable at all.
Since the rust wrappers in `esp-idf-svc` are using the esp-netif api to create interfaces, the raw lwip interface is abstracted and we can't access the mtu. And even if we could (it's most likely possible through shenanigans) I'm pretty sure changing the mtu at runtime is gonna make everything explode.

Soooo... How do we create a netif with a custom mtu? Is it even possible?

From what I could see the mtu is hardcoded here,

```c
/**
 * In this function, the hardware should be initialized.
 * Invoked by ethernetif_init().
 * @param netif lwip network interface which has already been initialized
 */
static void ethernet_low_level_init(struct netif *netif)
{
    netif->mtu = 1500;
    ...
}
```
And this function is called by 
```c
/**
 * @brief   LWIP's network stack init function for Ethernet
 * @param netif LWIP's network interface handle
 * @return ERR_OK on success
 */
err_t ethernetif_init(struct netif *netif);
```
Which *should* be the init function passed when initializing an ethernet netif.

Is the mtu parameter just not exposed at all? In which case, what would be required to expose it in the api?

david-cermak, 2025-02-05:
Thank you for suggesting this. It is in fact *possible* to update the MTU currently, but not in a nice way.

You can either create your own lwip netif_init() function and pass it to `esp_netif` configs, or update the lwip netif parameters directly (but inside the TCP/IP context).
Neither of these is very convenient (even less when you're using rust wrappers). I will make an API for that.

Is it okay to just add this option to the config parameters of `esp_netif_new()` or do you think an actual setter should be created, like `esp_netif_set_mtu()`? (cannot think of many usecases for runtime configuration, but maybe missing something).

indexds, 2025-02-05:
Hi, yes having it in `esp_netif_new()` would be the best case scenario for me as it would be really easy to update the wrappers downstream. I don't think a setter has any use, since if you need to change the mtu you might as well destroy and recreate the netif. I can't think of any use case for this.

Thanks a lot!