espressif/esp-idf #15322: btm_pm_proc_mode_change causes disconnect

mitchellcairns, 2025-02-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32-WROOM

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When the sniff/unsniff commands are sent (Host device is a Nintendo Switch here), the HID/ACL connections are severed. 

### What is the actual behavior?

Simply disabling the btm_pm_proc_mode_change function by performing an early return causes this issue to go away.

I'm not sure if there's a custom HCI command that is being misinterpreted as sniff/unsniff, but these are common commands and it seems like something is not right here with how the commands are being handled. 

[image]

### Steps to reproduce.

I'm not sure if there will be a simple way to reproduce this. I'm open to ideas, if there's another setup where we can test the sniff/unsniff commands in the context of a Bluetooth HID device. 

### Debug Logs.

```plain

```

BetterJincheng, 2025-02-08:
@mitchellcairns 
From the log you provided, no abnormal phenomena were found on the ESP32 side.
It looks like ESP32 always wants to enter the sniff, while peer devices always want to exit the sniff. Finally, the disconnection is initiated by the peer device. (`BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13` )

mitchellcairns, 2025-02-08:
> [@mitchellcairns](https://github.com/mitchellcairns) From the log you provided, no abnormal phenomena were found on the ESP32 side. It looks like ESP32 always wants to enter the sniff, while peer [...]

I believe that the ESP32 is not respecting the hcif mode change properly or reporting it back. 

When I override the behavior and adjust my polling rate manually according to the requested interval, the connection remains stable and works as expected. I do think something deeper is broken here.

The setup is an HID gamepad. I'm not doubting that the host is terminating the connection, but I think the ESP32 is not adjusting its reporting interval as it should.

BetterJincheng, 2025-02-11:
@mitchellcairns Is HID the only service connected between esp32 and the gamepad?

mitchellcairns, 2025-02-11:
> [@mitchellcairns](https://github.com/mitchellcairns) Is HID the only service connected between esp32 and the gamepad?

Yes.

jcbetter, 2026-10-14:
@mitchellcairns We have found some issues in sniff mode. Please test with the following changes.

```
-            {{BTA_DM_PM_SNIFF_HD_ACTIVE_IDX, 0},   {BTA_DM_PM_NO_ACTION, 0}},    /* busy */
+            {{BTA_DM_PM_SNIFF_HD_ACTIVE_IDX, 5000 + BTA_DM_PM_SPEC_TO_OFFSET},   {BTA_DM_PM_NO_ACTION, 0}},    /* busy */
```