espressif/esp-idf #15326: Issue with ESP32 NVS Encryption

KaeLL, 2025-02-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm experimenting with ESP32's security features, and I'm having an issue getting ESP32 autonomous NVS encryption workflow to work.

Here's an example repo with a script describing the build and flashing commands used. Secure Boot and Flash Encryption keys are the only artifacts not provided for obvious reasons.
https://github.com/KaeLL/nvs_enc

For the record, the DISABLE_DL_ENCRYPT efuse is burned.

<details>
<summary>Log of app run</summary>

```yaml
I (37) boot: ESP-IDF v5.5-dev-1656-g1d462a1298-dirty 2nd stage bootloader
I (38) boot: compile time Feb  3 2025 16:31:03
I (38) boot: Multicore bootloader
I (38) boot: chip revision: v1.0
I (39) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (42) boot: ## Label            Usage          Type ST Offset   Length
I (43) boot:  0 factory          factory app      00 00 00010000 00100000
I (44) boot:  1 nvs_keys         NVS keys         01 04 00110000 00001000
I (45) boot:  2 nvs              WiFi data        01 02 00111000 00004000
I (46) boot: End of partition table
I (47) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0bd10h ( 48400) map
I (61) esp_image: segment 1: paddr=0001bd38 vaddr=3ff80000 size=0001ch (    28) load
I (61) esp_image: segment 2: paddr=0001bd5c vaddr=3ffb0000 size=024cch (  9420) load
I (64) esp_image: segment 3: paddr=0001e230 vaddr=40080000 size=01de8h (  7656) load
I (67) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=19784h (104324) map
I (95) esp_image: segment 5: paddr=000397ac vaddr=40081de8 size=0b7b8h ( 47032) load
I (110) esp_image: segment 6: paddr=00044f6c vaddr=00000000 size=0b014h ( 45076) 
I (123) esp_image: Verifying image signature...
I (451) boot: Loaded app from partition at offset 0x10000
I (451) secure_boot_v1: bootloader secure boot is already enabled. No need to generate digest. continuing..
I (452) boot: Checking flash encryption...
I (452) flash_encrypt: flash encryption is enabled (0 plaintext flashes left)
I (453) boot: Checking secure boot...
I (454) secure_boot_v1: bootloader secure boot is already enabled, continuing..
I (455) boot: Disabling RNG early entropy source...
I (457) cpu_start: Multicore app
D (458) cpu_start: Pro cpu up
D (458) cpu_start: Starting app cpu, entry point is 0x40081054
--- 0x40081054: call_start_cpu1 at /esp-idf/components/esp_system/port/cpu_start.c:190

D (0) cpu_start: App cpu up
D (465) clk: RTC_SLOW_CLK calibration value: 3418957
D (466) cpu_start: calling init function: 0x400d1294 on core: 0
--- 0x400d1294: __esp_system_init_fn_init_efuse_check at /esp-idf/components/efuse/src/esp_efuse_startup.c:37

D (466) cpu_start: calling init function: 0x400d2370 on core: 0
--- 0x400d2370: __esp_system_init_fn_init_show_cpu_freq at /esp-idf/components/esp_system/startup_funcs.c:64

I (466) cpu_start: Pro cpu start user code
I (467) cpu_start: cpu freq: 240000000 Hz
D (467) cpu_start: calling init function: 0x400d120c on core: 0
--- 0x400d120c: __esp_system_init_fn_init_show_app_info at /esp-idf/components/esp_app_format/esp_app_desc.c:123

I (468) app_init: Application information:
I (468) app_init: Project name:     nvs_enc
I (469) app_init: App version:      e0f7c50-dirty
[... 162 lines cut ...]
ESP_ERROR_CHECK failed: esp_err_t 0x1102 (ESP_ERR_NVS_NOT_FOUND) at 0x400d7ae8
--- 0x400d7ae8: app_main at /nvs_enc/main/nvs_enc.c:24 (discriminator 1)

file: "./main/nvs_enc.c" line 24
func: app_main
expression: nvs_get_str( nvs_h, "test_key", out, &out_sz )

abort() was called at PC 0x40086003 on core 0
--- 0x40086003: _esp_error_check_failed at /esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40081801:0x3ffb3f30 0x4008600d:0x3ffb3f50 0x4008bd51:0x3ffb3f70 0x40086003:0x3ffb3fe0 0x400d7ae8:0x3ffb4010 0x400e9159:0x3ffb4050 0x40086801:0x3ffb4080
--- 0x40081801: panic_abort at /esp-idf/components/esp_system/panic.c:454
0x4008600d: esp_system_abort at /esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4008bd51: abort at /esp-idf/components/newlib/src/abort.c:38
0x40086003: _esp_error_check_failed at /esp-idf/components/esp_system/esp_err.c:49
0x400d7ae8: app_main at /nvs_enc/main/nvs_enc.c:24 (discriminator 1)
0x400e9159: main_task at /esp-idf/components/freertos/app_startup.c:208
0x40086801: vPortTaskWrapper at /esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 176a08109

CPU halted.
```

</details>

mahavirj, 2025-02-04:
@KaeLL 

I could not find the command to generate and flash the NVS encryption keys partition in build script. Please note that, encryption of pre-flashed NVS contents on the device is not supported (https://github.com/espressif/esp-idf/issues/14848#issuecomment-2467363322).

Please refer to this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/security-features-enablement-workflows.html#enable-nvs-encryption-based-on-flash-encryption), it should help to fix the problem.

KaeLL, 2025-02-04:
Interesting. As said on #14551, it seems to work if the partition is internally generated, which should mean that as far as the NVS encryption component is concerned, there's no difference. So why not?

mahavirj, 2025-02-05:
Yes, it should be possible. However, NVS encryption on device for pre-stored data would add additional complexity and increase the time for the first boot. 

We recommend using host based workflow instead, and tools like [mass manufacturing utility](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/mass_mfg.html).

KaeLL, 2025-02-05:
I don't see how that's any different from opting to use flash encryption with device-generated key, but I'll leave it at that, as this isn't a feature request, and the host-based workflow is enough for me.
Thanks for the pointer on mfg.
The issue may be closed.

closed 2025-02-06 by mahavirj