espressif/esp-idf #15327: BLE Appearance Characteristic only supports an incomplete hard-coded list of icons

qwertymodo, 2025-02-03:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf-release_v5.3-cfea4f7c-v1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

When I call esp_ble_gap_config_local_icon() with a valid Appearance value, I expect that value to be accepted and the Appearance to be updated accordingly.

### What is the actual behavior?

esp_ble_gap_config_local_icon() seems to have a hard-coded list of Appearance values [here](https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/esp_gap_ble_api.c#L304) that it will accept, and any other values are rejected with an ESP_ERR_INVALID_ARG error. Unfortunately, this hard-coded list is both woefully incomplete and also unmaintained, as the git blame seems to indicate only a single value has been added to the list since it was originally created 7 years ago, so many completely valid Appearance values are rejected.

### Steps to reproduce.

Attempt to call esp_ble_gap_config_local_icon with any Appearance value not explicitly defined in the switch case found at https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/esp_gap_ble_api.c#L304

e.g.:
0x03CA (Presentation Remote)
0x04C1 (Switch)
0x0540 (Generic Sensor)
etc.

Check the return value, which will be 0x102 ESP_ERR_INVALID_ARG and the Appearance will remain 0x0000 Unknown when connecting to the device.

### Debug Logs.

```plain

```

### More Information.

I recommend simply removing the switch statement entirely.  It would be one thing if the list was actively maintained and up-to-date with all current valid Appearance definitions, but as that is clearly not the case, the "validation" is not itself valid and is instead rejecting perfectly valid inputs for no reason.

qwertymodo, 2025-02-03:
For reference, here is the Bluetooth SIG list of assigned numbers, the relevant list of Appearance values is in table 2.6.3 starting on page 30: https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Assigned_Numbers/out/en/Assigned_Numbers.pdf

Also, I've found several of the tables in that document to be out-of-date, or otherwise missing entries, so there may in fact be more currently assigned Appearance values than listed there.  I'm not sure if they have other documents elsewhere that are more complete, but I've found this project by Nordic Semiconductor to be a more complete reference: https://github.com/NordicSemiconductor/bluetooth-numbers-database/blob/master/v1/gap_appearance.json

esp-zhp, 2025-02-07:
Hi @qwertymodo,

Thank you for bringing this to our attention. We have considered removing these restrictions, but doing so may introduce other issues.

In your code, you can manually remove these restrictions if necessary.

We will evaluate the possibility of removing these restrictions in future updates, but further assessment is needed to ensure no unintended issues arise.

qwertymodo, 2025-02-26:
It's not possible to remove these restrictions in my code without recompiling the entire esp-idf.  If you don't want to remove the restrictions, you should at the very least update the list of known appearance values, as your existing list currently contains less than 20% of the currently defined appearance characteristics listed in the official Assigned Numbers document that I linked in my second comment.