espressif/esp-idf #15330: sdmmc driver deinitializing too many GPIOs for 1 bit bus (slot_width)

cp-pkasemir, 2025-02-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraT-Mini V1.2

### Power Supply used.

USB

### What is the expected behavior?

After calling `esp_vfs_fat_sdmmc_mount()` with `slot_config.width = 1` the SD card lines are initialized as expected [clk, cmd, d0]

After calling `esp_vfs_fat_sdcard_unmount()` the SD card lines should be deinitialized [clk, cmd, d0]

### What is the actual behavior?

After calling `esp_vfs_fat_sdmmc_mount()` with `slot_config.width = 1` the SD card lines are initialized as expected [clk, cmd, d0]

After calling `esp_vfs_fat_sdcard_unmount()` the too many SD card lines are deinitialized [clk, cmd, d0, **d1, d2, d3**]

### Steps to reproduce.

1. Call `esp_vfs_fat_sdmmc_mount()` with `slot_config.width = 1`
2. Call `esp_vfs_fat_sdcard_unmount()` 

```c
#define SDIO_GPIO_PWR_CTRL 13
#define SDIO_GPIO_PWR_ACTIVE_LOW
#define SDIO_GPIO_CD 34

sdmmc_card_t *sd_card;

void sd_card_power_on(bool on)
{
#ifdef SDIO_GPIO_PWR_CTRL
    int level_active;
#ifdef SDIO_GPIO_PWR_ACTIVE_LOW
    level_active = !on;
#else
    level_active = on;
#endif
    gpio_config_t cfg = {
        .pin_bit_mask = BIT64(SDIO_GPIO_PWR_CTRL),
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = false,
        .pull_down_en = false,
        .intr_type = GPIO_INTR_DISABLE,
    };
    ESP_LOGI(TAG, "Powering the SD card %s", on ? "on" : "off");
    gpio_config(&cfg);
    gpio_set_level(SDIO_GPIO_PWR_CTRL, level_active);
    vTaskDelay(10 / portTICK_PERIOD_MS);
#endif /* SDIO_GPIO_PWR_CTRL */
}

int sd_card_mount(bool format_if_fail)
{
    esp_err_t ret;

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .max_files = 5,
        .allocation_unit_size = 4 * 1024,
        .format_if_mount_failed = format_if_fail,
    };

    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
    slot_config.width = 1;

    sd_card_power_on(true);

    ESP_LOGI(TAG, "Mounting SD card filesystem");
    ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &sd_card);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to mount the SD card (%s)", esp_err_to_name(ret));
        sd_card.card = NULL;
        sd_card_power_on(false);
        return ret;
    }
    ESP_LOGI(TAG, "SD card filesystem mounted");

    return ESP_OK;
}

void sd_card_unmount(void)
{
        esp_err_t ret = esp_vfs_fat_sdcard_unmount("/sdcard", sd_card);
        ESP_LOGI(TAG, "SD card unmounted (%s)", esp_err_to_name(ret));
        sd_card.card = NULL;
        sd_card_power_on(false);
}

```

### Debug Logs.

```plain
esp32> sd_card_mount
I (6248) sd_card: Powering the SD card on
I (6258) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (6268) sd_card: Mounting SD card filesystem
I (6588) sd_card: SD card filesystem mounted
esp32> sd_card_unmount
I (14738) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14738) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14748) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14748) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14758) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14778) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14778) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (14788) sd_card: SD card unmounted (ESP_OK)
I (14788) sd_card: Powering the SD card off
I (14788) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
esp32>

Notice how GPIO 13 gets configured twice (that is because IO13 is mapped to D3, but the board has IO13 used to control the power to the SD card.
```

### More Information.

For lower bus widths, the IO lines for d1, d2, d3 could be used for other functions on a board, and thus deinitializing them incorrectly like what is happening could cause issues for the application.

adokitkat, 2025-02-06:
Ok I see the patch I sent won't help in this situation. I will create a new MR and also send the new patch here so you don't have to wait for repo sync and IDF version release.

cp-pkasemir, 2025-02-06:
The fix needs to happen in `sdmmc_host_deinit_slot_internal()`

adokitkat, 2025-02-06:
Yes I know. Please try this patch: 

[sdmmc_gpio_deinit_fix.patch](https://github.com/user-attachments/files/18693219/sdmmc_gpio_deinit_fix.patch)

cp-pkasemir, 2025-02-06:
The patch you gave works. Thanks.

It is a little prone to another bug though if the `sdmmc_slot_io_info_t` structure changes. I wonder if a compile time assert would be useful to make sure the structure has the right number of elements in it that the deinit code expects.

igrr, 2025-02-17:
Should be fixed with https://github.com/espressif/esp-idf/commit/f8f40ad94fab2bda9d3383b023e3fcc5b1749fd0.

igrr, 2025-02-17:
Thanks for spotting that @cp-pkasemir! Will fix that.