espressif/esp-idf #15333: In esp32, flash encryption was set to development mode, but encryption was performed in release mode.

hgwc, 2025-02-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I was trying to use Flash Encryption.

using esp-idf v4.3.1-3-g954d44c7b9-dirty 2nd stage bootloader
Chip is ESP32-D0WD-V3 (revision 3)

# Security features
CONFIG_SECURE_FLASH_ENC_ENABLED=y
**CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y**
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE is not set

I (985) spi_flash: detected chip: generic
I (989) spi_flash: flash io: qio
**I (993) flash_encrypt: Flash encryption mode is RELEASE**
I (999) cpu_start: Starting scheduler on PRO CPU.
I (1005) nvs: nvs_key partition found
I (1005) nvs: Success to read nvs keys 

I don't know what's wrong.
please help me
I am attaching the sdkconfig and output log files.

[initial_log.txt](https://github.com/user-attachments/files/18667774/initial_log.txt)
file:///home/hgwc/st-device-sdk-c-ref/apps/esp32/light_example/sdkconfig

hgwc, 2025-02-10:
anyone please help

mahavirj, 2025-02-10:
@hgwc 

> using esp-idf v4.3.1-3-g954d44c7b9-dirty 2nd stage bootloader

This version of IDF is EOL. Can you please try with latest IDF version like v5.4 and this example https://github.com/espressif/esp-idf/tree/master/examples/security/flash_encryption?

>  I (993) flash_encrypt: Flash encryption mode is RELEASE

Can you please shared detailed steps that you had followed to enabled flash encryption? It will help if you can share first boot console log and efuse summary here.

hgwc, 2025-02-11:
Hello. 
Development began in June 2023 and ended in May 2024.  We are currently preparing for mass production, but we are having the same problem as the title, so we are asking this question.

1. detailed steps 
1) python ../espefuse.py --port /dev/ttyUSB0 burn_key secure_boot_v1 secure-bootloader-key-256.bin
2) python ../esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-reflash-digest.bin
3) python ../esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin
4) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800  --before default_reset --after no_reset write_flash 0x22000 ~/st-device-sdk-c-ref/iot-core/tools/keygen/keys/nvs_keys.bin
5) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after no_reset write_flash 0x11000 ~/st-device-sdk-c-ref/iot-core/tools/keygen/encrypted_STDK34567890.bin
6) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after no_reset write_flash 0x19000 ~/st-device-sdk-c-ref/bsp/esp32/components/nvs_flash/nvs_partition_generator/test/encrypted_nvs_info.bin
7) esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x10000 partition_table/partition-table.bin 0x1f000 ota_data_initial.bin 0x30000 SLBLDM-1000MOS-SCHOOL.bin

2. partition table
Please refer to the attached file

3. first_booting_log
Please refer to the attached file

4. efuse summary
Please refer to the attached file

[espefuse_summary_log_git.txt](https://github.com/user-attachments/files/18744820/espefuse_summary_log_git.txt)

[first_booting_log_git.txt](https://github.com/user-attachments/files/18744724/first_booting_log_git.txt)

[partitions.2MB.csv](https://github.com/user-attachments/files/18742606/partitions.2MB.csv)

mahavirj, 2025-02-11:
@hgwc 

I suspect that somehow flash-enc release mode config is getting passed to the bootloader build here, please see following code snippet:

https://github.com/espressif/esp-idf/blob/2e74914051d14ec2290fc751a8486fb51d73a31e/components/bootloader_support/src/esp32/flash_encrypt.c#L225-L227

I can see this info log in the first boot log that you shared.

Is your bootloader image pre-built one? Can you please put some `#warning` prints in the above file during build process?

hgwc, 2025-02-11:
Thanks for your answer.
1. Is your bootloader image pre-built one?
    yes
    When receiving UL certification, I placed the bootloader at address 0x0 and changed flash encryption to release mode only in 
     menuconfig.
2. Can you please put some #warning prints in the above file during build process?
    Where and what "#warning" should I put?
3. Do I need to recreate the bootloader based on my current sdkconfig?

mahavirj, 2025-02-11:
> When receiving UL certification, I placed the bootloader at address 0x0 and changed flash encryption to release mode only in menuconfig.

I think your bootloader has been built with release mode enabled config and hence the problem.

>  Where and what "#warning" should I put?

Code snippet I pointed above, just to confirm whether bootloader is getting built with release mode config or not..

> Do I need to recreate the bootloader based on my current sdkconfig?

Yes. For existing devices, bootloader update could be little tricky, only possible through OTA update via firmware (not power safe though).

hgwc, 2025-02-11:
Hello
I re-created the boot-related files based on the current sdkconfig.

"D (1422) memory_layout: Checking 6 reserved memory ranges:[0m"
This error occurred.

The entire log file is attached, so please review it.

[log1.txt](https://github.com/user-attachments/files/18746561/log1.txt)

AdityaHPatwardhan, 2025-02-11:
@hgwc 
In the log that you have shared,

The issue regarding flash encryption seems to be fixed now as Flash Encryption is now enabled in development mode -> (3 plaintext flashes left) instead of (0 plaintext flashes left)
```
[0;32mI (1317) flash_encrypt: flash encryption is enabled (3 plaintext flashes left)[0m
```

Regarding the crash that you are facing, 

``` 
[0;31mE (1921) nvs: nvs_open_from_partition Error (ESP_ERR_NVS_NOT_FOUND) opening NVS handle![0m
D (1931) nvs: nvs_get_str_or_blob onboardingKey[0m
[0;31mE (1941) MAIN: nvs_get_str(ESP_ERR_NVS_INVALID_HANDLE) [0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```
In this case, it seems that there is a read operation on a NULL pointer.
From the error log it seems that the nvs partition was not found and the handle returned is invalid, there seems to be a read operation performed on the invalid NVS handle and hence the crash.
This is not related to flash encryption or the bootloader. 
Please carefully take a look at your main application to fix the issue. 

Please help to close the issue if it is resolved.
Thanks

AdityaHPatwardhan, 2025-02-11:
Hi @hgwc 

> "D (1422) memory_layout: Checking 6 reserved memory ranges:�[0m"

Is not an error but a debug log.
If you set the default log verbosity to info then this debug log won't get printed.
You can set it at `Component Config -> Log output -> Default log verbosity`

Errors are marked as `E` and not `D`.
One error in your case is
```
[0;31mE (1941) MAIN: nvs_get_str(ESP_ERR_NVS_INVALID_HANDLE) �[0m
```

hgwc, 2025-02-12:
Thank you for your kind question and I will close.

closed 2025-02-12 by hgwc