espressif/esp-idf #15335: Question concerning I2S clock generation on ESP32-C6 (and presumably C5)

h-milz, 2025-02-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, for my project I need to justify the use of the internal clock generator (vs. an external xtal oscillator) for a sample rate of 44.1 kHz in TDM256 mode. The BCLK in this case is 11.2896 MHz, which the hardware generates using a fractional divider, with a divider of 14.172336 when derived from 160 MHz or  21.258503 from 240. The C6 Hardware Technical Reference does not say which kind of fractional divider is being implemented (and I assume the C5 which I am currently playing with uses the same building block). If it's a delta-sigma based fractional divider,  the jitter generated by the divider itself can be expected to be in the range of τ = 1.6 ps RMS which is negligible for audio applications, leading to a SNR of -20 log (2 pi f τ) ≈ 160 dB for f ≈ 1 kHz, i.e. way below the existing noise floor of the analog circuitry and the sampling itself. In case of a simple fractional divider switching between the two nearest integers in a more or less fixed fashion, however, the jitter would be in the range of 3 ns (!), and the SNR would be 94.5 dB, i.e. in the range of a high-quality audio ADC, reducing the effective resolution to about 15-16 bits, and thus not acceptable. 

So - which one is it, and would you mind adding this information to the Technical Reference which at the moment has only some wording like "Using fractional divider may introduce some clock jitter." which from an engineering standpoint is lacking? 

And while we're at it, how is the  40 / 48 MHz xtal oscillator in the WROOM modules specified as far as jitter? This jitter will have to be taken into account as well, plus some additional jitter by the PLL generating 480 MHz. 

Thank you! 

References: 
- Texas Instruments Technical Brief SWRA029 "Fractional/Integer-N PLL Basics" 
- Analog Devices Application Note AN-756 "Sampled Systems and the Effects of Clock Phase Noise and Jitter"

h-milz, 2025-02-13:
Update question: at the moment ESP-IDF supports only integer sample rates. In some applications, the absolute sample rate is not important, and then it would be useful if one could enter an integer divider instead. This could be easily achieved by exposing `mclk_div` in the various i2s_*_clk_config_t structs. This variable is uint16_t anyway.

esp-momo, 2025-03-03:
Hi, thank you for your inquiry.

> Which kind of fractional divider is being implemented?

Our fractional divider is not based on delta-sigma modulation, and it’s also not a simple integer switching approach. Instead, we implement alternating between different division ratios, which lies between the two approaches. However, we’re unable to share specific details about the implementation.

> How is the 40 / 48 MHz xtal oscillator in the WROOM modules specified as far as jitter?

The vendor datasheet does not provide this information.

I will keep you updated if more details are available.

closed 2025-11-26 by Alvin1Zhang