espressif/esp-idf #15336: Issue Connecting to Web Server on ESP32C6 ( mbedtls parse returned )

NirajSiroya, 2025-02-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Team,

I am using the **ESP32C6** series with the **ESP-IDF Eclipse plugin**. While connecting to a web server, I am encountering the following error:

E (16447) esp-tls-mbedtls: **mbedtls_x509_crt_parse of public cert returned -0x262E**
E (16447) esp-tls-mbedtls: Failed to set client PKI context
E (16450) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
E (16462) esp-tls: create_ssl_handle failed
E (16467) esp-tls: Failed to open new connection
E (16472) transport_base: Failed to open a new connection

If anyone is familiar with this issue or has encountered something similar, I would appreciate your help in finding a solution.
Thanks.

AdityaHPatwardhan, 2025-02-06:
Hi @NirajSiroya 
If you print the same error using mbedtls error log API i.e. (`mbedtls_strerror`)
Then the output is as follows:
```
Error: X509 - Signature algorithm (oid) is unsupported : OID - OID is not found
```

This is a combination of following two errors
```
#define MBEDTLS_ERR_X509_UNKNOWN_SIG_ALG                  -0x2600
#define MBEDTLS_ERR_OID_NOT_FOUND                         -0x002E
```

This means that the signature algorithm selected for the session establishment is unknown to the client.
You can check following things
1) Check the algorithm used for the server certificate which is used to establish the TLS connection.
2) Enable debug logs and find out which cipher suite is getting selected for the development at runtime ( This specific cipher suite is likely not available)
3) Ensure that this specific cipher suite (identified through step1 or step2) is enabled in the esp-idf mbedtls configuration. ( `idf.py menuconfig -> Component config -> mbedTLS`)

NirajSiroya, 2025-02-06:
Hii _@AdityaHPatwardhan_ , Thank you for the reply.

We are using the **ED25519** algorithm to generate certificates under the crypto algorithm section. However, we are unable to parse these certificates. 

Is there a specific configuration that needs to be enabled in the sdkconfig ?

AdityaHPatwardhan, 2025-02-06:
The `curve25519` is enabled by default, so should be available unless you have disabled it. It is under the option `menuconfig-> Component config -> mbedTLs -> Elliptic Curve Ciphers -> Enable CURVE25519 curve`.
I can tell for sure if you share your `sdkconfig` file.

Can you please also share the text output for your certificate?
You can get it with help of following command
`openssl x509 -in cert.pem -noout -text`
This would give more information regarding the curve, key size, algorithm used etc.
If possible please share the certificate itself.
(Ideally there shouldn't be any sensitive information in the certificate but if you have any, you may generate an identical certificate with sample inputs and share that one).

NirajSiroya, 2025-02-06:
Hii @AdityaHPatwardhan ,

I have attached both my sdkconfig and the certificate output text file. Please review them and let me know your feedback

[certificate output.txt](https://github.com/user-attachments/files/18685717/certificate.output.txt)

[sdkconfig.txt](https://github.com/user-attachments/files/18685706/sdkconfig.txt)

Thanks.

NirajSiroya, 2025-02-10:
Hi @AdityaHPatwardhan  , could you please help with the above topic ?

AdityaHPatwardhan, 2025-02-10:
@NirajSiroya Sorry for the delayed reply.
My bad, I just checked that the issue is because EdDSA itself is not supported in mbedTLS so a certificate generated with ED25519 algorithm won't work with mbedTLS.
See https://github.com/Mbed-TLS/mbedtls/issues/2452 for reference.

Is it possible for you to use alternate algorithms such as ECDSA ? Is there any specific requirement to use the ED25519 algorithm.

NirajSiroya, 2025-02-11:
Hi @AdityaHPatwardhan ,
Thank you for the reply.
Can you please suggest which other high-security algorithms are available in the ESP32-C6?

AdityaHPatwardhan, 2025-02-12:
Hi @NirajSiroya The algorithm would depend on your requirement, but in general you could consider using algorithms with similar bit security level (128 bit) such as ECDSA-256 or RSA-3072.

Since esp-idf uses mbedTLS as its default TLS stack, you can check which ciphersuites are supported in [mbedTLS](https://github.com/Mbed-TLS/mbedtls/blob/v3.6.2/include/mbedtls/ssl_ciphersuites.h) and decide the algorithm accordingly.

All these algorithms are available on esp32-c6.
Furthermore, there is partial/full hardware acceleration available for ECDSA-256, RSA-3072 signature generation.

AdityaHPatwardhan, 2025-02-17:
Closing this issue, since it should be answered with above reply.
@NirajSiroya, Please feel free to re-open the issue if required.
Thanks,
Aditya

closed 2025-02-17 by AdityaHPatwardhan

NirajSiroya, 2025-05-13:
Hello @AdityaHPatwardhan & team , please help for the below problem.

We are using the **ESP32-C6** along with the **ESP-IDF Eclipse plugin**.

Certificate Handling Summary:
We have stored the client certificate and key in SPIFFS memory.
After reading them from SPIFFS, the parsing completes successfully, 

Our controller acts as an HTTP client. For HTTP POST requests, we are including a client certificate. We are also using the same certificate for WebSocket connections, and it works correctly there.

However, when we use the same certificate with an HTTP POST request, we receive a 400 Bad Request response from the server, along with the error message: **"400 No required SSL certificate was sent"** . The server requires the Content-Type header to be set to application/json, and we are already sending that header in our HTTP POST request.

Please find the configuration as below.

 esp_http_client_config_t config_post = {
    		.port = 443,
			.method = HTTP_METHOD_POST,
			.timeout_ms = 6000,
			.skip_cert_common_name_check = true,
			.cert_pem = NULL,
			.client_cert_pem = (char*)client_cert_data,
			.client_key_pem = (char*)client_key_data,
			.transport_type = HTTP_TRANSPORT_OVER_SSL,
			.event_handler = _http_event_handler,
    };
    config_post.url = Cloudurl;

Point to be noted : On the server side, SSL/TLS client certificate verification is mandatory.
                               On the client side, we do not require server certificate verification.

Thanks in advance.