espressif/esp-idf #15339: 关于使用esp32s3n8r16的wss连接字节的扣子平台智能体出现的报错问题

PoohWoah, 2025-02-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

关于连接wss发生的错误

### What is the actual behavior?

我的wss连接特别简单
 url:wss://ws.coze.cn/v1/chat?bot_id=745783240
 Authorization:Bearer pat_aKiztw5bggzq7d1e3qM0Cxz4DyC4kJxXO3mQgRDjkjgQgykghLk18d

`               esp_websocket_client_config_t websocket_cfg = {
                    .uri = "wss://ws.coze.cn:443/v1/chat?bot_id=745783264",
                    .headers = "Authorization: Bearer pat_aKiztw5bggzq7d1e3qM0Cxz4DyC4kJgQgykghLk18d",
                    .task_prio = 5,
                    .task_stack = 4096,
                    .buffer_size = 1024,
                    .user_agent = "ESP32 Websocket Client"};
                client = esp_websocket_client_init(&websocket_cfg);
                esp_websocket_register_events(client, WEBSOCKET_EVENT_ANY, websocket_event_handler, client);

                esp_err_t err = esp_websocket_client_start(client);
                if (err != ESP_OK)
                {
                    ESP_LOGE("WebSocket", "Failed to start WebSocket client: %s", esp_err_to_name(err));
                }`

### Steps to reproduce.

在实际的调试过程中出现了如下报错：
`E (16109) transport_ws: Error read response for Upgrade header GET /v1/chat?bot_id=74578324474 HTTP/1.1
Connection: Upgrade
Host: ws.coze.cn:443
User-Agent: ESP32 Websocket Client
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: BdmZ1hCPWDvr7b2C92aVIQ==
Authorization: Bearer pat_aKiztw5bggzq7d1e3qM0Cx

E (16132) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=0, errno=119
W (16147) WebSocket: WebSocket Error: ERROR
I (16153) websocket_client: Reconnect after 10000 ms
W (16158) WebSocket: WebSocket Disconnected
E (16163) websocket_client: Client cannot be stopped from websocket task
E (16170) websocket_client: Client cannot be stopped from websocket task`

TLS配置如下：
`#
# ESP-TLS
#
CONFIG_ESP_TLS_USING_MBEDTLS=y
# CONFIG_ESP_TLS_USE_DS_PERIPHERAL is not set
# CONFIG_ESP_TLS_CLIENT_SESSION_TICKETS is not set
# CONFIG_ESP_TLS_SERVER_SESSION_TICKETS is not set
# CONFIG_ESP_TLS_SERVER_CERT_SELECT_HOOK is not set
CONFIG_ESP_TLS_SERVER_MIN_AUTH_MODE_OPTIONAL=y
# CONFIG_ESP_TLS_PSK_VERIFICATION is not set
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
# end of ESP-TLS`

请问我这应该如何去寻找问题？用电脑客户端postman可以正常连接到服务器

### Build or installation Logs.

```plain

```

PoohWoah, 2025-02-06:
> ### Answers checklist.
> * [x]  I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.我已阅读文档[ESP-IDF 编程指南](https://docs.espressif.com/projects/esp-idf/en/latest/)，但该问题并未得到解决。[x]  I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.我已将 IDF 分支（主版本或发行版）更新到最新版本，并检查该问题是否存在。[x]  I have searched the issue tracker for a similar issue and not found a similar issue.我在问题跟踪器中搜索了类似的问题，但没有找到类似的问题。
> 
> ### IDF version.
> idf5.3.1
> 
> ### Operating System used.
> Windows  视窗
> 
> ### How did you build your project?
> Command line with Make  命令行
> 
> ### If you are using Windows, please specify command line type.
> None  它不是
> 
> ### What is the expected behavior?
> 关于连接wss发生的错误
> 
> ### What is the actual behavior?
> 我的wss连接特别简单 url:wss://ws.coze.cn/v1/chat?bot_id=745783240 Authorization:Bearer pat_aKiztw5bggzq7d1e3qM0Cxz418d授权：PAT_AKIZTW5BGGZ
> 
> ` esp_websocket_client_config_t websocket_cfg = {`eps_websocket_client_config_t websocket_cfg = { .uri = "wss://ws.coze.cn:443/v1/chat?bot_id=745783264",。 .headers = "Authorization: Bearer pat_aKiztw5bggzq7d1e3qM0hLk18d",。 .task_prio = 5,  .task_prio = 5， .task_stack = 4096,  .task_stack = 4096， .buffer_size = 1024,  .boffer_sisize = 1024， .user_agent = "ESP32 Websocket Client"};.user_agent =“ ESP32 WebSockett客户”}; client = esp_websocket_client_init(&websocket_cfg);客户端= ESP_WEBSOCKECT_CLIENT_INIT（＆webSocket_cfg）; esp_websocket_register_events(client, WEBSOCKET_EVENT_ANY, websocket_event_handler, client);ESP_WEBSOCKECT_REGISTER_EVENTS（客户端，WebSocket_event_any，websocket_event_handler，client）;
> 
> ```
>             esp_err_t err = esp_websocket_client_start(client);
>             if (err != ESP_OK)
>             {
>                 ESP_LOGE("WebSocket", "Failed to start WebSocket client: %s", esp_err_to_name(err));
>             }`
> ```
> 
> ### Steps to reproduce.
> 在实际的调试过程中出现了如下报错： `E (16109) transport_ws: Error read response for Upgrade header GET /v1/chat?bot_id=74578324474 HTTP/1.1`e（16109）Transport_Ws：升级标题的错误响应get/v1/chat？ Connection: Upgrade  连接：升级 Host: ws.coze.cn:443  主持人：ws.coze.cn：443 User-Agent: ESP32 Websocket Client用户代理：ESP32 WebSocket客户 Upgrade: websocket  升级：Websocket Sec-WebSocket-Version: 13Sec-Websocket版本：13 Sec-WebSocket-Key: BdmZ1hCPWDvr7b2C92aVIQ==sec websecteet-key：bdmz1hcpwdvr7b2c92aviq == Authorization: Bearer pat_aKiztw5bggzq7d1e3qM0Cx授权：PARER PAT_AKIZTW5BGGZQ7D1E3QM0CX
> 
> E (16132) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=0, errno=119e（16132）websocket_client：ESP_TRANSPORT_CONNECT（）失败，cransport_error = esp_ok，tls_error_code = 0，tls_flags = 0，eps_ws_handshake_status_status_code = 0 W (16147) WebSocket: WebSocket Error: ERRORW（16147）WebSoctect：Web Socte错误：错误 I (16153) websocket_client: Reconnect after 10000 ms在（16153）WebSocket_client：10000 ms之后重新连接 W (16158) WebSocket: WebSocket DisconnectedW（16158）WebSecteet：网络杂志断开连接 E (16163) websocket_client: Client cannot be stopped from websocket taskE（16163）WebSocket_client：无法从WebSocket任务停止客户端 E (16170) websocket_client: Client cannot be stopped from websocket task`E（16170）WebSocket_client：无法从WebSocket任务停止客户端
> 
> TLS配置如下： `#
> 
> # ESP-TLS
> CONFIG_ESP_TLS_USING_MBEDTLS=yconfig_esp_tls_using_mbedtls = y
> 
> # CONFIG_ESP_TLS_USE_DS_PERIPHERAL is not set
> config_esp_tls_use_ds_peripheral未设置
> # CONFIG_ESP_TLS_CLIENT_SESSION_TICKETS is not set
> config_esp_tls_client_session_tickets未设置
> # CONFIG_ESP_TLS_SERVER_SESSION_TICKETS is not set
> config_esp_tls_server_session_tickets未设置
> # CONFIG_ESP_TLS_SERVER_CERT_SELECT_HOOK is not set
> config_esp_tls_server_cert_select_hook未设置
> CONFIG_ESP_TLS_SERVER_MIN_AUTH_MODE_OPTIONAL=yconfig_esp_tls_server_min_auth_mode_optional = y
> 
> # CONFIG_ESP_TLS_PSK_VERIFICATION is not set
> config_sp_tls_psk_verification未设置
> CONFIG_ESP_TLS_INSECURE=yconc_esp_tls_insecure = y CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=yconfig_esp_tls_skip_server_cert_verify = y
> 
> # end of ESP-TLS`  ESP-TLS的结尾
> 请问我这应该如何去寻找问题？用电脑客户端postman可以正常连接到服务器
> 
> ### Build or installation Logs.
> ```
> 
> ```
> 
> ### More Information.
> _No response  没有回应_

ssxx777, 2025-05-20:
兄弟解决了吗，我也这个问题，搞几天了

PoohWoah, 2025-05-20:
> 兄弟解决了吗，我也这个问题，搞几天了

现在新版本没问题了，他这个是服务器那边的问题，

ssxx777, 2025-05-20:
我用apifox是能连上的

ssxx777, 2025-05-20:
我用apifox是能连上的

PoohWoah, 2025-05-20:
我WS和RTC版本都跑完了，都可以直接出产品了。我不清楚你是什么问题。

ssxx777, 2025-05-20:
[image]

[image]

[image]
这是我的配置信息和错误信息，还有就是ESP-TLS中的两个选项不管是勾上还是取消我都试过了，都是不行，还有证书加不加我也都试过了

ssxx777, 2025-05-20:
ESP-TLS我重新改了一下，和你的改成一样的了但是还是有问题