espressif/esp-idf #15342: Hope to add HID and CDC composite methods, thanks!

mofa1, 2025-02-06:
### Is your feature request related to a problem?

I want to add HID and CDC features on my device, but I have limited level and have not succeeded, the composite program in the example is too simple for me to draw inferences, thanks!

tore-espressif, 2025-02-13:
He @mofa1 ,

The easiest way is to start from one of [USB device examples](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device) and modify it to your needs.

- Run `idf.py menuconfig` and in enable classes you want to use (CDC and HID) in `Component config → TinyUSB Stack ` menu. This will automaticly crate your USB descriptors that can be used as a starting point
- Use TinyUSB class API to control your device, eg for [HID class](https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid_device.h)

mofa1, 2025-02-13:
@tore-espressif Thanks for your reply,I'll give it a try

tore-espressif, 2025-03-04:
@mofa1 Feel free to reopen if you have any more questions