espressif/esp-idf #15343: CONFIG_ESP_CONSOLE_SECONDARY_NONE=y not entirely respected

florianhumblot, 2025-02-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to use the USB port of the ESP32C3 as a serial port.
To do this, without spamming the application logs over it, I set:
```
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
```
Now I can initialize the driver and install it as described in: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/usb-serial-jtag-console.html

However, after rebooting, the first time that I write to the USB port using `usb_serial_jtag_write_bytes` I'll get some logging that looks like it comes from application startup:

```
I (221) esp_image: segment 1: paddr=000728d0 vaddr=3fc956{"DeviceId":"mydevice","MacAddress":"f0-f5-bd-xx-xx-xx","SoftwareVersion":"dev"}
{"DeviceId":"mydevice","MacAddress":"f0-f5-bd-xx-xx-xx","SoftwareVersion":"dev"}
```

The code that I use to write to uart is:

```c++
void JtagConsole::Start() {
  usb_serial_jtag_driver_config_t usb_serial_jtag_config = {.tx_buffer_size = 255, .rx_buffer_size = 128};
  ESP_ERROR_CHECK(usb_serial_jtag_driver_install(&usb_serial_jtag_config));
}
void JtagConsole::WriteBytes(const char* data, int len) {
  usb_serial_jtag_write_bytes(data, len, 300 / portTICK_PERIOD_MS);
}
void JtagConsole::ReadBytes([[maybe_unused]] etl::istring& buffer) {
  usb_serial_jtag_read_bytes(buffer.begin(), buffer.max_size(), 0);
  buffer.trim_to_terminator();
}

JtagConsole::WriteBytes("{\"DeviceId\":\"mydevice\",\"MacAddress\":\"f0-f5-bd-xx-xx-xx\",\"SoftwareVersion\":\"dev\"}", 80);
```

Is there a way to prevent this from happening? 
I'm using esp-idf 5.1.5 and an ESP32-C3

suda-morris, 2025-02-07:
Hi @florianhumblot May I know if the [usb_serial_jtag_echo example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb_serial_jtag/usb_serial_jtag_echo) works for you?

florianhumblot, 2025-02-11:
@suda-morris can confirm that it also exhibits the behavior with the example.
I opened a serial monitor and typed `test` into it.

Log from the serial port (not USB):
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fcd5988,len:0x3250
load:0x403cc710,len:0x94c
load:0x403ce710,len:0x547c
entry 0x403cc710
I (74) boot: ESP-IDF v5.1.5-572-g8034b50d33 2nd stage bootloader
I (74) boot: chip revision: v0.4
I (74) boot.esp32c3: SPI Speed      : 80MHz
I (78) boot.esp32c3: SPI Mode       : DIO
I (83) boot.esp32c3: SPI Flash Size : 4MB
I (88) boot: Enabling RNG early entropy source...
I (93) boot: Partition Table:
I (97) boot: ## Label            Usage          Type ST Offset   Length
I (104) boot:  0 storage          WiFi data        01 02 00013000 00002000
I (111) boot:  1 phy_init         RF data          01 01 00015000 00001000
I (119) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (127) boot:  3 ota_0            OTA app          00 10 00020000 00180000
I (134) boot:  4 ota_1            OTA app          00 11 001a0000 00180000
I (142) boot:  5 nvs              WiFi data        01 02 00320000 00010000
I (149) boot: End of partition table
I (153) boot: No factory image, trying OTA 0
I (158) esp_image: segment 0: paddr=00020020 vaddr=3c020020 size=09eb8h ( 40632) map
I (173) esp_image: segment 1: paddr=00029ee0 vaddr=3fc8b000 size=01270h (  4720) load
I (176) esp_image: segment 2: paddr=0002b158 vaddr=40380000 size=04ec0h ( 20160) load
I (188) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=17b38h ( 97080) map
I (208) esp_image: segment 4: paddr=00047b60 vaddr=40384ec0 size=0603ch ( 24636) load
I (213) esp_image: segment 5: paddr=0004dba4 vaddr=00000000 size=0242ch (  9260)
I (221) esp_image: Verifying image signature...
I (221) secure_boot_v2: Verifying with RSA-PSS...
I (227) secure_boot_v2: Signature verified successfully!
I (233) boot: Loaded app from partition at offset 0x20000
I (266) boot: Set actual ota_seq=1 in otadata[0]
I (266) secure_boot_v2: enabling secure boot v2...
I (266) secure_boot_v2: secure boot v2 is already enabled, continuing..
I (273) boot: Disabling RNG early entropy source...
I (289) cpu_start: Unicore app
I (289) cpu_start: Pro cpu up.
I (298) cpu_start: Pro cpu start user code
I (298) cpu_start: cpu freq: 160000000 Hz
I (298) cpu_start: Application information:
I (301) cpu_start: Project name:     BCE_Monitor
I (306) cpu_start: App version:      0.1.0.0-dirty
I (312) cpu_start: Compile time:
I (316) cpu_start: ELF file SHA256:  2221067b1a6bff74...
I (322) cpu_start: ESP-IDF:          v5.1.4-674-gd38afc77db
I (328) cpu_start: Min chip rev:     v0.3
I (333) cpu_start: Max chip rev:     v1.99
I (338) cpu_start: Chip rev:         v0.4
I (343) heap_init: Initializing. RAM available for dynamic allocation:
I (350) heap_init: At 3FC8D270 len 00032D90 (203 KiB): DRAM
I (356) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (363) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (371) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (378) spi_flash: detected chip: generic
I (382) spi_flash: flash io: dio
I (386) sleep: Configure to isolate all GPIO pins in sleep state
I (392) sleep: Enable automatic switching of GPIO sleep configuration
I (400) coexist: coex firmware version: 27d8387
I (405) coexist: coexist rom version 9387209
[... 58 lines cut ...]
CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE=16
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=15
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUFFER=y
CONFIG_ESP_WIFI_MGMT_SBUF_NUM=16
# CONFIG_ESP_WIFI_IRAM_OPT is not set
# CONFIG_ESP_WIFI_RX_IRAM_OPT is not set
# CONFIG_ESP_WIFI_GMAC_SUPPORT is not set
# CONFIG_ESP_WIFI_SOFTAP_SUPPORT is not set
CONFIG_HEAP_TRACING_STANDALONE=y
CONFIG_HEAP_USE_HOOKS=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
CONFIG_LWIP_MAX_SOCKETS=5
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=4096
CONFIG_LWIP_TCP_WND_DEFAULT=4096
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA=y
CONFIG_MBEDTLS_ECP_RESTARTABLE=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
```

florianhumblot, 2025-02-11:
I've also attached a screen recording showing the issue: 
Left is the UART port, and right is the JTAG serial port
https://github.com/user-attachments/assets/24d8a741-f35c-4c6b-b2b7-05a5c7ea1ca4

mythbuster5, 2025-03-24:
It should be fixed by commit 12e9d6514af4f586765133ace13a3b2fa9ba7285, feel free to reopen if this can't solve your problem.

closed 2025-03-24 by mythbuster5

florianhumblot, 2025-03-24:
@mythbuster5 does this kind of work get backported to other releases like 5.1 or do I need to upgrade to the newer tags to get this fix? :)