espressif/esp-idf #15346: GPIO Wake has no affect on example with esp-zigbee-sdk

asward, 2025-02-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DEVKITM

### Power Supply used.

USB

### What is the expected behavior?

I expected to be able to modify the zigbee [light sleep example ](https://github.com/espressif/esp-zigbee-sdk/tree/8114916a4c6d1b4587a9fc24d2c85a1396328a28/examples/esp_zigbee_sleep/light_sleep) to use `gpio_wakeup_enable` and `esp_sleep_enable_gpio_wakeup`.

### What is the actual behavior?

It doesn't wake up on GPIO.

### Steps to reproduce.

1. Clone example code (https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_sleep/light_sleep)
2. Modify

```diff
80,81c80,81
<         ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(1ULL << CONFIG_GPIO_INPUT_IO_WAKEUP, ESP_EXT1_WAKEUP_ANY_LOW));
<  
---  
>         ESP_ERROR_CHECK(gpio_wakeup_enable(CONFIG_GPIO_INPUT_IO_WAKEUP, GPIO_INTR_LOW_LEVEL)) ;
>         ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup()) ;
```

3. `idf.py flash monitor`
4. Toggle `CONFIG_GPIO_INPUT_IO_WAKEUP` (defaults to 7 for C6)
5. No wake.

### Debug Logs.

```plain

```

### More Information.

Ref https://github.com/espressif/esp-zigbee-sdk/issues/550

1. With original config the wake reason is always `(0) other`, and randomly `(4) timer`
2. With original config the system wakes constantly (~50-100ms). It's difficult to determine if the ext pin is waking the chip or something else (ref 1 above).
3. With GPIO config it only wakes with `(0) other` but is always at the pre-set ED_KEEP_ALIVE time, which makes me think it's just the timer.
4. Changed to GPIO9 (C6 BOOT button): ext breaks (I think GPIO9 is not an RTC gpio on C6), but GPIO wake also does not work.

asward, 2025-02-20:
Hi @xieqinan . Is there anything I can provide you that would help? Additional logs or anything?

xieqinan, 2025-02-20:
> ESP_ERROR_CHECK(gpio_wakeup_enable(CONFIG_GPIO_INPUT_IO_WAKEUP, GPIO_INTR_LOW_LEVEL)) ;
> ESP_ERROR_CHECK(esp_sleep_enable_gpio_wakeup()) ;

If you intend to use the above method to wake up the device, ensure that `PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` is disabled. Please double-check this.

xieqinan, 2025-02-20:
By the way, if `PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` is disabled, the device's power consumption will increase during sleep since the peripheral remains active in light sleep.