espressif/esp-idf #15347: [ble-mesh] pass message to all elements when address is 0xFFFC-0xFFFF

chegewara, 2025-02-06:
Hi team,
sorry if i am wrong, because i am not 100% sure, but i believe that on node when destination address is special 0xfffX then message should be passed to all elements, not only root element (which is the case now).

Thanks

forx157, 2025-02-10:
Hi， @chegewara

forx157, 2025-02-10:
Hi, @chegewara 

[image]

According to spec, only the PRIMARY element handles these Fixed-group addresses.

chegewara, 2025-02-10:
Hi
thanks for explanation.
I understand it and accept it, but it is very odd. I mean, when i want to send device OFF to all models on all nodes with 0xFFFF to shutdown all then it wont be passed to models in non primary elements, unless i do it in my code (which is possible). For example when i have power supply splitters and i want to shutdown in case of fire.

Thanks

closed 2025-02-10 by chegewara

chegewara, 2025-02-10:
Hi @forx157 
maybe my english is not very good, but from my understanding the message shall be delivered to each instance of model when all conditions are meet
- condition 1 - meets
- condition 2 - meets
- condition 3 - destination address is set to any of the following

I mean it is not clear here, because part one of this sentence is excluding part 2
- `a fixed group address which is defined in table and meets the conditions` excluding `the message shall be delivered only to primary element`

and then this one: 
- `a group address, except the all-nodes address`

Thanks

BTW can you share which docs is it from, please