espressif/esp-idf #15348: Expose access to the wifi_pkt_rx_ctrl_t.timestamp source.

robiwano, 2025-02-06:
### Is your feature request related to a problem?

I have an application where I need to time events relative to the WiFi RX timestamp, and thus an exposure of the actual timer value would be very helpful.

zhangyanjiaoesp, 2025-02-10:
@robiwano 
Is the description in this issue (https://github.com/espressif/esp-idf/issues/9843) consistent with your requirements?

robiwano, 2025-02-10:
Unfortunately not. When a Wifi packet is received by the ESP32, somewhere the RX timestamp is set:

   rx_ctrl.timestamp = _wifi_timer_for_timestamps()_;

so what I am after is access to the _wifi_timer_for_timestamps()_ function. As I've surmised from the ESP32 forums, the timer is APB based (1MHz).

It need to be able to read this value for synchronizing events/peripherals later on in my app.

zhangyanjiaoesp, 2025-02-12:
@robiwano 
This `rx_ctrl.timestamp`  is obtained from the internal MAC clock. Would directly using the value of `rx_ctrl.timestamp` not meet your application requirements? Could you describe your synchronization requirements?

robiwano, 2025-02-12:
I have a scenario where I broadcast ESP-NOW messages with a "server" timestamp (1us timebase), and it is received by multiple client ESP32-S3s (in my case). I can then relate the `rx_ctrl.timestamp` to the server timestamp (which does suffer from network delay but should be the same delay experienced by all clients). By using the MAC clock to start a APB (1us) clock with the same value, I can then, given a later server timestamp, synchronize events with the server timestamp, like toggling a port for instance. The use case is for multiple ESP32s as clients in the system. Do I make any sense? :)

zhangyanjiaoesp, 2025-02-12:
@robiwano 
If you want to synchronize the client's timestamp with the server's timestamp, you can use let the clients(STA) connect to the server (softAP), and then use the tsf timer.

zhangyanjiaoesp, 2025-02-13:
@robiwano 
Which ESP device are you using? I can provide you with the register address, and you can try to see if it meets your expectations.

robiwano, 2025-02-13:
I'm using the ESP32-S3.

zhangyanjiaoesp, 2025-02-13:
> I'm using the ESP32-S3.

The register is `0x60035000`

robiwano, 2025-02-13:
32 or 64 bit?

zhangyanjiaoesp, 2025-02-13:
> 32 or 64 bit?

uint32_t

robiwano, 2025-02-13:
Confirming that the counter value from that register definitely matches rx_ctrl.timestamp. The wifi promiscuous callback is done in a FreeRTOS task, and I notice that the difference between the counter and rx_ctrl.timestamp always is around ~2150 us. This long between a received packet and processing it... is that to be expected?

zhangyanjiaoesp, 2025-02-14:
Due to task scheduling and other factors, this time difference is reasonable

robiwano, 2025-02-14:
Interestingly enough, now the latency from MAC timestamp to wifi callback is < 200 us which is very good :) (noted with the probe response message)

closed 2025-03-19 by espressif-bot