espressif/esp-idf #15349: Bluetooth mesh: Generic Battery Client parser does not correctly parse Generic Battery Status Messages

connyhald, 2025-02-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4 and git master

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed Studio XIAO ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

I'm sending `Generic Battery Status` messages from a `Generic Battery Server` model to a `Generic Battery Client` model.

I would expect the parameters for `time_to_discharge` and `time_to_charge` to be the same on the receiver side then they have been on the sender side. See: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/bluetooth/esp-ble-mesh.html#_CPPv436esp_ble_mesh_gen_battery_status_cb_t

### What is the actual behavior?

What I receive is something different then what I did send.

### Steps to reproduce.

1. Implement a `Generic Battery Server` on device A
2. Implement a `Generic Battery Client` on device A
3. Provision both devices (I did use the `nRF Mesh` app from Nordic)
4. Setup publication on the `Generic Battery Server` model to publish to group `X`
5. Setup subscription on the `Generic Battery Client` model to subscribe to group `X`
6. Observe what gets send by the server and what gets received by the client - e.g. by logging the values
7. Also use the `nRF Mesh Sniffer` app to sniff the traffic between the two nodes

### Debug Logs.

```plain

```

### More Information.

The following screenshot shows how the message looks like when sniffed and how it is (correctly I think) interpreted. Please focus on `Time to Discharge` and `Time to Charge` and ignore `Battery Flags` (there seems to be an issue on the Nordic side).
[image]

On the receiver side we log the following for the same mesh message.
```
I (1304601) Client -- Received Generic Battery Status
I (1304601) Client -- Battery Level: 100
I (1304601) Client -- Time to Discharge: 1576960
I (1304611) Client -- Time to Charge: 8656384
I (1304611) Client -- Flags: 0x00
```

Note that `Battery Level` is fine but `Time to Discharge` and `Time to Charge` is wrong.

I think the issue can be found here:
https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/components/bt/esp_ble_mesh/models/client/generic_client.c#L306-L312

Please see the following screenshot that shows the expected and the actual value. Please note how similar the bit pattern is. I think there is an error in the above code that leads to some bits being shifted wrongly.
[image]

connyhald, 2025-02-09:
My hacky workaround looks like this. I've changed the file `generic_client.c` as follows.

```c
    case BLE_MESH_MODEL_OP_GEN_BATTERY_STATUS: {
        struct bt_mesh_gen_battery_status *status = NULL;
        if (buf->len != 8) {
            BT_ERR("Invalid Generic Battery Status length %d", buf->len);
            return;
        }
        status = bt_mesh_calloc(sizeof(struct bt_mesh_gen_battery_status));
        if (!status) {
            BT_ERR("%s, Out of memory", __func__);
            return;
        }
        
        // Start change
        uint32_t byte0 = net_buf_simple_pull_u8(buf);
        uint32_t byte1 = net_buf_simple_pull_u8(buf);
        uint32_t byte2 = net_buf_simple_pull_u8(buf);
        uint32_t byte3 = net_buf_simple_pull_u8(buf);
        uint32_t byte4 = net_buf_simple_pull_u8(buf);
        uint32_t byte5 = net_buf_simple_pull_u8(buf);
        uint32_t byte6 = net_buf_simple_pull_u8(buf);
        uint32_t byte7 = net_buf_simple_pull_u8(buf);

	status->battery_level     = byte0;
    	status->time_to_discharge = byte3 | byte2 << 8 | byte1 << 16;
	status->time_to_charge    = byte6 | byte5 << 8 | byte4 << 16;
        status->flags             = byte7;
        // End change

        val = (uint8_t *)status;
        len = sizeof(struct bt_mesh_gen_battery_status);
        break;
    }
```

There are probably more elegant ways of doing this, but at least for me it work...

besbar, 2025-03-11:
Hi everyone, 

I am a enthusiast about ble mesh and I have build a network at home with multiple esp32 and vendor devices.
I have so far used quite a lot of blemesh models and also coding a provisionner. 

But I am still looking at examples for generic Battery Model (server and client).
Also I would be very happy to obtin some examples of Opcodes Aggregator (server and client).

Thanks in advance if any help can be provided.

connyhald, 2025-03-23:
@besbar I'm planning to write things down once I've figured everything out. In the mean time I've copied the most relevant pieces into a gist. We can continue discussions there if you want: https://gist.github.com/connyhald/8a14e573c41c03095452c80a3fcf3ef6

connyhald, 2026-02-10:
Did anyone ever look into this? I'm still seeing this issue on IDF 5.5.2