espressif/esp-idf #15351: "os_memblock_get failed.." in monitor log while trying to connect to WiFi AP.

scholayil, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release v5.4

### Espressif SoC revision.

ESP32-C3 (QFN32) (version v0.4)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with ESP32-C3-WROOM-02 module

### Power Supply used.

USB

### What is the expected behavior?

Previously provisioned WiFi connection attempt results in ""os_memblock_get failed.." in Monitor log. I do not see this with 5.3.1 or earlier idf versions. Also, free heap is 7K less for the same application compiled with 5.4 compared to 5.3.1 idf version. Not sure the two are related but the warning is a concern on its own.

The app also uses Nimble BLE in the central mode (normal boot mode) and in peripheral mode during WiFi provisioning.

### What is the actual behavior?

 os_memblock_get failed (size 4)  warning seen once to repeated 4 times during boot up wifi connection.

### Steps to reproduce.

Compile applicaiton with 5.4, program and boot device.

### Debug Logs.

```plain
I (7204) _blecent_update_status: Central status set: 8 
I (7204) _blecent_update_status: Central status get: 9
----
 os_memblock_get failed (size 4) 
I (8254) wifi:state: auth -> assoc (0x0) 
I (8264) wifi:Association refused temporarily time 1000, comeback time 1100 (TUs) 
I (9384) wifi:state: assoc -> auth (0xb0) 
os_memblock_get failed (size 4) 
os_memblock_get failed (size 4) 
os_memblock_get failed (size 4) 
I (10244) SYS_Init: Heap 14.2% free - 34472 of 243580 bytes
```

hrushikesh430, 2025-02-10:
Hi @scholayil, I compiled the wifi_prov_mgr application with verrsion 5.4 and 5.3.1 and provisioned wifi credentials. After resetting the device I do not encountered any `os_memblock_get failed`. 7k less free heap memory captured but not the monitor log. Could you please correct me if I am using incorrect way to reproduce or if possible please send the patch file of changes.

SCgenie, 2025-02-10:
@hrushikesh430 You are correct about the WiFi_Provisioning_Mgr application. I tried it under ESP-idf 5.4 on esp32-wroom-02 module. I do not see the "os_memblock_get failed.." with that application. I also tried the ESP HTTP Client Example with an actual wifi connection to AP. I did not see the "os_memblock_get failed.." warning.

I will try to test this with a new application using both WiFI and BLE central that I may be able to share.  I will let you know when I have the result.

For the 7K loss I mentioned, it may come down to the esp-idf 5.4 using more static DRAM memory and not related to the "os_memblock_get failed.." warning at all. Please find attached WiFi_Provisioning_Mgr application build result comparing DRAM usage on 5.4/5.3.1 . Will this ever improve or will 5.4 continue to use more static DRAM memory as seen in the attachment? 
Thanks

[image]

scholayil, 2025-02-11:
In my production code, I can avoid "os_memblock_get failed.." warnings, by disabling ble central startup as a test case.

scholayil, 2025-05-14:
Hi @mahavirj 

Is this 5K increased static memory utilization going to be in newer releases?

Thanks

> [@SCgenie](https://github.com/SCgenie)
> 
> You are right about the increased static memory utilization from v5.4 release. Most of the impact (around 5K) is coming from Bluetooth library (`libbtdm_app.a`). This can be found using following command:
> 
> ```
> idf_size.py --archives --diff <PREV_MAP_FILE>  <CURRENT_MAP_FILE>
> ```
> 
> Our Bluetooth team will look into this further, thanks for notifying.

Weijian-Espressif, 2025-05-16:
@scholayil After the IDF version upgrade, we fixed some bugs. The rom of this part of the code was placed in IRAM, which led to a reduction in the available system memory. On v5.4, we provide the configuration of the flash mode. After opening this configuration, all the code in IRAM can be placed in flash, saving memory. You can test whether it meets your requirements. Additionally, we are still optimizing the code to reduce the consumption of IRAM and flash. It will be merged in as soon as possible

esp-zhp, 2025-05-28:
commit c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74 (HEAD, tag: v5.3.1)
[image]
commit 67c1de1eebe095d554d281952fde63c16ee2dca0 (HEAD, tag: v5.4
[image]