espressif/esp-idf #15352: ESP-IDF Release v5.4 CONFIG_BT_SSP_ENABLED

tv4you2016, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I just updated my project to **ESP-IDF Release v5.4**.
Before the update I was using the **Release v5.1.5** version for **ESP-IDF Release v5.4** where I had pairing with a BT device with ESP_BT_PIN_TYPE_FIXED 
```
# CONFIG_BT_SPP_ENABLED is not defined
# CONFIG_BT_SSP_ENABLED is not defined
```

in sdkconfig.esp32 and everything works fine, 
Now that I've updated, when pairing a device, the pin is no longer asked for.

After comparing the sdkconfig. esp32 of the new idf with the old one, I discovered that the CONFIG_BT_SSP_ENABLED setting 
It's not in the new one.

I did a search at ESP-IDF Programming Guide[v5.1.5](https://docs.espressif.com/projects/esp-idf/en/v5.1.5/esp32/index.html#esp-idf-programming-guide)  by this config and it exists, search in ESP-IDF Programming Guide [v5.4](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/index.html#esp-idf-programming-guide) he doesn't exist

Is there any reason why this setting disappeared?

boblane1, 2025-02-07:
Hi @tv4you2016,

Thanks for your reporting. From release/v5.2, pairing mode can be configured at runtime. API `esp_bluedroid_init` has been updated with `esp_bluedroid_init_with_cfg`. You can use legacy pairing by calling `esp_bluedroid_init_with_cfg` with the parameter `ssp_en` setting to `false`. We will add some instruction about this change later.

Alvin1Zhang, 2025-02-08:
Thanks for reporting and sharing the updates, will reopen and keep the ticket open until the fix is available on GitHub, thanks.

closed 2025-02-13 by espressif-bot