espressif/esp-idf #15354: 使用idf.py build size 之后IRAM大小只有16K，怀疑只读到了RTC RAM的大小

ZhkuWuMingyang, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

IRAM应该显示有300+KB大小

### What is the actual behavior?

IRAM显示只有16KB

### Steps to reproduce.

1. 使用idf.py create-project ttttt创建一个空白工程
2. 使用idf.py set-target esp32s3切换目标芯片
3. 使用idf.py build size

### Build or installation Logs.

```plain
wmy@wmy:~/esp/work/ttttt$ idf.py build size
Executing action: all (aliases: build)
Running ninja in directory /home/wmy/esp/work/ttttt/build
Executing "ninja all"...
[1/4] cd /home/wmy/esp/work/ttttt/buil...ome/wmy/esp/work/ttttt/build/ttttt.bin
ttttt.bin binary size 0x357b0 bytes. Smallest app partition is 0x100000 bytes. 0xca850 bytes (79%) free.
[1/1] cd /home/wmy/esp/work/ttttt/buil.../ttttt/build/bootloader/bootloader.bin
Bootloader binary size 0x5250 bytes. 0x2db0 bytes (36%) free.
Executing action: size
Running ninja in directory /home/wmy/esp/work/ttttt/build
Executing "ninja all"...
[1/4] cd /home/wmy/esp/work/ttttt/buil...ome/wmy/esp/work/ttttt/build/ttttt.bin
ttttt.bin binary size 0x357b0 bytes. Smallest app partition is 0x100000 bytes. 0xca850 bytes (79%) free.
[1/1] cd /home/wmy/esp/work/ttttt/buil.../ttttt/build/bootloader/bootloader.bin
Bootloader binary size 0x5250 bytes. 0x2db0 bytes (36%) free.
Running ninja in directory /home/wmy/esp/work/ttttt/build
Executing "ninja size"...
[0/1] cd /home/wmy/esp/work/ttttt/build && /usr/bin/cmake -D "IDF_SIZE_TOOL=/home/wmy/.espressif/python_env/idf5.2_py3.8_env/bin/python;-m;esp_idf_size" -D MAP_FILE=/home/wmy/esp/work/ttttt/build/ttttt.map -D OUTPUT_JSON= -P /home/wmy/esp/esp-idf/tools/cmake/run_size_tool.cmake
Total sizes:
Used static IRAM:   16383 bytes (      1 remain, 100.0% used)
      .text size:   15356 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   52831 bytes ( 293025 remain, 15.3% used)
      .data size:   10432 bytes
      .bss  size:    2256 bytes
      .text size:   40143 bytes
Used Flash size :  151987 bytes
           .text:   99567 bytes
         .rodata:   52164 bytes
Total image size:  218945 bytes (.bin may be padded larger)

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/ttttt.bin
or from the "/home/wmy/esp/work/ttttt/build" directory
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
```

### More Information.

1.idf.py 版本为v5.2.1，尝试把版本升级到v5.2.3，问题都没有解决
2.尝试重新装esp-idf，也没有解决
3.目前问题只出现在esp32s3这个target，其他target正常
4.更换其他s3芯片也是一样的情况
5.尝试过新建一个空白工程，设置为esp32s3，IRAM大小也只显示16K
6.系统是在VM运行的：Ubuntu 20.04.6 LTS

fhrbata, 2025-02-07:
Hello @ZhkuWuMingyang ,

https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf

> 4.3.2 Internal Memory

> 3. Internal SRAM 0
> Internal SRAM 0 is a 32 KB, read-and-write memory space, addressed by the CPU through the instruction bus,
> as shown in Table 4.3-1.
> A 16 KB or the total 32 KB of this memory space can be configured as instruction cache (ICache) to store
> instructions or read-only data of the external memory. In this case, the occupied memory space cannot be
> accessed by the CPU, while the remaining can still can be accessed by the CPU.

> 4. Internal SRAM 1
> Internal SRAM 1 is a 416 KB, read-and-write memory space, addressed by the CPU through the data bus or
> instruction bus in the same order, as shown in Table 4.3-1.

SRAM0 is mapped at `0x4037_0000 - 0x4037_7FFF` via instruction bus. It has a total size of 32KB, but by default, 16KB is allocated for the instruction cache, as determined by `CONFIG_ESP32S3_INSTRUCTION_CACHE_SIZE`. This leaves 16KB of IRAM available in SRAM0. You can find this information at https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32s3/memory.ld.in#L40. The section labeled as `Used static IRAM:` in the output refers to SRAM0.

SRAM1 is mapped at `0x4037_8000 - 0x403D_FFFF` via instruction bus and directly follows the SRAM0 mapping. Additionally, SRAM1 can be accessed through the data bus at `0x3FC8_8000 - 0x3FCE_FFFF`, indicating that it is not just IRAM but DIRAM. This is reflected in the `Used stat D/IRAM:` section in the output, where `(293025 remain, 15.3% used)` signifies that 293KB is still available for code or data. Please be aware that the `Used stat D/IRAM` section includes both the `.data` and `.text` sections, whereas the `Used static IRAM` section contains only the `.text` section.

The code is first placed into SRAM0 (IRAM) and may extend into SRAM1 (DIRAM) if needed. This explains the full SRAM0 (IRAM) usage shown as `(1 remain, 100.0% used)`. The data is positioned after the code in SRAM1 (DIRAM), and its starting address depends on how much code is already in SRAM1 (DIRAM). This means there is no fixed starting address for the data in SRAM1 (DIRAM) and this is expressed by a dummy output section, which represent the part of SRAM1 (DIRAM) used for code (IRAM). It can be seen here https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32s3/sections.ld.in#L206.

Because SRAM1 (DIRAM) is utilized for both code and data, it's not possible to determine the exact amount of space left as IRAM and DRAM. Therefore, esp-idf-size reports it together as `D/IRAM`. This division, based on how SRAM can be accessed, aims to provide accurate remaining values. 

For a reference, I'm including a commit that resolved the outstanding DIRAM calculation issue: https://github.com/espressif/esp-idf-size/commit/88a8b132e1b9eb8f418e4155252eab336e7db871
Before this commit the remaining DIRAM was accounted twice, once as remaining IRAM and second time for remaining DRAM, providing a misleading information about how much memory remains available.

I hope this clarifies how the DIRAM is represented in the `esp-idf-size` output for the esp32s3.

Apologies for the lengthy post and thank you.

ZhkuWuMingyang, 2025-02-08:
感谢解答，虽然作为初学者还是不太能看懂，我会继续努力的

closed 2025-02-08 by ZhkuWuMingyang

ZhkuWuMingyang, 2025-02-08:
在看了ESP32和ESP32S3的手册关于Internal SRAM1部分的描述是都是：可以通过数据或指令总线访问。那么为什么ESP32可以分别计算其中IRAM和DRAM的占用，而ESP32S3要一起报告呢？

fhrbata, 2025-02-09:
Hello @ZhkuWuMingyang, the difference lies in how SRAM1 is utilized on the esp32 compared to the esp32s3. On the esp32 it is not by default used for application's IRAM nor DRAM, but SRAM0 is used for IRAM, and SRAM2 is for DRAM, with no memory aliasing (DIRAM) between IRAM and DRAM as seen in the esp32s3 usage of SRAM1. The linker script specifies which parts of SRAM are designated for IRAM and DRAM. You can examine the linker templates for the `MEMORY` command, which are used to generate the final linker script for each target. For instance, for the esp32, it's located at `components/esp_system/ld/esp32/memory.ld.in`. The memory regions for IRAM (iram0_0_seg) and DRAM (dram0_0_seg) are defined there. You can compare the address ranges with the TRM to determine which SRAM they map to. The final linker script with the `MEMORY` command can be found in the build directory at `build/esp-idf/esp_system/ld/memory.ld`. The memory regions as defined in the linker script are also listed in the link map file under the `Memory Configuration` section available in the project build directory(e.g. `build/hello_world.map`). This map file is what esp-idf-size uses to calculate the total available sizes for different memory regions. Additionally, esp-idf-size contains chip info configuration files for each target. These are available at https://github.com/espressif/esp-idf-size/tree/master/esp_idf_size/chip_info, where memory regions are mapped to different memory types(IRAM, DRAM, DIRAM, ...). The configuration files also contain information about memory aliasing(`secondary_address` entry for DIRAM) if any. Generally the key is the linker script which defines how the memory is used and what  goes into what memory region.

The final linker script is generated based on the configuration and the following templates
* https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32/memory.ld.in
* https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32/sections.ld.in
* https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/ld/esp32/bootloader.ld

The generated files related to the application link can be found in the `build` directory. For example for `hello_world`
* build/esp-idf/esp_system/ld/memory.ld
* build/esp-idf/esp_system/ld/sections.ld
* build/hello_world.map

esp-idf-size chip information for esp32 and esp32s3(note the `secondary_address`)
* https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/chip_info/esp32.yaml
* https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/chip_info/esp32s3.yaml

Hopefully this information clarifies it somewhat. Thank you