espressif/esp-idf #15357: Using GPIO 18 ledc pwm to chirp buzzer , getting Warning: ledc: GPIO 18 is not usable, maybe conflict with others

scholayil, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The buzzer on the gpio18 sounds ok. However getting the warning the second time (and thereafter) the buzzer is activated.

void BUZZ_init(gpio_num_t pin){
   gpio_reset_pin(pin);

   gpio_config_t io_conf;
   io_conf.intr_type = GPIO_INTR_DISABLE;       // disable interrupt
   io_conf.mode = GPIO_MODE_OUTPUT;             // set as output mode
   io_conf.pin_bit_mask = pin;                  // set bitmask
   io_conf.pull_down_en = 0;                    // disable pull-down mode
   io_conf.pull_up_en = 1;                      // enable pull-up mode
   gpio_config(&io_conf);                       // configure GPIO with the given settings
   
}

void _buzzOn(uint32_t dutyCycle) {
    // Configure the LEDC timer
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_RESOLUTION,  // Resolution of PWM duty
        .freq_hz = LEDC_FREQUENCY,           // Frequency of PWM signal
        .speed_mode = LEDC_MODE,             // Timer mode
        .timer_num = LEDC_TIMER              // Timer index
    };
    ledc_timer_config(&ledc_timer);

    // Configure the LEDC channel
    ledc_channel_config_t ledc_channel = {
        .intr_type = LEDC_INTR_DISABLE,
        .channel    = LEDC_CHANNEL,
        .duty       = dutyCycle,             // Initial duty cycle (off)
        .gpio_num   = BUZZER_GPIO_PIN,       // GPIO pin
        .speed_mode = LEDC_MODE,             // Speed mode
        .hpoint     = 0,                     // High point of PWM signal
        .timer_sel  = LEDC_TIMER             // Use the configured timer
    };
    ledc_channel_config(&ledc_channel);
}

void _buzzKill(void) {
   ledc_stop(LEDC_MODE, LEDC_CHANNEL, 0); 
}

EqUiNoX-Labs, 2025-02-07:
Just to confirm i also have same issue...

looks like its just a warning... but would be great to have a fix...

https://github.com/anujdeshpande/esp-idf/blob/cb3ac7429c73b6e7c6d79ae05de480d4b6706770/components/esp_driver_ledc/src/ledc.c#L809

songruo, 2025-02-10:
Your `_buzzOn` function is more like `BUZZ_init`. `ledc_timer_config` and `ledc_channel_config` do not need to be called every time you pause and resume the buzzer. (Another side note, `gpio_reset_pin` and `gpio_config` are unnecessary to be called)

If you use `ledc_stop` to pause the buzzer, then call `ledc_update_duty` should resume it.

And as another better option, `ledc_timer_pause` and `ledc_timer_resume` APIs are provided to pause and resume the PWM signal.

SCgenie, 2025-02-10:
@songruo Thank you for pointing to the error.

SCgenie, 2025-02-10:
Please close the issue. Thank you.

closed 2025-02-11 by scholayil