espressif/esp-idf #15359: Memory Protection causing reboot loop

paulblid, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3, v5.4

### Espressif SoC revision.

ESP32-S2 rev 1.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The code should throw a memory protection fault and then reboot into user code.

### What is the actual behavior?

After the first crash (with a memory protection fault), the cpu setup code encounters a Cache disabled exception, which causes a reboot loop.
Turning off memory protection avoids this behavior entirely, but is counterproductive to our app as we would like it to crash if it accesses memory in an unintended way.

### Steps to reproduce.

Minimal example code:

```c
#include <stdio.h>
#include <stdint.h>
#include <string.h>

void app_main(void) {
  uint8_t* heap = (uint8_t*)malloc(7);
  // void* bad_addr = (void*)0x39063232;
  void* bad_addr = (void*)0x39000000; // can be any address between 0x20000000-0x39FFFFFF
  memcpy(heap, bad_addr, 7);
}
 
```

All sdkconfig settings are identical to the official template-app.

### Debug Logs.

```plain
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe5110,len:0x1664
load:0x4004a000,len:0x4
[... 188 lines cut ...]
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000014  A9      : 0x00000001
A10     : 0x00000028  A11     : 0x00000001  A12     : 0x00000028  A13     : 0x00000003
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000007  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000028  LEND    : 0x00000003  LCOUNT  : 0x400250f0
--- 0x400250f0: xt_highint4 at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_system/port/soc/esp32s2/highint_hdl.S:52

Backtrace: 0x4002f9a2:0x3fffe260 0x40025c37:0x3fffe270 0x40083779:0x3fffe290 0x40083e18:0x3fffe2b0 0x40024fc5:0x3fffe2f0 0x4004f555:0x3fffe330 0x4004f969:0x3fffe370 0x4004a231:0x3fffe4a0 0x40011371:0x3fffe560 0x4000f788:0x3fffe6f0 0x400072a5:0x3fffe710
--- 0x4002f9a2: xt_ints_on at E:/espressif/v5.4.0/v5.4/esp-idf/components/xtensa/xtensa_intr_asm.S:178
0x40025c37: esp_cpu_intr_enable at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_hw_support/include/esp_cpu.h:388
 (inlined by) esp_intr_enable_source at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_hw_support/intr_alloc.c:939
0x40083779: esp_memprot_intr_init at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_hw_support/port/esp32s2/memprot.c:144
0x40083e18: esp_memprot_set_prot at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_hw_support/port/esp32s2/memprot.c:819 (discriminator 1)
0x40024fc5: call_start_cpu0 at E:/espressif/v5.4.0/v5.4/esp-idf/components/esp_system/port/cpu_start.c:767
0x40011371: ets_run_flash_bootloader in ROM
0x4000f788: main in ROM
0x400072a5: _start in ROM
```

### More Information.

* There seems to be a related issue 2 years ago but it has remained open.
* Putting the panic handler in ISR does not help.
* Changing coredump settings does not help (coredump config is also showing up as corrupted if coredump is enabled)
* The specific line causing the Disabled cache error seems to be memprot interrupt enabling code `esp_memprot_intr_init`. 
* My current (probably wrong) theory is that there is some state left over from the crash, or some interrupt that is not cleared, and upon it being enabled by the memprot code, it calls into that interrupt handler, which is probably not located in IRAM, but I did not have enough time to go down this rabbit hole any further.

o-marshmallow, 2025-02-08:
Hello @paulblid ,

Thanks for reporting this issue, I was able to reproduce it on my side, I am going to investigate it.

o-marshmallow, 2025-02-11:
Hello @paulblid ,

Sorry for not giving an update, I have already opened an internal PR to fix this issue, it is currently under review. 

I confirm that the source of the issue is that the memory protection interrupt state is not cleaned before re-enabling the interrupts on boot, hence the boot loop.

o-marshmallow, 2025-02-14:
This fix will be backported to the older versions that are still maintained

mahavirj, 2025-02-18:
> Shouldn't this be a call to memprot_ll_peri1_is_assoc_intr instead of memprot_ll_dram0_is_assoc_intr considering it's referring to peri1 not dram0?

You are right here. We will create an internal fix for this, thanks for pointing it out!

pacucha42, 2025-02-18:
@paulblid thanks for noticing the 'memprot_ll_dram0_is_assoc_intr' issue, seemingly it's been sleeping there for years. I'll provide the fix and port it back to all the relevant IDF versions. You will be posted

paulblid, 2025-02-18:
@pacucha42 thanks for the update.

Just a note, I opened the initial issue, but the credit for the fixes posted above and the discovery of `memprot_ll_dram0` bug go to my colleague @lucic71 .

pacucha42, 2025-02-18:
Ok, thank you both 🙂 It was "standard" copy&paste error, unfortunately, as the RTCFAST shares the violation interrupts with DRAM0 bus and the handling logic is the same for all the RTC memory ranges. Coincidentally, this setup also kept passing the tests and thus stayed hidden.

The code is fixed, the tests are updated. The patch will be released shortly after all rounds of internal verification

closed 2025-02-28 by espressif-bot