espressif/esp-idf #15360: DHCP OPTIONS NOT WORKING - Esp-Idf 5.4 maybe others

filzek, 2025-02-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Trying to setup a.dhcpc options vendor information or any other and the return is invalid always.

If it connects to the wifi its return dhcp already started

It seems a bug.

So, how to correct set it?

zhangyanjiaoesp, 2025-02-19:
@filzek 
Can you provide a demo for your issue?

zhangyanjiaoesp, 2025-02-20:
@filzek 
In the menuconfig, the default value of `LWIP_DHCP_DISABLE_VENDOR_CLASS_ID` is `yes`, have you set it to `no`?

[image]

filzek, 2025-02-20:
#
# LWIP
#
CONFIG_LWIP_ENABLE=y
CONFIG_LWIP_LOCAL_HOSTNAME="wisehome"
# CONFIG_LWIP_NETIF_API is not set
CONFIG_LWIP_TCPIP_TASK_PRIO=18
# CONFIG_LWIP_TCPIP_CORE_LOCKING is not set
# CONFIG_LWIP_CHECK_THREAD_SAFETY is not set
CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES=y
# CONFIG_LWIP_L2_TO_L3_COPY is not set
# CONFIG_LWIP_IRAM_OPTIMIZATION is not set
# CONFIG_LWIP_EXTRA_IRAM_OPTIMIZATION is not set
CONFIG_LWIP_TIMERS_ONDEMAND=y
CONFIG_LWIP_ND6=y
# CONFIG_LWIP_FORCE_ROUTER_FORWARDING is not set
CONFIG_LWIP_MAX_SOCKETS=10
# CONFIG_LWIP_USE_ONLY_LWIP_SELECT is not set
# CONFIG_LWIP_SO_LINGER is not set
CONFIG_LWIP_SO_REUSE=y
CONFIG_LWIP_SO_REUSE_RXTOALL=y
# CONFIG_LWIP_SO_RCVBUF is not set
# CONFIG_LWIP_NETBUF_RECVINFO is not set
CONFIG_LWIP_IP_DEFAULT_TTL=64
CONFIG_LWIP_IP4_FRAG=y
CONFIG_LWIP_IP6_FRAG=y
# CONFIG_LWIP_IP4_REASSEMBLY is not set
# CONFIG_LWIP_IP6_REASSEMBLY is not set
CONFIG_LWIP_IP_REASS_MAX_PBUFS=10
# CONFIG_LWIP_IP_FORWARD is not set
# CONFIG_LWIP_STATS is not set
CONFIG_LWIP_ESP_GRATUITOUS_ARP=y
CONFIG_LWIP_GARP_TMR_INTERVAL=60
CONFIG_LWIP_ESP_MLDV6_REPORT=y
CONFIG_LWIP_MLDV6_TMR_INTERVAL=40
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=32
CONFIG_LWIP_DHCP_DOES_ARP_CHECK=y
# CONFIG_LWIP_DHCP_DOES_ACD_CHECK is not set
# CONFIG_LWIP_DHCP_DOES_NOT_CHECK_OFFERED_IP is not set
# CONFIG_LWIP_DHCP_DISABLE_CLIENT_ID is not set
# CONFIG_LWIP_DHCP_DISABLE_VENDOR_CLASS_ID is not set
# CONFIG_LWIP_DHCP_RESTORE_LAST_IP is not set
CONFIG_LWIP_DHCP_OPTIONS_LEN=68
CONFIG_LWIP_NUM_NETIF_CLIENT_DATA=0
CONFIG_LWIP_DHCP_COARSE_TIMER_SECS=1

david-cermak, 2025-02-27:
Hi @filzek 

The reason you're seeing this error:

> Failed to set Vendor Class Identifier: ESP_ERR_ESP_NETIF_INVALID_PARAMS

is that you're calling `esp_netif_dhcpc_option()` immediately after starting wifi, but before wifi's netif starts, so its dhcp is not available yet. Now this is a little tricky with the default wifi interface, since we automatically start the DHCP client upon WiFI start event, so you'd get `ESP_ERR_ESP_NETIF_DHCP_ALREADY_STOPPED` if you called it after wifi's netif got started.

to configure DHCP options you'd have to stop the client, set selected options and start it again.
```cpp
    esp_netif_dhcpc_stop(netif);
    esp_netif_dhcpc_option(netif, ESP_NETIF_OP_SET, ESP_NETIF_VENDOR_CLASS_IDENTIFIER, (void*)vendor_class_id, strlen(vendor_class_id));
    esp_netif_dhcpc_start(netif);
```

filzek, 2025-02-27:
@david-cermak,

same problem!

#ifndef CONFIG_LWIP_DHCP_DISABLE_VENDOR_CLASS_ID
esp_netif_dhcpc_stop(sta_netif);
const char *vendor_class_id = "Testing";
esp_err_t err = esp_netif_dhcpc_option(sta_netif, ESP_NETIF_OP_SET, ESP_NETIF_VENDOR_CLASS_IDENTIFIER, &vendor_class_id, strlen(vendor_class_id));
if (err != ESP_OK) {
     printf("Failed to set Vendor Class Identifier: %s\n", esp_err_to_name(err));
} else {
     printf("Vendor Class Identifier set to: %s\n", vendor_class_id);
}
esp_netif_dhcpc_start(netif);

#endif

RESULT:
Failed to set Vendor Class Identifier: ESP_ERR_ESP_NETIF_INVALID_PARAMS

So still not working

david-cermak, 2025-02-28:
Created a minimal project using VCI and VSI options. Tested on `v5.4` and master, and also with WiFi and Ethernet and it is working on my side.

@filzek Could you please give it a try: https://github.com/david-cermak/esp-network-examples/tree/main/dhcp-vendor-spec

filzek, 2025-02-28:
@david-cermak  in a real application scenario it doesnt work.

simple return the same problem, 
Failed to set Vendor Class Identifier: ESP_ERR_ESP_NETIF_INVALID_PARAMS

we have refatored our code to be called in the event handler to do the same as you are doing in your application:

we have called it void set_dhcp_options().
in the event handler:

if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_CONNECTED){
xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
set_dhcp_options();
}

but no luck, same error.

the netif and wifi initialization is the same as example connect.

its only trigger the error:
Failed to set Vendor Class Identifier: ESP_ERR_ESP_NETIF_INVALID_PARAMS
Failed to set Vendor Class Identifier: ESP_ERR_ESP_NETIF_INVALID_PARAMS

david-cermak, 2025-02-28:
> in the event handler:

At which point do you register that event handler? Please make sure that you call `esp_event_handler_register()` after `esp_netif_create_default_wifi_sta()`.

Could you please share the initialization steps of your project? the snippet above doesn't include event handlers.