espressif/esp-idf #15363: uart_async_rxtxtasks 例程栈溢出

EwaneGao, 2025-02-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

测试uart_async_rxtxtasks

### What is the actual behavior?

编译后提示Total image size: 227572 bytes (.bin may be padded larger)，运行后向设备发送数据后重启
***ERROR*** A stack overflow in task uart_rx_task has been detected.

Backtrace: 0x40375b71:0x3fc9b130 0x4037a9b9:0x3fc9b150 0x4037b71e:0x3fc9b170 0x4037c917:0x3fc9b1f0 0x4037b7e4:0x3fc9b210 0x4037b7da:0x3c02a842 |<-CORRUPTED
--- 0x40375b71: panic_abort at E:/program/.espressif/v5.4/esp-idf/components/esp_system/panic.c:454
0x4037a9b9: esp_system_abort at E:/program/.espressif/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4037b71e: vApplicationStackOverflowHook at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
0x4037c917: vTaskSwitchContext at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x4037b7e4: _frxt_dispatch at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4037b7da: _frxt_int_exit at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

[image]

### Steps to reproduce.

1. 选择esp32-s3
2. 编译-内存过大
3. 发送12345678901234567890

### Debug Logs.

```plain
I (148285) TX_TASK: Wrote 11 bytes
I (150155) RX_TASK: Read 22 bytes: '12345678901234567890
'
I (150155) RX_TASK: 0x3fc9b6e8   31 32 33 34 35 36 37 38  39 30 31 32 33 34 35 36  |1234567890123456|
I (150155) RX_TASK: 0x3fc9b6f8   37 38 39 30 0d 0a                                 |7890..|

***ERROR*** A stack overflow in task uart_rx_task has been detected.

Backtrace: 0x40375b71:0x3fc9b130 0x4037a9b9:0x3fc9b150 0x4037b71e:0x3fc9b170 0x4037c917:0x3fc9b1f0 0x4037b7e4:0x3fc9b210 0x4037b7da:0x3c02a842 |<-CORRUPTED
--- 0x40375b71: panic_abort at E:/program/.espressif/v5.4/esp-idf/components/esp_system/panic.c:454
0x4037a9b9: esp_system_abort at E:/program/.espressif/v5.4/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4037b71e: vApplicationStackOverflowHook at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:563
0x4037c917: vTaskSwitchContext at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x4037b7e4: _frxt_dispatch at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4037b7da: _frxt_int_exit at E:/program/.espressif/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

ELF file SHA256: c166dd763

Rebooting...
```

suda-morris, 2025-02-10:
> ***ERROR*** A stack overflow in task uart_rx_task has been detected.

Does increase the stack size help?

https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/examples/peripherals/uart/uart_async_rxtxtasks/main/uart_async_rxtxtasks_main.c#L75

EwaneGao, 2025-02-13:
通过增大栈空间，可有效解决
` xTaskCreate(rx_task, "uart_rx_task", 1024 * 3, NULL, configMAX_PRIORITIES - 1, NULL); `

closed 2025-02-19 by espressif-bot