espressif/esp-idf #15365: Modbus Serial example not working as expected, only read 2/9 Characteristics

dzungpv, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

First, I tested both slave and master board with uart_echo_rs485 example here https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485 and they working fine.

And nex I try examples here [modbus](https://github.com/espressif/esp-idf/tree/master/examples/protocols/modbus/serial)
with two of my custom board using ESP32 with MAX13487EESA chip, and not using RTS pin. 
1 board with master and other with slave example. It not working as expected, it show may error and can **only read 2/9 Characteristics**.
Only Characteristic 0, Characteristic 1 the rest shows:
` read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).`

The log with master example:
```
I (495) MASTER_TEST: Start modbus test...
I (495) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (505) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.340000 (0x3fab851f) read successful.
E (515) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (515) MASTER_TEST: Characteristic #2 (Temperature_1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (535) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (545) MASTER_TEST: Characteristic #3 (Humidity_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (555) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (565) MASTER_TEST: Characteristic #4 (Temperature_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (585) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (585) MASTER_TEST: Characteristic #5 (Humidity_3) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (615) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (615) MASTER_TEST: Characteristic #6 (Test_regs) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (635) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (635) MASTER_TEST: Characteristic #7 (RelayP1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (655) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (665) MASTER_TEST: Characteristic #8 (RelayP2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (675) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (685) MASTER_TEST: Characteristic #9 (DiscreteInpP1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
I (1195) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 1.120000 (0x3f8f5c29) read successful.
I (1205) MASTER_TEST: Characteristic #1 Humidity_1 (%rH) value = 1.540000 (0x3fc51eb9) read successful.
E (1215) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1215) MASTER_TEST: Characteristic #2 (Temperature_1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1235) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1245) MASTER_TEST: Characteristic #3 (Humidity_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1255) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1265) MASTER_TEST: Characteristic #4 (Temperature_2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1275) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1285) MASTER_TEST: Characteristic #5 (Humidity_3) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1315) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1315) MASTER_TEST: Characteristic #6 (Test_regs) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1325) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1335) MASTER_TEST: Characteristic #7 (RelayP1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1355) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1355) MASTER_TEST: Characteristic #8 (RelayP2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).
E (1375) MB_CONTROLLER_MASTER: mbc_master_get_parameter(75): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (1385) MASTER_TEST: Characteristic #9 (DiscreteInpP1) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).

```

Logs with slave example:

```
I (343) SLAVE_TEST: Start modbus test...
I (4853) SLAVE_TEST: INPUT READ (4551320 us), ADDR:0, TYPE:8, INST_ADDR:0x3ffb2e64, SIZE:2
I (4863) SLAVE_TEST: HOLDING READ (4558143 us), ADDR:0, TYPE:2, INST_ADDR:0x3ffb30dc, SIZE:2
I (4873) SLAVE_TEST: INPUT READ (4567392 us), ADDR:2, TYPE:8, INST_ADDR:0x3ffb2e68, SIZE:2
I (4893) SLAVE_TEST: HOLDING READ (4590032 us), ADDR:2, TYPE:2, INST_ADDR:0x3ffb30e0, SIZE:2
I (4923) SLAVE_TEST: INPUT READ (4612502 us), ADDR:4, TYPE:8, INST_ADDR:0x3ffb2e6c, SIZE:2
I (4943) SLAVE_TEST: HOLDING READ (4635277 us), ADDR:4, TYPE:2, INST_ADDR:0x3ffb30e4, SIZE:2
I (4963) SLAVE_TEST: HOLDING READ (4657761 us), ADDR:10, TYPE:2, INST_ADDR:0x3ffb30f0, SIZE:58
I (4993) SLAVE_TEST: HOLDING WRITE (4688539 us), ADDR:10, TYPE:1, INST_ADDR:0x3ffb30f0, SIZE:58
[... 11 lines cut ...]
I (5723) SLAVE_TEST: COILS READ (5414150 us), ADDR:10, TYPE:32, INST_ADDR:0x3ffb2e61, SIZE:6
I (5743) SLAVE_TEST: DISCRETE READ (5436563 us): ADDR:2, TYPE:64, INST_ADDR:0x3ffb2e5c, SIZE:7
I (6253) SLAVE_TEST: INPUT READ (5950395 us), ADDR:0, TYPE:8, INST_ADDR:0x3ffb2e64, SIZE:2
I (6263) SLAVE_TEST: HOLDING READ (5956237 us), ADDR:0, TYPE:2, INST_ADDR:0x3ffb30dc, SIZE:2
I (6273) SLAVE_TEST: INPUT READ (5965659 us), ADDR:2, TYPE:8, INST_ADDR:0x3ffb2e68, SIZE:2
I (6293) SLAVE_TEST: HOLDING READ (5988581 us), ADDR:2, TYPE:2, INST_ADDR:0x3ffb30e0, SIZE:2
I (6313) SLAVE_TEST: INPUT READ (6011190 us), ADDR:4, TYPE:8, INST_ADDR:0x3ffb2e6c, SIZE:2
I (6343) SLAVE_TEST: HOLDING READ (6034104 us), ADDR:4, TYPE:2, INST_ADDR:0x3ffb30e4, SIZE:2
I (6363) SLAVE_TEST: HOLDING READ (6056796 us), ADDR:10, TYPE:2, INST_ADDR:0x3ffb30f0, SIZE:58
I (6383) SLAVE_TEST: COILS READ (6073931 us), ADDR:2, TYPE:32, INST_ADDR:0x3ffb2e60, SIZE:6
I (6403) SLAVE_TEST: COILS READ (6094208 us), ADDR:10, TYPE:32, INST_ADDR:0x3ffb2e61, SIZE:6
I (6423) SLAVE_TEST: DISCRETE READ (6116613 us): ADDR:2, TYPE:64, INST_ADDR:0x3ffb2e5c, SIZE:7
I (6933) SLAVE_TEST: INPUT READ (6630377 us), ADDR:0, TYPE:8, INST_ADDR:0x3ffb2e64, SIZE:2
I (6943) SLAVE_TEST: HOLDING READ (6636218 us), ADDR:0, TYPE:2, INST_ADDR:0x3ffb30dc, SIZE:2
I (6953) SLAVE_TEST: INPUT READ (6645641 us), ADDR:2, TYPE:8, INST_ADDR:0x3ffb2e68, SIZE:2
I (6973) SLAVE_TEST: HOLDING READ (6668493 us), ADDR:2, TYPE:2, INST_ADDR:0x3ffb30e0, SIZE:2
I (6993) SLAVE_TEST: INPUT READ (6691171 us), ADDR:4, TYPE:8, INST_ADDR:0x3ffb2e6c, SIZE:2
I (7023) SLAVE_TEST: HOLDING READ (6714085 us), ADDR:4, TYPE:2, INST_ADDR:0x3ffb30e4, SIZE:2
I (7043) SLAVE_TEST: HOLDING READ (6736777 us), ADDR:10, TYPE:2, INST_ADDR:0x3ffb30f0, SIZE:58
I (7063) SLAVE_TEST: COILS READ (6753913 us), ADDR:2, TYPE:32, INST_ADDR:0x3ffb2e60, SIZE:6
I (7083) SLAVE_TEST: COILS READ (6774182 us), ADDR:10, TYPE:32, INST_ADDR:0x3ffb2e61, SIZE:6
I (7103) SLAVE_TEST: DISCRETE READ (6796595 us): ADDR:2, TYPE:64, INST_ADDR:0x3ffb2e5c, SIZE:7
I (7613) SLAVE_TEST: INPUT READ (7310358 us), ADDR:0, TYPE:8, INST_ADDR:0x3ffb2e64, SIZE:2
I (7623) SLAVE_TEST: HOLDING READ (7316199 us), ADDR:0
```

dzungpv, 2025-02-10:
@alisitsyn Can you look into this?

alisitsyn, 2025-02-11:
@dzungpv,

Could you provide more information:
1. The logs for master and slave with debug verbosity in kconfig menu?
2. Which version of esp-idf you are using with the examples?

alisitsyn, 2025-02-12:
I guess you are using the latest esp-idf. This could be access feature de-synchronization  issue between stack and examples which happens only on latest esp-idf. As a simple fix for this issue please change [the line](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_slave/main/slave.c#L94) in the examples as below:
```
mb_register_area_descriptor_t reg_area = {0}; // Modbus register area descriptor structure
```
`idf.py fullclean; idf.py build;`
Please give me answers and feedback ASAP. Thanks.

dzungpv, 2025-02-13:
> [@dzungpv](https://github.com/dzungpv),
> 
> Could you provide more information:
> 
>     1. The logs for master and slave with debug verbosity in kconfig menu?
> 
>     2. Which version of esp-idf you are using with the examples?

It is IDF **v5.3.2-282-gcfea4f7c98**, I also apply fix for mode config here https://github.com/espressif/esp-idf/commit/8d2a1720640449d7e9b0fec6c1e49cd60ca301ef.

This is master logs: 

[master_mb_log.log](https://github.com/user-attachments/files/18779195/master_mb_log.log)

This is slave logs: 

[slave_mb_log.log](https://github.com/user-attachments/files/18779197/slave_mb_log.log)

I try to use "Modbus Poll" PC app to test the slave example and it can be read and write the registers, so no error in slave example. It may cause by the master, read write wrong address.

I made change in two examples:

```
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
CONFIG_MB_COMM_MODE_ASCII=n
CONFIG_MB_COMM_MODE_RTU=y
CONFIG_MB_UART_RXD=18
CONFIG_MB_UART_TXD=15
CONFIG_MB_UART_RTS=21
```

I also make change like you said to slave.c but it is the same.

```
mb_register_area_descriptor_t reg_area = {0}; // Modbus register area descriptor structure
```

dzungpv, 2025-02-13:
@alisitsyn It working now, it cause by may rainbow cable. After not see any receive data in the log, I try to change it and working now.
Thank you for the support.

[image]

closed 2025-02-13 by dzungpv

alisitsyn, 2025-02-13:
@dzungpv ,

Thank you for feedback. It is good to hear the issue is resolved.