espressif/esp-idf #15366: macfilter OT CLI calling issue

donaldkohcw, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have enabled MAC filtering (OPENTHREAD_CONFIG_MAC_FILTER_ENABLE = 1) in my OpenThread build, but the macfilter command always returns "Disabled", even after attempting to add addresses and configure filtering modes.

Example build :  \openthread\ot_cli release\5.4
Platform: ESP32-H2/ESP32-C6
IDF: [5.3.2]

Enabled via Menuconfig and checked "OPENTHREAD_CONFIG_MAC_FILTER_ENABLE = 1"

> macfilter

Address Mode: Disabled
RssIn List:
Done

> macfilter addr

Disabled
Done

> macfilter addr add 0XFEDCBA0987653421

Error 7: InvalidArgs

donaldkohcw, 2025-02-10:
correction it is on Platform: ESP32-C6 not -H2

zwx1995esp, 2025-02-11:
Hi @donaldkohcw  , please refer to the command [macfilter](https://github.com/openthread/openthread/tree/main/src/cli#macfilter) in the openthread official docs. You might need to set the [macfilter mode](https://github.com/openthread/openthread/tree/main/src/cli#macfilter-addr-allowlist) first.

closed 2025-02-11 by Alvin1Zhang