espressif/esp-idf #15368: ESP32-S3 Risc-V gpio interrupts

Mychaylo-evox, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I cannot implement any gpio interrupts, after enabling interrupts in the menuconfig, the macro CONFIG_ULP_RISCV_INTERRUPT_ENABLE is never used from the general sdkconfig.h file present in the build directory, the components/ulp/ulp_riscv/ulp_core/ulp_riscv_gpio.c, which has the interrupt registering functions, has a call to a different sdkconfig.h that is located in components/esp_system/test_eh_frame_parser and has 2 macros:
```
#define CONFIG_ESP_SYSTEM_USE_EH_FRAME 1
#define CONFIG_IDF_TARGET_X86 1
```
, but it doesn't have the CONFIG_ULP_RISCV_INTERRUPT_ENABLE macro.

In the general sdkconfig.h, CONFIG_ULP_RISCV_INTERRUPT_ENABLE is present and #defined as 1.

Also, I don't know if it's related, but in the components/riscv/interrupt.c, the RV_EXTERNAL_INT_COUNT in line 22 is not declared and i cannot find it anywhere that should be declared.

### What is the actual behavior?

To be able to access the functions that are blocked by the macro

### Steps to reproduce.

No significant step

### Build or installation Logs.

```plain

```

o-marshmallow, 2025-02-12:
Hello @Mychaylo-evox ,

To use interrupts on the RSICV-ULP (on the S3), you may want to take a look at this example: https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/interrupts

The `sdkconfig.h` is a file generated at compile time, it is based on the `sdkconfig` file itself generated by `menuconfig`. Its path after compilation will be `<your_project>/build/config/sdkconfig.h`.

Mychaylo-evox, 2025-02-12:
What about the undefined RV_EXTERNAL_INT_COUNT in the components/riscv/interrupt.c file? When including the file it appears with red squiggles. And when I look at the file, RV_EXTERNAL_INT_COUNT is not defined or implemented.

o-marshmallow, 2025-02-13:
@Mychaylo-evox  This file is not related to ULP RISC-V either, this file is for RISC-V targets (C3, C2, H2, C6, ...). The files you'll be interested in are in `esp-idf/components/ulp/` and `esp-components/ulp/ulp_riscv`, but again, the interrupt example should have all the information you need to setup interrupts on the ULP RISC-V core.

Mychaylo-evox, 2025-02-13:
If that file is related to RISC-V targets, then I assume that it's an error that it's being included by the `components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_interrupt.h`, which is included in `components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h`, which is used by the example you provided.

BitsForPeople, 2025-03-01:
I doubt that components/riscv/interrupt.c is included by any (ULP) header file.
Don't rely too much on VS Code's ability to resolve all includes and paths correctly, especially when editing ULP code in a project which to VS Code appears to be a main CPU application.

o-marshmallow, 2025-03-12:
Agreed with @BitsForPeople , do not rely on VSCode symbol resolution, it will point you to symbols with the same name without actually knowing whether it will be included in the build or not.

Closing this issue, since it's not an issue in ESP-IDF. As said above, please refer to this example for using interrupts on the ULP: https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/interrupts

closed 2025-03-12 by o-marshmallow