espressif/esp-idf #15369: Shared kconfig

malachib, 2025-02-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a set of components which wants to include SPI settings across them. Only one component will truly activate at a time, but all are available. I `rsource` them in so that I can use relative pathing to get to a parent folder (i.e. `../Kconfig.spi`).

In certain environments, this results in "INFO: Symbol ... defined in multiple locations"

Example project linked here [https://bitbucket.org/malachib/playgrou ... /PGESP-69/](https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-69/) exhibits the problem. You can see there are some IF guards there, but the INFO warning persists.

Please advise.

chegewara, 2025-02-10:
duplicate
https://github.com/espressif/esp-idf/issues/15242

Honza0297, 2025-02-11:
Hello @malachib ! 

First of all, thank you for reporting this issue!  This info statement is caused by recent update of esp-idf-kconfig, in which we started to check if some config options are defined multiple times (see https://github.com/espressif/esp-idf/issues/15242#issuecomment-2636133153 for detailed context).

How to get rid of these: in the near future, I'll introduce a way how to manually mark config options which should ignored for this check. I'll notify you once it will be officially released.

Note regarding IF statements and conditioned multiple definition in general: during the parse time, we cannot evaluate any condition (not speaking about general satisfiability of the conditions as an alternative - that can be done, but it is NP-complete problem in general = VERY time consuming) and thus we cannot check whether these conditions truly guarantee mutual exclusion.  To play it on the safe side, we decided to  report every case of multiple definition that will happen. Unfortunately, practice of using multiple definitions is more widespread than we anticipated, which is the reason we are implementing the aforementioned way how to manually turn this check off. 

I hope I answered the situation acceptably, if you have any more question, do not hesitate to ask and thank you for your patience!

malachib, 2025-02-11:
Thank you for the informative response!

A manual marking system would work just fine for me - my use case really need to make sure I'm only _truly_ including it once anyway.  I appreciate your time and effort on this!

Honza0297, 2025-02-21:
@malachib [latest version of esp-idf-kconfig](https://github.com/espressif/esp-idf-kconfig/releases/tag/v2.5.0) (supported by ESP-IDF v5.3 and higher) should provide a way fix this issue.  Please, re-run install script in ESP-IDF repository to ensure you have the latest esp-idf-kconfig version. 

If you want to manually mark config options as allowed to be set multiple times, add `# ignore: multiple-definition` pragma comment to the config definition (it is sufficient to put it only to one definition, but you can put the pragma to all the definition locations, that is up to you):

```
config FOO # ignore: multiple-definition 
    (...)
```

Please, let me know if any issues will arise. Thank you for your patience!

malachib, 2025-02-22:
Seems to do the trick.  I was thrown a little that merely re-installing existing ESP-IDF would work, but it did.  My attempt to over complicate things was only so successful :)

Honza0297, 2025-02-26:
I am happy it works now! This issue will be closed in a short time, but feel free to reopen or open a new one if something arise. Thanks again for the patience! :)

closed 2025-02-27 by Indastri