espressif/esp-idf #15370: Connection to WPA2-Enterprise PEAP WiFi fails after enabling performance optimization.

thomas201012, 2025-02-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

The device should connect successfully to the WPA2-Enterprise WiFi.

### What is the actual behavior?

The device fails to connect after attempting for some time.

### Steps to reproduce.

1.Use the example examples/wifi/wifi_enterprise.
2.Enable "Optimize for performance" in menuconfig.
3.Attempt to connect to a WPA2_ENT PEAP WiFi.
4.The connection fails, but other optimization methods succeed.

### Debug Logs.

```plain
I (26) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (27) boot: compile time Feb 11 2025 10:07:54
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (33) boot: efuse block revision: v1.3
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=1b3b0h (111536) map
I (123) esp_image: segment 1: paddr=0002b3d8 vaddr=3fc98a00 size=04868h ( 18536) load
I (127) esp_image: segment 2: paddr=0002fc48 vaddr=40374000 size=003d0h (   976) load
I (129) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=90cf4h (593140) map
I (242) esp_image: segment 4: paddr=000c0d1c vaddr=403743d0 size=14578h ( 83320) load
I (269) boot: Loaded app from partition at offset 0x10000
I (269) boot: Disabling RNG early entropy source...
I (281) cpu_start: Multicore app
I (290) cpu_start: Pro cpu start user code
I (290) cpu_start: cpu freq: 160000000 Hz
I (291) app_init: Application information:
I (293) app_init: Project name:     wifi_enterprise
I (299) app_init: App version:      v5.3.2
I (304) app_init: Compile time:     Feb 11 2025 10:08:08
I (310) app_init: ELF file SHA256:  b2f9983e7...
I (315) app_init: ESP-IDF:          v5.3.2
I (320) efuse_init: Min chip rev:     v0.0
I (324) efuse_init: Max chip rev:     v0.99 
I (329) efuse_init: Chip rev:         v0.2
I (334) heap_init: Initializing. RAM available for dynamic allocation:
I (341) heap_init: At 3FCA0FF8 len 00048718 (289 KiB): RAM
I (348) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (354) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (360) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (367) spi_flash: detected chip: gd
I (370) spi_flash: flash io: dio
W (374) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (388) sleep: Configure to isolate all GPIO pins in sleep state
I (394) sleep: Enable automatic switching of GPIO sleep configuration
I (402) main_task: Started on CPU0
I (412) main_task: Calling app_main()
I (442) pp: pp rom version: e7ae62f
I (442) net80211: net80211 rom version: e7ae62f
I (452) wifi:wifi driver task: 3fcab364, prio:23, stack:6656, core=0
I (452) wifi:wifi firmware version: b0fd6006b
I (462) wifi:wifi certification version: v7.0
I (462) wifi:config NVS flash: enabled
I (462) wifi:config nano formating: disabled
I (462) wifi:Init data frame dynamic rx buffer num: 32
I (462) wifi:Init static rx mgmt buffer num: 5
I (472) wifi:Init management short buffer num: 32
I (472) wifi:Init dynamic tx buffer num: 32
I (482) wifi:Init static tx FG buffer num: 2
I (482) wifi:Init static rx buffer size: 1600
I (492) wifi:Init static rx buffer num: 10
I (492) wifi:Init dynamic rx buffer num: 32
I (492) wifi_init: rx ba win: 6
I (502) wifi_init: accept mbox: 6
I (502) wifi_init: tcpip mbox: 32
I (502) wifi_init: udp mbox: 6
I (512) wifi_init: tcp mbox: 6
I (512) wifi_init: tcp tx win: 5760
I (522) wifi_init: tcp rx win: 5760
I (522) wifi_init: tcp mss: 1440
I (522) wifi_init: WiFi IRAM OP enabled
I (532) wifi_init: WiFi RX IRAM OP enabled
I (532) example: Setting WiFi configuration SSID test...
I (542) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (592) wifi:mode : sta (dc:da:0c:1c:53:78)
I (592) wifi:enable tsf
I (602) main_task: Returned from app_main()
I (3462) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (3462) wifi:state: init -> auth (0xb0)
I (3482) wifi:state: auth -> assoc (0x0)
I (3492) wifi:state: assoc -> run (0x10)
I (4602) example: ~~~~~~~~~~~
I (4602) example: IP:0.0.0.0
I (4602) example: MASK:0.0.0.0
I (4602) example: GW:0.0.0.0
I (4602) example: ~~~~~~~~~~~
I (6602) example: ~~~~~~~~~~~
I (6602) example: IP:0.0.0.0
I (6602) example: MASK:0.0.0.0
I (6602) example: GW:0.0.0.0
I (6602) example: ~~~~~~~~~~~
I (8602) example: ~~~~~~~~~~~
I (8602) example: IP:0.0.0.0
I (8602) example: MASK:0.0.0.0
I (8602) example: GW:0.0.0.0
I (8602) example: ~~~~~~~~~~~
I (10602) example: ~~~~~~~~~~~
I (10602) example: IP:0.0.0.0
I (10602) example: MASK:0.0.0.0
I (10602) example: GW:0.0.0.0
I (10602) example: ~~~~~~~~~~~
I (12602) example: ~~~~~~~~~~~
I (12602) example: IP:0.0.0.0
I (12602) example: MASK:0.0.0.0
I (12602) example: GW:0.0.0.0
I (12602) example: ~~~~~~~~~~~
I (14602) example: ~~~~~~~~~~~
I (14602) example: IP:0.0.0.0
I (14602) example: MASK:0.0.0.0
I (14602) example: GW:0.0.0.0
I (14602) example: ~~~~~~~~~~~
I (16602) example: ~~~~~~~~~~~
I (16602) example: IP:0.0.0.0
I (16602) example: MASK:0.0.0.0
I (16602) example: GW:0.0.0.0
I (16602) example: ~~~~~~~~~~~
I (18602) example: ~~~~~~~~~~~
I (18602) example: IP:0.0.0.0
I (18602) example: MASK:0.0.0.0
I (18602) example: GW:0.0.0.0
I (18602) example: ~~~~~~~~~~~
I (20602) example: ~~~~~~~~~~~
I (20602) example: IP:0.0.0.0
I (20602) example: MASK:0.0.0.0
I (20602) example: GW:0.0.0.0
I (20602) example: ~~~~~~~~~~~
I (22602) example: ~~~~~~~~~~~
I (22602) example: IP:0.0.0.0
I (22602) example: MASK:0.0.0.0
I (22602) example: GW:0.0.0.0
I (22602) example: ~~~~~~~~~~~
I (24602) example: ~~~~~~~~~~~
I (24602) example: IP:0.0.0.0
I (24602) example: MASK:0.0.0.0
I (24602) example: GW:0.0.0.0
I (24602) example: ~~~~~~~~~~~
I (26602) example: ~~~~~~~~~~~
I (26602) example: IP:0.0.0.0
I (26602) example: MASK:0.0.0.0
I (26602) example: GW:0.0.0.0
I (26602) example: ~~~~~~~~~~~
I (28602) example: ~~~~~~~~~~~
I (28602) example: IP:0.0.0.0
I (28602) example: MASK:0.0.0.0
I (28602) example: GW:0.0.0.0
I (28602) example: ~~~~~~~~~~~
I (30602) example: ~~~~~~~~~~~
I (30602) example: IP:0.0.0.0
I (30602) example: MASK:0.0.0.0
I (30602) example: GW:0.0.0.0
I (30602) example: ~~~~~~~~~~~
I (32602) example: ~~~~~~~~~~~
I (32602) example: IP:0.0.0.0
I (32602) example: MASK:0.0.0.0
I (32602) example: GW:0.0.0.0
I (32602) example: ~~~~~~~~~~~
I (33492) wifi:state: run -> init (0xcc00)
I (33492) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (34602) example: ~~~~~~~~~~~
I (34602) example: IP:0.0.0.0
I (34602) example: MASK:0.0.0.0
I (34602) example: GW:0.0.0.0
I (34602) example: ~~~~~~~~~~~
I (35912) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (35912) wifi:state: init -> auth (0xb0)
I (35922) wifi:state: auth -> assoc (0x0)
I (35932) wifi:state: assoc -> run (0x10)
```

### More Information.

In components/wpa_supplicant/esp_supplicant/src/esp_eap_client.c, within the wpa2_task function, the addresses of sm->wpa2_sig_cnt[e->sig] during read and increment operations in wpa2_post are inconsistent, causing abnormal data reads from wpa2_sig_cnt.

kapilkedawat, 2025-02-11:
Hi @thomas201012, could you please share your sdkconfig, I just tried after enabling both `CONFIG_COMPILER_OPTIMIZATION_PERF` and `CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF` and able to connect without any issues.

```
I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (27) boot: compile time Feb 11 2025 14:04:58
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=206f0h (132848) map
I (112) esp_image: segment 1: paddr=00030718 vaddr=3fc98a00 size=04800h ( 18432) load
I (116) esp_image: segment 2: paddr=00034f20 vaddr=40374000 size=0b0f8h ( 45304) load
I (127) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=94a50h (608848) map
I (234) esp_image: segment 4: paddr=000d4a78 vaddr=4037f0f8 size=09850h ( 38992) load
I (251) boot: Loaded app from partition at offset 0x10000
I (252) boot: Disabling RNG early entropy source...
I (262) cpu_start: Multicore app
I (271) cpu_start: Pro cpu start user code
I (271) cpu_start: cpu freq: 160000000 Hz
I (271) app_init: Application information:
I (271) app_init: Project name:     wifi_enterprise
I (276) app_init: App version:      v5.3.2-dirty
I (280) app_init: Compile time:     Feb 11 2025 14:04:29
I (285) app_init: ELF file SHA256:  35535b1e7...
I (289) app_init: ESP-IDF:          v5.3.2-dirty
I (294) efuse_init: Min chip rev:     v0.0
I (298) efuse_init: Max chip rev:     v0.99 
I (302) efuse_init: Chip rev:         v0.1
I (305) heap_init: Initializing. RAM available for dynamic allocation:
I (312) heap_init: At 3FCA0F90 len 00048780 (289 KiB): RAM
I (317) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (322) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (327) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (334) spi_flash: detected chip: generic
I (336) spi_flash: flash io: dio
W (339) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (352) sleep: Configure to isolate all GPIO pins in sleep state
I (357) sleep: Enable automatic switching of GPIO sleep configuration
I (364) main_task: Started on CPU0
I (384) main_task: Calling app_main()
I (394) pp: pp rom version: e7ae62f
I (394) net80211: net80211 rom version: e7ae62f
I (404) wifi:wifi driver task: 3fcab130, prio:23, stack:6656, core=0
I (404) wifi:wifi firmware version: b0fd6006b
I (404) wifi:wifi certification version: v7.0
I (404) wifi:config NVS flash: enabled
I (414) wifi:config nano formating: disabled
I (414) wifi:Init data frame dynamic rx buffer num: 32
I (414) wifi:Init static rx mgmt buffer num: 5
I (424) wifi:Init management short buffer num: 32
I (424) wifi:Init dynamic tx buffer num: 32
I (434) wifi:Init static tx FG buffer num: 2
I (434) wifi:Init static rx buffer size: 1600
I (444) wifi:Init static rx buffer num: 10
I (444) wifi:Init dynamic rx buffer num: 32
I (444) wifi_init: rx ba win: 6
I (454) wifi_init: accept mbox: 6
I (454) wifi_init: tcpip mbox: 32
I (454) wifi_init: udp mbox: 6
I (454) wifi_init: tcp mbox: 6
I (464) wifi_init: tcp tx win: 5760
I (464) wifi_init: tcp rx win: 5760
I (464) wifi_init: tcp mss: 1440
I (474) wifi_init: WiFi IRAM OP enabled
I (474) wifi_init: WiFi RX IRAM OP enabled
I (474) example: Setting WiFi configuration SSID wpa2_enterprise...
I (484) wpa: WiFi Enterprise enable
I (484) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (644) wifi:mode : sta (7c:df:a1:e2:9e:b8)
I (644) wifi:enable tsf
I (654) main_task: Returned from app_main()
I (1754) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (1754) wifi:state: init -> auth (0xb0)
I (1784) wifi:state: auth -> assoc (0x0)
I (1804) wifi:state: assoc -> run (0x10)
I (1804) wpa: wifi_task prio:7, stack:6656
I (2014) wpa: len=5 not available in input
I (2084) wpa: SSL: Need 1416 bytes more input data
I (2124) wpa: SSL: Need 422 bytes more input data
I (2454) wpa: len=5 not available in input
I (2484) wpa: application data is null, adding one byte for ack
I (2574) wpa: EAP-MSCHAPV2: Authentication succeeded
I (2594) wpa: EAP-TLV: TLV Result - Success - EAP-TLV/Phase2 Completed
I (2614) wpa: >>>>>EAP FINISH
I (2704) wifi:connected with wpa2_enterprise, aid = 1, channel 6, BW20, bssid = da:21:f9:31:2f:b0
I (2704) wifi:security: WPA2-ENT, phy: bgn, rssi: -35
I (2704) wifi:pm start, type: 1

I (2704) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (2714) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (2854) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (3724) esp_netif_handlers: sta ip: 192.168.50.113, mask: 255.255.255.0, gw: 192.168.50.1
I (4654) example: ~~~~~~~~~~~
I (4654) example: IP:192.168.50.113
I (4654) example: MASK:255.255.255.0
I (4654) example: GW:192.168.50.1
I (4654) example: ~~~~~~~~~~~
I (6654) example: ~~~~~~~~~~~
I (6654) example: IP:192.168.50.113
I (6654) example: MASK:255.255.255.0
I (6654) example: GW:192.168.50.1
I (6654) example: ~~~~~~~~~~~
I (8654) example: ~~~~~~~~~~~
```

thomas201012, 2025-02-11:
> Hi [@thomas201012](https://github.com/thomas201012), could you please share your sdkconfig, I just tried after enabling both `CONFIG_COMPILER_OPTIMIZATION_PERF` and `CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF` and able to connect without any issues.您好，您能否分享您的 sdkconfig，我刚刚在启用两者 `CONFIG_COMPILER_OPTIMIZATION_PERF` 后尝试了一下， `CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF` 并且能够连接而没有任何问题。
> 
> ```
> I (27) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
> I (27) boot: compile time Feb 11 2025 14:04:58
> I (27) boot: Multicore bootloader
> I (28) boot: chip revision: v0.1
> I (30) boot: efuse block revision: v1.2
> I (34) boot.esp32s3: Boot SPI Speed : 80MHz
> I (38) boot.esp32s3: SPI Mode       : DIO
> I (42) boot.esp32s3: SPI Flash Size : 2MB
> I (45) boot: Enabling RNG early entropy source...
> I (50) boot: Partition Table:
> I (52) boot: ## Label            Usage          Type ST Offset   Length
> I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (72) boot:  2 factory          factory app      00 00 00010000 00100000
> I (78) boot: End of partition table
> I (82) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=206f0h (132848) map
> I (112) esp_image: segment 1: paddr=00030718 vaddr=3fc98a00 size=04800h ( 18432) load
> I (116) esp_image: segment 2: paddr=00034f20 vaddr=40374000 size=0b0f8h ( 45304) load
> I (127) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=94a50h (608848) map
> I (234) esp_image: segment 4: paddr=000d4a78 vaddr=4037f0f8 size=09850h ( 38992) load
> I (251) boot: Loaded app from partition at offset 0x10000
> I (252) boot: Disabling RNG early entropy source...
> I (262) cpu_start: Multicore app
> I (271) cpu_start: Pro cpu start user code
> I (271) cpu_start: cpu freq: 160000000 Hz
> I (271) app_init: Application information:
> I (271) app_init: Project name:     wifi_enterprise
> I (276) app_init: App version:      v5.3.2-dirty
> I (280) app_init: Compile time:     Feb 11 2025 14:04:29
> I (285) app_init: ELF file SHA256:  35535b1e7...
> I (289) app_init: ESP-IDF:          v5.3.2-dirty
> I (294) efuse_init: Min chip rev:     v0.0
> I (298) efuse_init: Max chip rev:     v0.99 
> I (302) efuse_init: Chip rev:         v0.1
> I (305) heap_init: Initializing. RAM available for dynamic allocation:
> I (312) heap_init: At 3FCA0F90 len 00048780 (289 KiB): RAM
> I (317) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
> I (322) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (327) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
> I (334) spi_flash: detected chip: generic
> I (336) spi_flash: flash io: dio
> W (339) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> I (352) sleep: Configure to isolate all GPIO pins in sleep state
> I (357) sleep: Enable automatic switching of GPIO sleep configuration
> I (364) main_task: Started on CPU0
> I (384) main_task: Calling app_main()
> I (394) pp: pp rom version: e7ae62f
> I (394) net80211: net80211 rom version: e7ae62f
> I (404) wifi:wifi driver task: 3fcab130, prio:23, stack:6656, core=0
> I (404) wifi:wifi firmware version: b0fd6006b
> I (404) wifi:wifi certification version: v7.0
> I (404) wifi:config NVS flash: enabled
> I (414) wifi:config nano formating: disabled
> I (414) wifi:Init data frame dynamic rx buffer num: 32
> I (414) wifi:Init static rx mgmt buffer num: 5
> I (424) wifi:Init management short buffer num: 32
> I (424) wifi:Init dynamic tx buffer num: 32
> I (434) wifi:Init static tx FG buffer num: 2
> I (434) wifi:Init static rx buffer size: 1600
> I (444) wifi:Init static rx buffer num: 10
> I (444) wifi:Init dynamic rx buffer num: 32
> I (444) wifi_init: rx ba win: 6
> I (454) wifi_init: accept mbox: 6
> I (454) wifi_init: tcpip mbox: 32
> I (454) wifi_init: udp mbox: 6
> I (454) wifi_init: tcp mbox: 6
> I (464) wifi_init: tcp tx win: 5760
[... 20 lines cut ...]
> I (2574) wpa: EAP-MSCHAPV2: Authentication succeeded
> I (2594) wpa: EAP-TLV: TLV Result - Success - EAP-TLV/Phase2 Completed
> I (2614) wpa: >>>>>EAP FINISH
> I (2704) wifi:connected with wpa2_enterprise, aid = 1, channel 6, BW20, bssid = da:21:f9:31:2f:b0
> I (2704) wifi:security: WPA2-ENT, phy: bgn, rssi: -35
> I (2704) wifi:pm start, type: 1
> 
> I (2704) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
> I (2714) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
> I (2854) wifi:AP's beacon interval = 102400 us, DTIM period = 3
> I (3724) esp_netif_handlers: sta ip: 192.168.50.113, mask: 255.255.255.0, gw: 192.168.50.1
> I (4654) example: ~~~~~~~~~~~
> I (4654) example: IP:192.168.50.113
> I (4654) example: MASK:255.255.255.0
> I (4654) example: GW:192.168.50.1
> I (4654) example: ~~~~~~~~~~~
> I (6654) example: ~~~~~~~~~~~
> I (6654) example: IP:192.168.50.113
> I (6654) example: MASK:255.255.255.0
> I (6654) example: GW:192.168.50.1
> I (6654) example: ~~~~~~~~~~~
> I (8654) example: ~~~~~~~~~~~
> ```

Please do not enable CONFIG_ESP_WIFI_DEBUG_PRINT. When CONFIG_ESP_WIFI_DEBUG_PRINT is enabled during testing, the connection works fine. However, when it is disabled, the connection fails.

thomas201012, 2025-02-11:
[sdkconfig.txt](https://github.com/user-attachments/files/18747748/sdkconfig.txt)

AxelLin, 2025-02-11:
> Please do not enable CONFIG_ESP_WIFI_DEBUG_PRINT. When CONFIG_ESP_WIFI_DEBUG_PRINT is enabled during testing, the connection works fine. However, when it is disabled, the connection fails.

I have hit similar issue when test with WPA2-enterprise TTLS using v5.2 branch, and it works if enable CONFIG_ESP_WIFI_DEBUG_PRINT.

kapilkedawat, 2025-02-12:
@thomas201012 could you please try with this patch once?

```
--- a/components/wpa_supplicant/esp_supplicant/src/esp_eap_client.c
+++ b/components/wpa_supplicant/esp_supplicant/src/esp_eap_client.c
@@ -243,6 +243,7 @@ void wpa2_task(void *pvParameters)
         } else {
             if (s_wifi_wpa2_sync_sem) {
                 wpa_printf(MSG_DEBUG, "EAP: wifi->EAP api completed");
+               __asm__ volatile ("" ::: "memory");
                 os_semphr_give(s_wifi_wpa2_sync_sem);
             } else {
                 wpa_printf(MSG_ERROR, "EAP: null wifi->EAP sync sem");
```

thomas201012, 2025-02-12:
> __asm__ volatile ("" ::: "memory");

After applying this patch, it works correctly. Will this patch be included in the next version 5.3.3?

kapilkedawat, 2025-02-13:
@thomas201012 @KaeLL Seems like this was just masking the actual issue, this is a stack corruption and will be fixed properly in next releases. You can use this patch meanwhile, please update if you face any issues.

```
--- a/components/wpa_supplicant/esp_supplicant/src/esp_eap_client.c
+++ b/components/wpa_supplicant/esp_supplicant/src/esp_eap_client.c
@@ -725,7 +725,7 @@ static int eap_peer_sm_init(void)
 
     gEapSm = sm;
 #ifdef USE_WPA2_TASK
-    s_wpa2_queue = os_queue_create(SIG_WPA2_MAX, sizeof(ETSEvent));
+    s_wpa2_queue = os_queue_create(SIG_WPA2_MAX, sizeof(s_wpa2_queue));
     ret = os_task_create(wpa2_task, "wpa2T", WPA2_TASK_STACK_SIZE, NULL, WPA2_TASK_PRIORITY, &s_wpa2_task_hdl);
     if (ret != TRUE) {
         wpa_printf(MSG_ERROR, "wps enable: failed to create task");
```

AxelLin, 2025-02-13:
> [@thomas201012](https://github.com/thomas201012) [@KaeLL](https://github.com/KaeLL) Seems like this was just masking the actual issue, this is a stack corruption and will be fixed properly in next [...]

@kapilkedawat 
Are you sure this change is correct? Can you double check?

kapilkedawat, 2025-02-13:
@AxelLin we are passing address of `ETSEvent`, please try out the change once. Also as I said this is to unblock @thomas201012 , we will be fixing this properly and change will be available in all the IDF release branches.

AxelLin, 2025-02-19:
@thomas201012 
The fix is on github now.
Can you test again to confirm if this issue is fixed?

IMHO, the fix does not explain why this issue only happens when enabling performance optimization.