espressif/esp-idf #15373: Using memcpy with target_add_binary_data(x) produces corrupted read

scholayil, 2025-02-11:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. target_add_binary_data(${CMAKE_PROJECT_NAME}.elf "${CMAKE_CURRENT_LIST_DIR}/certs/root_cert_auth.crt" TEXT)
2. extern const char root_cert_start [] asm("_binary_root_cert_auth_crt_start");
3. char rca[CERT_BUFFER_LEN];
4. (void)memcpy(rca, root_cert_start , size); 
5. Results in tls parse error when rca is passed to aws mqtt broker, no error when root_cert_start is passed directly.

E (41289) esp-tls-mbedtls: mbedtls_x509_crt_parse of public cert returned -0x2180
E (41289) esp-tls-mbedtls: Failed to set client pki context
E (41299) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
E (41309) esp-tls: create_ssl_handle failed
E (41309) esp-tls: Failed to open new connection

6. This happens consistently on certain builds and no errors on other builds. This leads me to suspect flash cache clearing or TEXT location across 32K boundary. 

What is a good way to avoid it?

Thanks

AdityaHPatwardhan, 2025-02-12:
@scholayil I had couple of questions
1) Did you print the content on console before providing it to the crt_parse API? You could print it as-is or in hex format.
2) how did you calculate the size for the memcpy operation? While doing that, did you consider the requirement that the crt_parse API needs certificate length to include the NULL byte at the end and thus size = size + 1 ?

It is interesting that this happens on certain builds and not on others. Is it possible to share some information about builds? What is the difference between them?

scholayil, 2025-02-12:
@AdityaHPatwardhan here is my reply to your questions,

1. I printed out the the cert and diffed it with original cert. They match. 
2. size + 1 is used. 
3. The difference between build is I clean before the build. With some builds , I am unable to create the error, most builds, I get the error in the original post.

AdityaHPatwardhan, 2025-02-13:
@scholayil,

If the certificate matches as you say, then there shouldn't be a reason for the certificate copied into the buffer to fail when the certificate stored in the text section to pass.

Is it possible for you to produce a small example that is able to reproduce the issue?

FYI: The same approach is already used in our [examples](https://github.com/espressif/esp-idf/blob/c5865270b50529cd32353f588d8a917d89f3dba4/examples/protocols/mqtt/ssl_mutual_auth/CMakeLists.txt#L11) and we have not faced any issue.

scholayil, 2025-02-13:
@AdityaHPatwardhan We have couple of released products with different IDF versions in production already. I started with your examples for those. They are ok. We have also now established that memcpy that I put in the issue name is not a real problem either. 
 
I also found that with this new project, the same cert work ok all the time if I reset the device (step 4. below) when it gets cert error during cloud provisioning(step 3). Costs some time to restart and connect to aws.
 
1. Use wifi mgr to wifi provision via BLE. 
2. Reset
3. Connect to mqtt and cloud provision device on aws. 
4. Restart device, mqtt connects ok to aws.
5. Start BLE central , scan for peripherals at the same time. 

Typically have 27K heap free with idf v 5.4 (38K with idf 5.3.1) once mqtt connects ok.
I also tried not starting ble central at all, that leaves me with 100K free heap and still get same cert parse error during provisioning step 3.

scholayil, 2025-02-17:
Setting the cert/key size in Mqtt_Context_t after step 3 solved the issue. Thanks all.

closed 2025-02-17 by scholayil