espressif/esp-idf #15374: ESP32-P4 v1.0 Silicon idf 5.5 (master) i2c communications issue.

csalinasonline, 2025-02-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32-P4 v1.0 Silicon idf 5.5 (master) i2c communications issue.

General issue report
I am using esp-idf 5.5 via master branch (https://github.com/espressif/esp-idf.git).

I have new ESP32-P4 v1.0 Silicon.
Before I had v0.1 Silicon. Working on a prototype that mirrors most of the hardware on ESP32-P4-eval board.
I am also using example project 'i2c_basic' in sdk (esp-idf/examples/peripherals/i2c/i2c_basic)

Since changing to ESP32-P4 v1.0 Silicon I have issues with i2c communications this new chip.
I get this error:
I (104) example: I2C initialized successfully
E (204) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (264) i2c.master: clear bus failed.
E (264) i2c.master: i2c_master_transmit_receive(1227): I2C transaction failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4000be60
--- 0x4000be60: app_main at /home/xxx/esp/esp-idf-master-xxx/examples/peripherals/i2c/i2c_basic/main/i2c_basic_example_main.c:84 (discriminator 1)

But ESP32-P4 v0.1 Silicon works fine?
I (94) main_task: Calling app_main()
I (94) example: I2C initialized successfully
I (94) example: WHO_AM_I = 0
I (94) example: I2C de-initialized successfully

Please look at my attached log files, old (p4_Silicon_v0.1_i2c_basic_debug_log.txt) vs new (p4_Silicon_v1.0_i2c_basic_debug_log.txt) silicon log, and sdkconfig below.

Please advise.

[p4_Silicon_v0.1_i2c_basic_debug_log.txt](https://github.com/user-attachments/files/18760500/p4_Silicon_v0.1_i2c_basic_debug_log.txt)

[p4_Silicon_v1.0_i2c_basic_debug_log.txt](https://github.com/user-attachments/files/18760501/p4_Silicon_v1.0_i2c_basic_debug_log.txt)

[sdkconfig.txt](https://github.com/user-attachments/files/18760538/sdkconfig.txt)

mythbuster5, 2025-02-12:
It works fine on my side
```
I (204) efuse_init: Chip rev:         v1.0
...
I (299) example: WHO_AM_I = FF
I (299) example: I2C de-initialized successfully
I (299) main_task: Returned from app_main()
```

So more information would be helpful. 1). Did you try any other i2c devices? 2) Could I see any online logic wave on your i2c device on v1.0? (important) 3) Did pull-up be applied properly?

mythbuster5, 2025-02-12:
And I go through the code. The potential thing might cause this issue is the last parameter of `i2c_master_transmit(dev_handle, write_buf, sizeof(write_buf), I2C_MASTER_TIMEOUT_MS / portTICK_PERIOD_MS);` Try to increase it, just try to make it largest as `-1` : `i2c_master_transmit(.., .., .., -1)` and see if the issue will gone.

csalinasonline, 2025-02-12:
@mythbuster5 thanks for suggestions. I will apply and report back to you soon.

csalinasonline, 2025-02-13:
@mythbuster5 I found the issue on our side. V1.0 Silicon and idf5.5 are good. Seems the 3.3V power on new board (P4 v1.0) vs old board (P4 v0.1) was not functioning well. Fixed it. So i2c working great! Thanks for confirming new v1.0 Silicon and idf5.5 i2c.

mythbuster5, 2025-02-14:
Great! I will close this issue.

closed 2025-02-14 by mythbuster5