espressif/esp-idf #15376: i2c_master.c writes spurious warning when internal pull-ups are disabled

bdpdx, 2025-02-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

all

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

all

### Power Supply used.

USB

### What is the expected behavior?

In v5.4 i2c_master.c lines 957-959 is this code:

`if (i2c_master->base->pull_up_enable == false) {
        ESP_LOGW(TAG, "Please check pull-up resistances whether be connected properly. Otherwise unexpected behavior would happen. For more detailed information, please read docs");
    }`

Notice the code prints a warning to the console if internal pull-ups are disabled. This is NOT desirable as it leads one to think something is wrong with their configuration when using external pull-ups.

I understand this is here as a precaution, but it's unnecessary as anyone using the i2c master code should have at least a basic understanding of how pull-ups should be connected.

I've seen numerous reports online where people report this error and think there's a problem.

Best would be to remove this warning entirely, or in the worst case add some Kconfig setting to allow us to suppress it.

### What is the actual behavior?

see above report

### Steps to reproduce.

when configuring i2c_master_bus_config_t, set flags.enable_internal_pullup = false.

### Debug Logs.

```plain
n/a, there is just a warning printed as described above.
```

suda-morris, 2025-02-13:
As in https://github.com/espressif/esp-idf/pull/13963

KaeLL, 2025-02-13:
@suda-morris #12325 ?

bdpdx, 2025-02-13:
@suda-morris Yes, this is the same issue at #13963, although the title for that ticket is very broad. I'll close this issue as a duplicate.

closed 2025-02-13 by bdpdx