espressif/esp-idf #15379: BLE-HID-Device Re-Connection not working with the esp_hid_host example

turgu1, 2025-02-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ProS3 ESP32-S3 Dev board

### Power Supply used.

USB

### What is the expected behavior?

Expected the keyboard to reconnect when the ESP32s3 reboots or when the keyboard is back On and advertising it's request to reconnect.

### What is the actual behavior?

Using a BLE Keyboard (Logitech MX Keys Mini), was able to do pairing without any change to the esp_hid_host example. As the ESP32s3 reboot and the keyboard is advertising to reconnect, we can see the keyboard address in the scan's list, but nothing happen.

In the log, the device address corresponding to the keyboard is d5:97:39:9a:a9:db.

### Steps to reproduce.

1. idf.py set-target esp32s3
2. idf.py build flash
3. put keyboard in pairing mode
4. idf.py monitor
5. Enter the requested pairing number on the keyboard
6. (pairing successful) press some keys to see the connection is working
7. Leave the monitoring
8. idf.py monitor (rebooting the esp32s3)
9. press a key on the keyboard to get advertising for re-connection before the start of the scan process
10. (no re-connection occurs)

### Debug Logs.

```plain
turgu1@phobos:~/Dev/esp_hid_host$ idf.py --port /dev/ttyACM0  monitor
Executing action: monitor
Running idf_monitor in directory /home/turgu1/Dev/esp_hid_host
Executing "/home/turgu1/.espressif/python_env/idf5.4_py3.13_env/bin/python /home/turgu1/esp/esp-idf-v5.4/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/turgu1/Dev/esp_hid_host/build/esp_hid_host.elf -m '/home/turgu1/.espressif/python_env/idf5.4_py3.13_env/bin/python' '/home/turgu1/esp/esp-idf-v5.4/tools/idf.py' '--port' '/dev/ttyACM0'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037e846
[... 199 lines cut ...]
I (484) main_task: Returned from app_main()
BLE: 41:81:f6:04:87:7f, RSSI: -59, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 62:b1:07:ca:dc:6b, RSSI: -63, UUID: 0xfeb8, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM', NAME: 'Meta Quest 3'
BLE: 5e:d9:bb:ef:cb:8c, RSSI: -50, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: d5:d2:18:fa:ca:4b, RSSI: -68, UUID: 0x1523, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM', NAME: 'Sol Page Turner'
BLE: 53:d3:ed:5c:e5:ee, RSSI: -48, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 74:46:b3:70:c1:e2, RSSI: -94, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'PUBLIC'
BLE: d6:d2:d2:92:8e:a1, RSSI: -68, UUID: 0x1523, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM', NAME: 'Sol Page Turner'
BLE: c8:9b:d7:a5:0a:38, RSSI: -75, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM', NAME: 'Luxe'
BLE: d5:97:39:9a:a9:db, RSSI: -77, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: f9:60:2b:0a:ba:73, RSSI: -50, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: e5:08:cf:b4:a6:6f, RSSI: -72, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 80:8a:bd:23:b2:0f, RSSI: -90, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'PUBLIC'
BLE: f1:09:97:a4:eb:5e, RSSI: -55, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'RANDOM'
BLE: 80:8a:bd:23:b2:0f, RSSI: -87, UUID: 0x0000, APPEARANCE: 0x0000, ADDR_TYPE: 'PUBLIC', NAME: '[TV] Samsung Q50AA 32 TV'
I (5484) ESP_HIDH_DEMO: SCAN: 0 results

turgu1@phobos:~/Dev/esp_hid_host$
```

### More Information.

Recompiling for the ESP32 and plugging an appropriate ESP32 Dev board, the same behavior is shown on an ESP32 with the same keyboard. Using a BT keyboard (Logitech K380) everything works as expected. So the BT portion is working well, but not the BLE portion.

I've tried to apply the patch shown in Issue #11067. As expected, it didn't resolved the problem. As you can see in the log, when the ESP32s3 reboots, the advertising for the keyboard (d5:97:39:9a:a9:db) show an appearance of 0x0000 because it tries to reconnect, not pairing. So the SCAN result show that 0 entries was discovered.

esp-cjh, 2025-02-14:
@turgu1 The patch you apply is for HID host to reconnect. After disconnection, the HID host will restart scanning and attempt to reconnect. Based on the logs you provided, the failure to reconnect is likely because the HID device did not start advertising again, so the HID host could not detect it.

esp-zhp, 2025-02-14:
You can use another device to scan and check whether your HID Device is broadcasting.  

Additionally, please note that the **HID host example** scans for **5 seconds** by default and then stops scanning. This duration is defined by the `SCAN_DURATION_SECONDS` macro.  

For more details, refer to the code in:  
`esp-idf/examples/bluetooth/esp_hid_host/main/esp_hid_host_main.c`  

The scanning function is:  
```c
esp_hid_scan(SCAN_DURATION_SECONDS, &results_len, &results);
```

esp-zhp, 2025-02-14:
Make sure that the **HID Device** is broadcasting during the scanning period.

turgu1, 2025-02-14:
- The problem is not with the K380 as this one is working perfectly (it is a classic BT device, not BLE). No need to call the bt scan method to have it reconnect.
- the problem is with the MX Keys Mini. It does advertise, but because it was already paired, the UUiD is set to 0x0000 instead of 0x1812. You can see that in the log I provided.
- this keyboard works perfectly with Windows, Android and IOS devices

turgu1, 2025-02-14:
Iḿ using the bluedroid, not nimble as I need to be able to use BT and BLE keyboards in the same app.

turgu1, 2025-02-14:
Another important aspect for the end-user: reconnecting a device must be almost instantaneous. Waiting for a scan that would take more than a very few seconds wouldn't be acceptable. The BLE reconnect must be working as fast as it is for the BT parts (that is working very well).

turgu1, 2025-02-19:
Any idea to help me get it works?

turgu1, 2025-08-14:
Today, I was able to get the patched example compiled and uploaded to a ESP32s3. The patch, for what I understand about it, allows the ESP32s3 to be rebooted and reconnect quickly to the keyboard without  requiring a pairing phase. That seems to work well as long as the keyboard remains ON all the time.

When the keyboard is then turned OFF then ON, I expect the keyboard to be recognized quickly without requiring a new pairing, but this doesn’t work (that works well when connecting the keyboard with a Linux or Windows. Computer). I don’t see any log message appearing when monitoring. The keyboard is then probably interacting behind the scene with the ESP32s3 to regain proper credentials, and it is now using a different MAC address (the last nibble of the address is incremented by 1). The example code doesn’t show any processing and typing on the keyboard has no effect.

Can you confirm me if the different MAC address is related to some privacy mode supported by the ESP-IDF? If this is the case, is there anyway we can select the kind of privacy mode the ESP32s3 will use to configure the interaction with the keyboard so the keyboard would use the same MAC address as per the pairing? I’m not that familiar with these aspects of the BLE protocols so forgive me if I’m wrong.

turgu1, 2025-08-16:
I thing the patch resolve the initial issue. Thanks a lot! 

There is still the problem of the device that get a device address change when it resets or when the power is toggled. This is another issue that I'm now trying to resolve on my side for now. As stated in my preceding message, I will look at the BLE protocols if it's possible to get a fix random device address (MAC) between device resets.

Thanks again!