espressif/esp-idf #15380: OTA update with secure boot and flash encryption - guru meditation in `esp_ota_set_boot_partition`

PetyM, 2025-02-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4, v5.3.2, v5.2.3

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I am trying to achieve firmware update on device with secure boot and flash encryption enabled. Update data is sent to device from webserver. Device confirms sucessfull update to webserver and reboots to new version.

### What is the actual behavior?

Update is sucessfully written to OTA partition, but guru meditation occurs in `esp_ota_set_boot_partition`. Device then restarts and boots to new version of firmware. As FW crashes update confirmation cannot be sent back to websever.

### Steps to reproduce.

[sdkconfig](https://github.com/user-attachments/files/18768935/sdkconfig.txt) (sorry, I could not figure how to upload without adding .txt)
[partitions.csv](https://github.com/user-attachments/files/18768929/partitions.csv)

Update is recieved from webserver here:
```cpp
esp_err_t Protocols::UriHandlers::postWebUpdateHandler(httpd_req_t *req)
{
    constexpr size_t MAX_SIZE = 1024;
    static std::span<uint8_t, MAX_SIZE> dataBuffer = std::span<uint8_t, MAX_SIZE>(reinterpret_cast<uint8_t *>(heap_caps_malloc(MAX_SIZE, MALLOC_CAP_SPIRAM)), MAX_SIZE);
    size_t alredyReceivedDataLen = 0;
    int readBytes;
    auto updater = GlobalInstances::UpdaterInstance::instance().getPartialUpdater();

    allowCORS(req);
    setJSONResponse(req);
    if (!checkAuthentication(req))
    {
        return httpd_resp_send_err(req, HTTPD_401_UNAUTHORIZED, "{}");
    }

    while (req->content_len > alredyReceivedDataLen)
    {
        readBytes = httpd_req_recv(req, reinterpret_cast<char *>(dataBuffer.data()), dataBuffer.size_bytes());
        if (readBytes <= 0)
        {
            return httpd_resp_send_500(req);
[... 290 lines cut ...]
I (992) app_init: App version:      3.3.76
I (996) app_init: ELF file SHA256:  129cdf5ce...
I (1000) app_init: ESP-IDF:          v5.4
I (1004) efuse_init: Min chip rev:     v3.0
I (1008) efuse_init: Max chip rev:     v3.99 
I (1012) efuse_init: Chip rev:         v3.0
I (1016) heap_init: Initializing. RAM available for dynamic allocation:
I (1022) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1027) heap_init: At 3FFBFF38 len 000200C8 (128 KiB): DRAM
I (1033) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1038) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1044) heap_init: At 4009B3F4 len 00004C0C (19 KiB): IRAM
I (1049) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1056) spi_flash: detected chip: gd
I (1059) spi_flash: flash io: qio
W (1062) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (1074) nvs_sec_provider: NVS Encryption - Registering Flash encryption-based scheme...
I (00:05:37.073) main_task: Started on CPU0
I (1102) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (00:05:37.084) main_task: Calling app_main()
I (00:05:37.091) MAIN: --------------------------------------------------------
I (00:05:37.093) MAIN: Version 3.3.76 started
I (00:05:37.097) MAIN: --------------------------------------------------------
I (00:05:37.192) nvs: NVS partition "nvs" is encrypted.
```

mahavirj, 2025-02-24:
@PetyM 

Is it possible for you to share minimal application using any of the IDF security examples that can recreate this issue?

deldering-momo, 2025-02-25:
I'm experiencing the same issue. This is the traceback for the Guru error that I get:
```
0x400f61a5: spi_flash_op_unlock at /COMPONENT_SPI_FLASH_DIR/cache_utils.c:112
0x40084f35: s_verify_write at /COMPONENT_SPI_FLASH_DIR/esp_flash_api.c:976
0x40085332: esp_flash_write_encrypted at /COMPONENT_SPI_FLASH_DIR/esp_flash_api.c:1380
0x400f48ea: esp_partition_write at /COMPONENT_ESP_PARTITION_DIR/partition_target.c:85
0x400f3d7f: rewrite_ota_seq at /COMPONENT_APP_UPDATE_DIR/esp_ota_ops.c:387
0x400f40f1: rewrite_ota_seq at /COMPONENT_APP_UPDATE_DIR/esp_ota_ops.c:376
 (inlined by) esp_rewrite_ota_data at /COMPONENT_APP_UPDATE_DIR/esp_ota_ops.c:445
 (inlined by) esp_ota_set_boot_partition at /COMPONENT_APP_UPDATE_DIR/esp_ota_ops.c:487
0x4012a8e2: esp_https_ota_finish at /COMPONENT_ESP_HTTPS_OTA_DIR/src/esp_https_ota.c:663
0x4012a9e4: esp_https_ota at /COMPONENT_ESP_HTTPS_OTA_DIR/src/esp_https_ota.c:766
```

deldering-momo, 2025-02-26:
I've honed in on the issue (for me at least). In `esp_flash_write_encrypted` the `lock_once` variable seems to determine if the Guru is triggered. 
```
    /* For buffer in internal RAM already, we only need to lock only once.
       While for buffer in flash, we need to copy data from flash to internal RAM before
       encrypted write every time. That means we need to lock/unlock before/after encrypted
       write every time.
    */
    lock_once = esp_ptr_in_dram(buffer);
```
Having the `s_encryption_write_lock` still active during `s_verify_write` seems to cause the problem. Forcing `lock_once` to `false` solves the problem for me as it calls `s_encryption_write_unlock` before every `s_verify_write`.

mahavirj, 2025-02-26:
@deldering-momo  Good catch! Rather then forcing `lock_once` to `false`, we can handle it like this: [patch](https://gist.github.com/mahavirj/ece7853ae9acc595dd055008568ec03e)

deldering-momo, 2025-02-26:
@mahavirj I agree that just disabling it shouldn't be the long term solution 😉. Thanks!

closed 2025-03-06 by espressif-bot

AxelLin, 2025-03-30:
@mahavirj @mythbuster5 

This issue was reported against v5.4, v5.3.2, v5.2.3.
But I don't find this fix in any release branches, can you apply this fix to impacted branches?