espressif/esp-idf #15382: Captive portal not working

JVKran, 2025-02-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The [captive portal example of release V5.4](https://github.com/espressif/esp-idf/tree/release/v5.4/examples/protocols/http_server/captive_portal) as well as [on version V5.3.2](https://github.com/espressif/esp-idf/tree/v5.3.2/examples/protocols/http_server/captive_portal) doesn't work.

On my Galaxy S23, the captive portal doesn't open; it's simply recognized as network without internet access. Windows 11 does connect, but most of the times it disconnects not much later.

I've done nothing but create the example through Visual Studio Code, called 'idf.py set-target esp32s3' and flashed it to my ESP32-S3-DevKitC-1.

Best regards,

Jochem

JVKran, 2025-02-13:
To provide some further details, I've made [a screencapture of my phone](https://drive.google.com/file/d/1aJ5ncQIsDMJFB9A3bbUxqULZoipmGrny/view?usp=sharing) and provided the logs of said screencapture underneath.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4 2nd stage bootloader
I (27) boot: compile time Feb 12 2025 13:07:27
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1bc6ch (113772) map
I (108) esp_image: segment 1: paddr=0002bc94 vaddr=3fc99c00 size=04384h ( 17284) load
I (112) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8a5f8h (566776) map
I (212) esp_image: segment 3: paddr=000ba620 vaddr=3fc9df84 size=005e4h (  1508) load
I (213) esp_image: segment 4: paddr=000bac0c vaddr=40374000 size=15b14h ( 88852) load
I (236) esp_image: segment 5: paddr=000d0728 vaddr=600fe100 size=0001ch (    28) load
I (244) boot: Loaded app from partition at offset 0x10000
I (244) boot: Disabling RNG early entropy source...
I (255) cpu_start: Multicore app
I (264) cpu_start: Pro cpu start user code
I (264) cpu_start: cpu freq: 160000000 Hz
I (264) app_init: Application information:
I (264) app_init: Project name:     captive_portal
I (268) app_init: App version:      1
I (272) app_init: Compile time:     Feb 13 2025 07:53:56
I (277) app_init: ELF file SHA256:  bb9b9809f...
I (281) app_init: ESP-IDF:          v5.4
I (285) efuse_init: Min chip rev:     v0.0
I (288) efuse_init: Max chip rev:     v0.99
I (292) efuse_init: Chip rev:         v0.1
I (296) heap_init: Initializing. RAM available for dynamic allocation:
I (303) heap_init: At 3FCA2300 len 00047410 (285 KiB): RAM
I (308) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (313) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (318) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (325) spi_flash: detected chip: generic
I (327) spi_flash: flash io: dio
W (330) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (343) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (349) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (356) main_task: Started on CPU0
I (376) main_task: Calling app_main()
I (406) pp: pp rom version: e7ae62f
I (406) net80211: net80211 rom version: e7ae62f
I (416) wifi:wifi driver task: 3fcac53c, prio:23, stack:6656, core=0
I (416) wifi:wifi firmware version: 48ea317a7
I (416) wifi:wifi certification version: v7.0
I (416) wifi:config NVS flash: disabled
I (416) wifi:config nano formatting: disabled
I (426) wifi:Init data frame dynamic rx buffer num: 32
I (426) wifi:Init static rx mgmt buffer num: 5
I (436) wifi:Init management short buffer num: 32
[... 38 lines cut ...]
I (31096) example_dns_redirect_server: Waiting for data
I (31146) example_dns_redirect_server: Received 60 bytes from 192.168.4.2 | DNS reply with len: 76
I (31146) example_dns_redirect_server: Waiting for data
I (31336) example_dns_redirect_server: Received 34 bytes from 192.168.4.2 | DNS reply with len: 50
I (31346) example_dns_redirect_server: Waiting for data
I (31386) example_dns_redirect_server: Received 42 bytes from 192.168.4.2 | DNS reply with len: 58
I (31386) example_dns_redirect_server: Waiting for data
I (31406) example_dns_redirect_server: Received 37 bytes from 192.168.4.2 | DNS reply with len: 53
I (31406) example_dns_redirect_server: Waiting for data
I (31516) example: Serve root
I (38536) example_dns_redirect_server: Received 32 bytes from 192.168.4.2 | DNS reply with len: 48
I (38546) example_dns_redirect_server: Waiting for data
I (42916) example_dns_redirect_server: Received 56 bytes from 192.168.4.2 | DNS reply with len: 72
I (42916) example_dns_redirect_server: Waiting for data
I (46076) example_dns_redirect_server: Received 39 bytes from 192.168.4.2 | DNS reply with len: 55
I (46076) example_dns_redirect_server: Waiting for data
I (50466) example_dns_redirect_server: Received 37 bytes from 192.168.4.2 | DNS reply with len: 53
I (50476) example_dns_redirect_server: Waiting for data
I (56136) example_dns_redirect_server: Received 39 bytes from 192.168.4.2 | DNS reply with len: 55
I (56136) example_dns_redirect_server: Waiting for data
I (56406) wifi:station: 56:c0:0f:d5:ac:a5 leave, AID = 1, reason = 3, bss_flags is 691315, bss:0x3fca4fa0
I (56406) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (56406) wifi:<ba-del>idx:2, tid:0
I (56416) example: station 56:c0:0f:d5:ac:a5 leave, AID=1, reason=3
```

hrushikesh430, 2025-02-19:
Hi @JVKran, I tried captive portal on Samsung S23, and it is totally working fine. I watched your demo video, and seems that you are doing `connect only this time`, which is mobile feature that if the internet is not available then it disconnects and connects to wifi which has internet access. Instead of this click on `Always connect` and in wifi setting of `esp32_ssid` and click on `manage route`, you will able to see captive portal hello world page.

JVKran, 2025-02-19:
Manage router works just fine as well using connect only this time. The idea of a captive portal is to pop up when connecting to a network that requires agreeing to terms of some kind in buses and in hotels. None of that is happening. What you're doing with manage router is how an Access Point works, you don't need a captive portal for that functionality.

hrushikesh430, 2025-02-21:
Hi @JVKran, This seems to be issue regarding Samsung mobiles, you can refer this docs [doc1](https://github.com/tonyp7/esp32-wifi-manager/issues/57) and [doc2](https://forum.developer.samsung.com/t/not-showing-popup-window-with-auth-when-connect-to-captive-portal/23164). You can check if it works well by changing the IP range or try out with another mobile.

hrushikesh430, 2025-03-10:
Hello @JVKran, closing this issue to response delay. Please feel free to reopen the issue for any further queries. Thank you.

closed 2025-03-10 by hrushikesh430