espressif/esp-idf #15384: The esp32 product was developed in a Linux environment, but there are inquiries about how to produce it in a Windows environment.

hgwc, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
using esp-idf v4.3.1-3-g954d44c7b9-dirty 2nd stage bootloader
Chip is ESP32-D0WD-V3 (revision 3)

We developed in the ESP-IDF environment on Ubuntu.
However, since the production department is in a Windows environment, I would like to produce in a Windows environment.

We use secure boot v1, flash encryption, nvs encryption.
The command to insert a program into esp32 when developing is as follows.
1) python ../espefuse.py --port /dev/ttyUSB0 burn_key secure_boot_v1 secure-bootloader-key-256.bin
2) python ../esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x00 bootloader-digest.bin
3) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800  --before default_reset --after no_reset write_flash 0x22000 ~/st-device-sdk-c-ref/iot-core/tools/keygen/keys/nvs_keys.bin
4) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after no_reset write_flash 0x11000 ~/st-device-sdk-c-ref/iot-core/tools/keygen/encrypted_L2510004.bin
5) ~/st-device-sdk-c-ref/bsp/esp32/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after no_reset write_flash 0x19000 ~/st-device-sdk-c-ref/bsp/esp32/components/nvs_flash/nvs_partition_generator/test/encrypted_nvs_info.bin
6) sptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x10000 partition_table/partition-table.bin 0x1f000 ota_data_initial.bin 0x30000 SLBLDM-1000MOS-SCHOOL.bin

My questions
How should I use Flash Download Tools available in a Windows environment?
1) Can I import the hex file created in Ubuntu to Windows and download it to the address?
2) Which address should I download “secure-bootloader-key-256.bin” to?

Best Regards,

hgwc, 2025-02-24:
Please give me some advice.

arosyc, 2025-02-24:
Hi @hgwc ,
1. Can I import the hex file created in Ubuntu to Windows and download it to the address?
  Yes.

2. Which address should I download “secure-bootloader-key-256.bin” to?

This key file is random and associated with secure-bootloader-key-256.bin.  

Therefore, when using the Flash Download Tool, you only need to enable Secure Boot V1. The tool will generate a new random key each time it flashes a new device.

https://docs.espressif.com/projects/esp-test-tools/en/latest/esp32/production_stage/tools/flash_download_tool.html#encrypted-download

closed 2025-03-27 by Alvin1Zhang