espressif/esp-idf #15385: Slow File Download Speed in file_serving Example

DuRuofu, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

#### **Issue Description**  

I encountered an issue with slow file download speed while using the [file_serving](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving) example.  

#### **Test Environment**  

- **Hardware**: ESP32-S3  
- **Storage**: SD card (SDMMC interface, 1-bit mode)  
- **Network**: Local LAN  

#### **Test Results**  
- The file download speed via HTTP from ESP32 is around **100-300 KB/s**, as shown in the image below:  

[image]

- This is far below my expectation (at least **1MB/s**).  

#### **Optimization Attempts**  
To rule out network bottlenecks, I tested the TCP throughput using the [Iperf](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/iperf) example (ESP32 as the server, PC as the client):  

[image]

The results indicate that ESP32's TCP transmission capability is significantly higher than the current file download speed.  

I applied the **menuconfig settings** from the [Iperf ](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/iperf) example to the `file_serving` example to improve network performance:  

```
#
# ESP32S3-specific
#
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=32
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65535
CONFIG_LWIP_TCP_WND_DEFAULT=65535
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64

CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ=240

CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y

CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB=y
CONFIG_ESP32S3_INSTRUCTION_CACHE_LINE_32B=y
CONFIG_ESP32S3_INSTRUCTION_CACHE_WRAP=y

```

However, these adjustments did not bring significant improvements, and the file download speed remains at 100-300 KB/s.

##### Questions & Requests

1. What is the theoretical maximum file download speed via HTTP on ESP32?
2. How can I further optimize the file_serving example to achieve a download speed closer to ESP32's network throughput capability?
3. Are there other key factors affecting file download speed that should be adjusted?

I would appreciate any suggestions. Thank you!

igrr, 2025-02-13:
To get higher throughput when writing to SD cards, you need to perform wrote operations in large chunks, up to FAT filesystem cluster size. I suggest starting with at least 16kB chunks.

SD card writes are DMA-driven, therefore you also can to let them run in parallel with TCP download process. You can do this by having two tasks, one task would be only reading from TCP and writing into a ringbuffer, another task reading from the ringbuffer and writing into the SD card.

I made a related example some time ago, although that one was set up as an HTTP client, rather than an HTTP server, but the principle is the same:
https://github.com/espressif/idf-extra-components/tree/0486c7e09337589f6d3a0711c67301042e701037/download_file

Hope this helps.

nileshkale123, 2025-02-26:
Hello @DuRuofu 
Thank you for reaching out. 

The download speed from the ESP device is not only limited by the available network bandwidth or TCP configuration but is also significantly influenced by the storage read speed (flash or SD card). The ESP device has a hardware-imposed limit on read throughput, which affects how fast the file can be served.

If you are serving files from internal flash, the read speed is typically lower than an SD card, especially if the flash is being accessed via SPI. Similarly, if using an SD card, the performance can vary based on the SD card type.

I conducted tests to measure the average download speed using both flash and an SD card on the ESP32-LyraTD-MSC board:

-     Flash:        Average Download Speed: **48,312.90 bytes/sec**
-     SD Card:   Average Download Speed: **85,264.60 bytes/sec**

Please feel free to reach out if you have any questions. If there are no further concerns, you may close the issue.

nileshkale123, 2025-03-05:
Hello @DuRuofu ,

I am closing this issue, and I hope we have addressed your query adequately. Please feel free to reopen it if needed.
Thank you

closed 2025-03-05 by nileshkale123